arendst/Tasmota #7986: Tasmota hangs after typing 'Wificonfig 2' followed by 'Wificonfig'

ma261065, 2020-03-23:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Tasmota hangs after typing 'Wificonfig 2' followed by 'Wificonfig'

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Brilliant Smart Wifi Plug
- [ ] Tasmota binary firmware version number used: 8.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
05:31:21 RSL: stat/tasmota_BB2A78/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
05:31:21 RSL: stat/tasmota_BB2A78/RESULT = {"Module":{"1":"Sonoff Basic"}}
05:31:22 RSL: stat/tasmota_BB2A78/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
05:32:00 RSL: stat/tasmota_BB2A78/STATUS = {"Status":{"Module":1,"FriendlyName":["Tasmota"],"Topic":"tasmota_BB2A78","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
05:32:00 RSL: stat/tasmota_BB2A78/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:06:08","StartupUTC":"2020-03-23T04:25:52","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2020-03-23T05:21:08","SaveCount":11,"SaveAddress":"F9000"}}
05:32:00 RSL: stat/tasmota_BB2A78/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-20T14:45:23","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"314/699"}}
05:32:00 RSL: stat/tasmota_BB2A78/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["DODO-31BF",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
05:32:00 RSL: stat/tasmota_BB2A78/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
05:32:00 RSL: stat/tasmota_BB2A78/STATUS5 = {"StatusNET":{"Hostname":"tasmota_BB2A78-2680","IPAddress":"192.168.1.7","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:BB:2A:78","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
05:32:00 RSL: stat/tasmota_BB2A78/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_BB2A78","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
05:32:00 RSL: stat/tasmota_BB2A78/STATUS7 = {"StatusTIM":{"UTC":"2020-03-23T04:32:00","Local":"2020-03-23T05:32:00","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:45","Sunset":"19:07"}}
05:32:00 RSL: stat/tasmota_BB2A78/STATUS10 = {"StatusSNS":{"Time":"2020-03-23T05:32:00"}}
05:32:00 RSL: stat/tasmota_BB2A78/STATUS11 = {"StatusSTS":{"Time":"2020-03-23T05:32:00","Uptime":"0T00:06:08","UptimeSec":368,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"DODO-31BF","BSSId":"D0:6F:82:43:31:C8","Channel":1,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
05:48:51 CMD: weblog 4
05:48:51 RSL: stat/tasmota_BB2A78/RESULT = {"WebLog":4}
05:48:52 CFG: Saved to flash at F6, Count 14, Bytes 4096
05:48:56 CMD: wificonfig
05:48:56 SRC: WebConsole from 192.168.1.13
05:48:56 CMD: Group 0, Index 1, Command "WIFICONFIG", Data ""
05:48:56 RSL: stat/tasmota_BB2A78/RESULT = {"WifiConfig":{"2":"WifiManager"}}
05:48:59 WIF: Checking connection...
05:49:04 CMD: wificonfig 2
05:49:04 SRC: WebConsole from 192.168.1.13
05:49:04 CMD: Group 0, Index 1, Command "WIFICONFIG", Data "2"
05:49:04 RSL: stat/tasmota_BB2A78/RESULT = {"WifiConfig":{"2":"WifiManager"}}

<Type 'wificonfig <CR><LF>' but it doesn't appear in the console>
<Tasmota hangs>
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Type 'wificonfig'
Type 'wificonfig 2'
Type 'wificonfig' again
Tasmota stops responding, and I need to re-connect to the 'tasmota_xxx' access point and re-enter the WiFi access point details

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Tasmota continues to respond

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ma261065, 2020-03-23:
Just to pre-empt this, I have done a 'reset 1' but the problem persists

arendst, 2020-03-23:
As designed! Wificonfig 2 stops all comms and starts wifimanager. There is no console so you will never see any response.

I suggest you try, while driving your car in the fourth gear, to set it in reverse and see what happens. Contact your dealer too.

ma261065, 2020-03-23:
Thanks for the useful, but also slightly snarky reply.

closed 2020-03-23 by ma261065

arendst, 2020-03-23:
Sorry, I couldn't resist. Won't do it again.