arendst/Tasmota #7991: Shelly 2.5 gets significantly hotter since 8.2.0.1

kugelkopf123, 2020-03-24:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
With the current version from the Dev Branch my Shelly2.5 is obviously much warmer than with the version from 15.03.2020.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Shelly2.5
- [ ] Tasmota binary firmware version number used: 8.2.0.1
  - [ ] Pre-compiled
  - [x ] Self-compiled
    - [ x] IDE / Compiler used: PlatformIO (MS VS)
- [ ] Flashing tools used: OTA
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
Shelly with Version 8.1.0.11:

03:53:22 MQT: stat/Shelly25_5/RESULT = {"NAME":"Shelly 2.5 (bu","GPIO":[56,0,19,0,21,127,0,0,6,126,5,22,156],"FLAG":2,"BASE":18}
03:53:22 MQT: stat/Shelly25_5/RESULT = {"Module":{"0":"Shelly 2.5 (bu"}}
03:53:23 MQT: stat/Shelly25_5/RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"0":"None"},"GPIO2":{"19":"Button3"},"GPIO3":{"0":"None"},"GPIO4":{"21":"Relay1"},"GPIO5":{"127":"Button2in"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"6":"I2C SDA"},"GPIO13":{"126":"Button1in"},"GPIO14":{"5":"I2C SCL"},"GPIO15":{"22":"Relay2"},"GPIO16":{"156":"ADE7953 IRQ"}}

Shelly with Version 8.2.0.1:

04:12:29 MQT: stat/Shelly25_4/RESULT = {"NAME":"Shelly 2.5 (bu","GPIO":[56,0,19,0,21,127,0,0,6,126,5,22,156],"FLAG":2,"BASE":18}
04:12:30 MQT: stat/Shelly25_4/RESULT = {"Module":{"0":"Shelly 2.5 (bu"}}
04:12:30 MQT: stat/Shelly25_4/RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"0":"None"},"GPIO2":{"19":"Button3"},"GPIO3":{"0":"None"},"GPIO4":{"21":"Relay1"},"GPIO5":{"127":"Button2in"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"6":"I2C SDA"},"GPIO13":{"126":"Button1in"},"GPIO14":{"5":"I2C SCL"},"GPIO15":{"22":"Relay2"},"GPIO16":{"156":"ADE7953 IRQ"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
Shelly with Version 8.1.0.11:
:53:49 MQT: stat/Shelly25_5/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
03:53:49 MQT: stat/Shelly25_5/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Free":370,"Rules":"on SHUTTER1#Position do backlog publish stat/%topic%/RESULT {"ShutterPosition1":%value%}; publish cmnd/%topic%/ShutterPosition1 %value% endon"}
03:53:49 MQT: stat/Shelly25_5/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

Shelly with Version 8.2.0.1:
04:13:57 MQT: stat/Shelly25_4/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
04:13:57 MQT: stat/Shelly25_4/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Free":370,"Rules":"on SHUTTER1#Position do backlog publish stat/%topic%/RESULT {"ShutterPosition1":%value%}; publish cmnd/%topic%/ShutterPosition1 %value% endon"}
04:13:57 MQT: stat/Shelly25_4/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

Shelly with Version 8.1.0.11:

03:55:19 MQT: stat/Shelly25_5/STATUS = {"Status":{"Module":0,"FriendlyName":["Shelly25_5","Shelly25_52"],"Topic":"Shelly25_5","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
03:55:19 MQT: stat/Shelly25_5/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"wohnzimmer","OtaUrl":"http://192.168.1.32:8000/shelly25.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:57:39","StartupUTC":"2020-03-24T01:57:40","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2020-03-20T18:24:36","SaveCount":763,"SaveAddress":"1F9000"}}
03:55:19 MQT: stat/Shelly25_5/STATUS2 = {"StatusFWR":{"Version":"8.1.0.11(tasmota)","BuildDateTime":"2020.03.24 02:55:46","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"414/699"}}
03:55:19 MQT: stat/Shelly25_5/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Eiswuerstchen",""],"TelePeriod":300,"Resolution":"559980C0","SetOption":["0000C80B","1405C8000100060000005A00000000000000","408000E8","00000000"]}}
03:55:19 MQT: stat/Shelly25_5/STATUS4 = {"StatusMEM":{"ProgramSize":497,"Free":1528,"Heap":23,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000407","8F000197","040003A1","00000004","010000C0","40008100","00000000"],"Drivers":"1,2,3,4,9,10,27,30","Sensors":"2,3"}}
03:55:19 MQT: stat/Shelly25_5/STATUS5 = {"StatusNET":{"Hostname":"shelly25-5","IPAddress":"192.168.1.207","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"98:F4:AB:F3:0F:80","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
03:55:19 MQT: stat/Shelly25_5/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.73","MqttPort":1883,"MqttClientMask":"Shelly25_5","MqttClient":"Shelly25_5","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
03:55:19 MQT: stat/Shelly25_5/STATUS7 = {"StatusTIM":{"UTC":"2020-03-24T02:55:19","Local":"2020-03-24T03:55:19","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"06:23","Sunset":"18:51"}}
03:55:19 MQT: stat/Shelly25_5/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
03:55:19 MQT: stat/Shelly25_5/STATUS10 = {"StatusSNS":{"Time":"2020-03-24T03:55:19","ANALOG":{"Temperature":47.9},"ENERGY":{"TotalStartTime":"2020-03-18T23:06:37","Total":0.202,"Yesterday":0.005,"Today":0.000,"Power":[0.000,0.000],"ApparentPower":[0.000,0.000],"ReactivePower":[0.000,0.000],"Factor":[0.00,0.00],"Voltage":0,"Current":[0.000,0.000]},"Shutter1":{"Position":0,"Direction":0,"Target":0},"TempUnit":"C"}}
03:55:19 MQT: stat/Shelly25_5/STATUS11 = {"StatusSTS":{"Time":"2020-03-24T03:55:19","Uptime":"0T00:57:39","UptimeSec":3459,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Eiswuerstchen","BSSId":"10:9A:DD:84:A0:07","Channel":6,"RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:09"}}}

Shelly with Version 8.2.0.1:

04:37:45 MQT: stat/Shelly25_4/STATUS = {"Status":{"Module":0,"FriendlyName":["Shelly25_4","Shelly25_42"],"Topic":"Shelly25_4","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
04:37:45 MQT: stat/Shelly25_4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"wohnzimmer","OtaUrl":"http://192.168.1.32:8000/shelly25.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:03:09","StartupUTC":"2020-03-24T03:34:36","Sleep":50,"CfgHolder":4617,"BootCount":2,"BCResetTime":"2020-03-20T18:24:44","SaveCount":10,"SaveAddress":"1F8000"}}
04:37:45 MQT: stat/Shelly25_4/STATUS2 = {"StatusFWR":{"Version":"8.2.0.1(tasmota)","BuildDateTime":"2020.03.24 04:29:37","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"414/699"}}
[... 22 lines cut ...]
#undef MQTT_HOST_DISCOVERY   
#define MDNS_ENABLED           1
#undef USE_ADC_VCC
#define USE_ADE7953
#define USE_SHUTTER
#define USE_EXPRESSION
#define SUPPORT_IF_STATEMENT
```

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Run with normal Temperature.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

The two Shellys2.5 are both flush-mounted in one wall directly next to each other. To have a direct comparison, I flashed only one of the two Shellys back to the state of the 15.03.2020. 

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-03-24:
Just did a quick check and I see indeed a higher temperature reading but using an external thermometer it is still at the "old" temperature.

Apparently something has changed aroung the analog input (where the shelly temp sensor is connected). I'll investigate.

BTW. Could you report the output of command ``adcparam`` for both shellies pls.

arendst, 2020-03-24:
As I upggraded my shelly 2.5 yesterday I see indeed a jump in temperature (+10 C) as soon as the latest development firmware version is active.

So this must be a software fault and not a real high temp error.  Continue investigating...

kugelkopf123, 2020-03-24:
Hi!
Thanks for checking. It's good to hear that only the display has changed and not the real temperature.

adcparam for 8.11.0.1: 
`11:30:51 MQT: stat/Shelly25_5/RESULT = {"AdcParam":[2,32000,10000,3350]}`

for 8.2.0.1:
`11:31:05 MQT: stat/Shelly25_4/RESULT = {"AdcParam":[2,32000,10000,3350]}`

arendst, 2020-03-24:
Pls also provide the output of command ``webrefresh``

arendst, 2020-03-24:
I currently see no relation to firmware changes. I do see an increase in measured temperature but not the amount you encounter. Also opening the webpage increases temp with 10 C but that's normal as it has to update the webpage regularly.

Are you sure the "hot" shelly reported cooler temps before or is this a new shelly with possible different NTC hardware?

kugelkopf123, 2020-03-24:
Both Shellys give:

`11:58:13 MQT: stat/Shelly25_4/RESULT = {"WebRefresh":2345}`
as answer. 
Yes I had investigate myself in the Code but cant find anything. 
This Morning I had updated all Shellys2.5 to V8.2.0.1 then I noticed the higher values in all of them. To have a better comparison, I took exactly the pair that sits directly next to each other in the wall and only brought one back to the state of 15.03.2020 and thus had directly lower values. 
Apart from that, both Shellys came from the same box, I assume they have the same hardware.

arendst, 2020-03-24:
Agree. Will need more investigation....

arendst, 2020-03-24:
Problem is I do not feel the higher temp. I also do not measure the higher temp.

meingraham, 2020-03-24:
@kugelkopf123 

To eliminate difference in the hardware, perhaps you can reverse the firmware versions on your two Shelly devices to see if the higher temp readings follow the firmware regardless of which device. I know you had both on the new firmware but having one each on the older & the newer with the same environmental conditions and then "swapping" may provide some additional clues.

Mike

kugelkopf123, 2020-03-24:
@meingraham Did it. I have tried your suggestion and can confirm that the error goes with it. In general it was obvious, because Theo could also confirm it with his Shelly25 in his frist post.

kugelkopf123, 2020-03-24:
[image]

The ambient Temp has rised that is the reason why the overall Temp is higher than this morning at 4 o Clock.

meingraham, 2020-03-24:
Yes... it just makes it "definitive". Eliminates that question.

enricota66, 2020-03-24:
My device also heats up a lot.
I tried with "Sleep 250" and "WifiPower" in command.
But WifiPower doesn't seem to work.
I put WifiPower 1 with the device near the router, but the signal does not go down
[image]

kugelkopf123, 2020-03-24:
> My device also heats up a lot.
> I tried with "Sleep 250" and "WifiPower" in command.
> But WifiPower doesn't seem to work.
> I put WifiPower 1 with the device near the router, but the signal does not go down
> [image]

BTW. I tested this WiFipower a week before on another Shelly that is 1m away from the router,too. Tried it down to the value of 0.1. nothing changed. Nor in the information or on the router setup page.

Jason2866, 2020-03-25:
First step it needs a measurement if the device is really! getting warmer.
This can NOT be done with Tasmota. Use a extra thermometer to measure with old and new firmware.
Depending on this we know (a bit more) where to look.

enricota66, 2020-03-25:
I have tested with version 8.1.0.I have read similar heat values of version 8.2.0.1
I don't have a thermoscanner but at the touch of the device the values seem plausible.
I have seen that temperatures rise a lot when the reays are turned on.
However, high temperatures do not depend on the load applied.
I have no experience with Shelly because I have sonoffs in my home.
I'm afraid Shelly's heat may be a device construction problem

kugelkopf123, 2020-03-25:
Unfortunately I currently have no Shelly2.5 that is not yet installed in any wall on my desk. I have to remove one first. Furthermore I also lack a suitable thermometer to validate the values. Also I could only operate the Shelly via 3.3V and feel it with my finger when the case is open. But this would be a very subjective evaluation. 
@enricota66 Actually the design of the Shelly2.5 is not yet the yellow one. Switching on the relays increases the temperature. That is correct. However, I use all of the Shelly2.5 in my home to control my blinds. So the relays are only running for a few seconds. My tests have always been done with relays switched off for a long time. Does i get you right? Have you done your test now with a sonoff device or shelly?

If I understood Theo's first post correctly, he could not detect any physical increase in temperature. 
I can see that the temperature is high at the beginning and then lower when flashing back, but this can also be related to the fact that the built-in ESP works more when flashing and therefore a high temperature is reached.

enricota66, 2020-03-25:
I Have sonoff basic and sonoff mini inside the electrical boxes of the wall to control the house lights and HomeAssitant automatisms and MQTT protocol. Everything works perfectly with the Tasmota firmware.
My intent was to replace 2 basic sonoff devices inside a box with Shelly 2.5
Before doing this I am tensing shelly on the desk for my use the relays must remain on for a long time.
I have 2 shelly2.5 devices on the desk, the first with one with tasmota firmware and the second with original firmware.
From the tests I'm doing with 2 relays on, the original one seems to be more cooler.
However I want to measure the temperature with an instrument

Jason2866, 2020-03-25:
Please dont mix up different devices in this thread. It is about Shelly 2.5
Feeling is no validate way of measuring ;-) 
Please post only measured values of a Shelly 2.5

arendst, 2020-03-25:
@enricota66 pls do measure externally. The shelly 2.5 internal temperature is measured by a NTC thermistor. To get a reasonable reading it's result needs to be calculated. The current calculation is based on parameters I once received from Shelly. These parameters might have been changed so in that case the ``AdcParam`` parameters need a change too.

enricota66, 2020-03-25:
> @enricota66 pls do measure externally. The shelly 2.5 internal temperature is measured by a NTC thermistor. To get a reasonable reading it's result needs to be calculated. The current calculation is [...]

Tried one hour and 30 minutes with the relays on.
Indeed, there seem to be no important differences in the devices
( Tasmota ver is 8.2.0.1)

[image]
[image]

arendst, 2020-03-25:
Great! Now what are the temperatures reported by Shelly and Tasmota in those situations?

enricota66, 2020-03-25:
> Great! Now what are the temperatures reported by Shelly and Tasmota in those situations?

Sure
The Tasmota temperature in that situation is around 60 c.
But I was unable to find Shelly's data in its application.
Can anyone help me, where can I find it?

arendst, 2020-03-25:
Background info shelly  https://www.facebook.com/groups/1686781668087857/permalink/2054834997949187/

ascillato, 2020-03-25:
From previous link:

> About question:How much is PCB operating temperature to Shelly2.5, Shelly1PM and Shelly PlugS:
> Max ambient temperature is 40 degree. With no load PCB temperature is 55-60 degree. At MAX load continuesly is 87-90 degree.
> Heating protection will switch off device at 95 degree.
> All parts inside are 105-120 degree certifed for continuous usage.

So, the actual value of Tasmota is Right and in previous versions was wrong.

effelle, 2020-03-25:
Meaning the Shelly 2.5 needs a special `adcparam` setup to work properly, I suppose. 
Not easy without knowing the actual resistor used and what NTC is using...

kugelkopf123, 2020-03-25:
> > @enricota66 pls do measure externally. The shelly 2.5 internal temperature is measured by a NTC thermistor. To get a reasonable reading it's result needs to be calculated. The current calculation is based on parameters I once received from Shelly. These parameters might have been changed so in that case the `AdcParam` parameters need a change too.
> 
> Tried one hour and 30 minutes with the relays on.
> Indeed, there seem to be no important differences on the devices
> ( Tasmota ver is 8.2.0.1)
> 
> [image]
> [image]

First of all, thank you very much for your commitment!
So how exactly is this meant?
The temperature of the measured external thermometer and the one displayed in the Tasmota WebGUI did not differ?

Or both devices, one with the current Dev Branch and one with the original FW, got almost identical temperatures on the external thermometer?

If that was the case, what did the ShellyConnect app provide for a temperature value at that time? 
And was this value identical to the value from the Tasmota WebGUI?

effelle, 2020-03-25:
@enricota66 now that you have your Shellys in "test mode", you could try to change the value of the resistor using the command `adcparm` and see if you can match the temperature shown by the one with Tasmota and your thermometer. 
The default values are `2,32000,10000,3350`, where the third value is the resistor (10k). Changing the resistor value you will have a proportional higher/lower reading. 
Dai che risolviamo! :)

arendst, 2020-03-25:
Stop!

No need to change adcparam. The internal shelly tmperature IS 60 degrees C as also reported by shelly. See link above.

The shelly should report it's internal temp and not the external temp.

As already was concluded the tasmota adcparam are ok (I got them from shelly) and result in the same theoretic shelly temp.

To extract the original shelly temp a mqtt server is needed to address a documented topic within the device.

effelle, 2020-03-25:
Understood, then the NTC is **NOT** for external reading but just for thermal control. 
That's changes all.

ascillato, 2020-03-25:
the temperature is from the PCB. Measuring from the exterior is going to be a lot lower than the actual pcb temperature. As it is Tasmota now is OK. About 40 to 60C when relays are OFF.

kugelkopf123, 2020-04-01:
I found out that if you remove the ADE7953 chip from the template, the temperature is much lower. 
Maybe interesting for some people who use the Shelly2.5 only for the shutters and therefore don't need a power meter. 

`{"NAME":"Shelly 2.5 test","GPIO":[56,0,19,0,21,127,0,0,0,126,0,22,0],"FLAG":2,"BASE":18}`

ascillato, 2020-04-30:
From the shelly support group on Facebook. Posted by Shelly.

[image]

kugelkopf123, 2020-04-30:
Didnt knew that its possible to run it without N. I saw it yesterday with the Shelly Dimmer SL? I think. Would be nice to know how big this load was in this picture.

BassT23, 2020-05-18:
> I found out that if you remove the ADE7953 chip from the template, the temperature is much lower.
> Maybe interesting for some people who use the Shelly2.5 only for the shutters and therefore don't [...]

Thanks for the tip.
-8°C with ADE7953 disabled!

Smandurlo, 2020-07-26:
I have the same issue with my shelly(s) 2.5. With original firmware the temperature is 60°C with both relays ON, with Tasmota 8.3.1 the temperature is 87°C (and probably increasing) with both relays ON. Nothing connected to the relays. I flashed 8 of them with the same behaviour. Four of them were in use with the original firmware (so I know their temperature while in use), I flashed the other 4 directly from the box.
Did you find a solution for it?
Thanks!

Smandurlo, 2020-08-27:
> @ascillato <https://github.com/ascillato> , I have to disagree. There is
> a significant temperature difference between the stock firmware and Tasmota
> 8.4.0. Without any load the difference is around 20 degrees in my tests,
> with similar circumstances.
>
> Between the stock firmware and Tasmota 8.1.0 the difference is negligible.

I agree with stefxx. The only way to keep the temperature the same like the
stock firmware is to disable the ADE chip. With the ADE chip the
temperature goes too high. If I keep the web page open, without any load,
the shelly goes to 90°C and more and it switches off due to the termal
protection. This is obviously totally wrong.

kugelkopf123, 2020-08-27:
I don’t know how precisely This ade chip works. But perhaps the original software doesn’t read the chip that often as Tasmota do? Perhaps this could be a reason. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

stefxx, 2020-09-12:
Can we re-open this issue? It is a real issue, not something with wrong readings

> 
> 
> From previous link:
> 
> > About question:How much is PCB operating temperature to Shelly2.5, Shelly1PM and Shelly PlugS:
> > Max ambient temperature is 40 degree. With no load PCB temperature is 55-60 degree. At MAX load continuesly is 87-90 degree.
> > Heating protection will switch off device at 95 degree.
> > All parts inside are 105-120 degree certifed for continuous usage.
> 
> So, the actual reported value of Tasmota is Right and in previous versions was reported wrongly. No real increase of PCB temperature between Tasmota versions.

Either way, there is a real increase in the actual temperature of the device since version 8.2.0. It can be felt easily by simply touching the Shelly 2.5 by hand. I have two of them next to each other, one with 8.1 and one with 8.4 and the difference is huge!

Can we please reopen this issue and have another look into this? Thanks!

Smandurlo, 2020-09-12:
I think so. I keep the ADE chip not reading the power consumption otherwise
the Shelly switch off because of overheating eventually.

kugelkopf123, 2020-09-12:
I have no Idea how that can be. If you have two devices, perhaps it is possible for you to make a real mesurament. With two Sensors. One on each shelly. 
Best if its open.
Perhaps in the latest Version from Tasmota, the ADE Chip gets triggerd more often? Perhaps this is the reason...

stefxx, 2020-09-13:
To get out of the right/wrong game here, I did some more testing in a more controlled way. Note that all tests have been done with an ambient temperature of 22-22 degrees, and the Shelly 2.5 PM devices are on my tabletop, not enclosed in any way.

One device has stock firmware 1.8.3 installed. The other one has Tasmota 8.5.0, using the default Blackadder device template. They are only connected to L and N, no load connected to them.

First, with both switches ON, I’ve let them run for about 45 minutes. At that point the temperature stabilized around 75 degrees on both.
[image]
 
Then I let the browser open on both devices, which has a (known) effect on the temperature of a Tasmota device of ~10 degrees. And indeed, after 10 minutes, the Tasmota stabilized around 85 degrees. No changes to the stock Shelly:
[image]
 
Next, I stopped actively monitoring the web interface and I switched both switches off, on both devices. They stabilized just above 60 degrees.
[image]

So far, so good. Still think that the temperature is quite high, but the difference with the stock firmware is negligible. 

Next step is to downgrade the Tasmota device to firmware 8.1.0. Switches still of, no webserver session active, and after some time the temperature stabilizes around 50 degrees! More than 10 degrees less than 8.5.0!
[image]

Final test with both switches on, on both devices. Still no load connected. Stock firmware 75 degrees, Tasmota 8.1.0 “only” 62.
[image]
 
My conclusion: as the issue starter stated, there is a significant difference in Shelly 2.5PM temperatures between Tasmota versions, most likely introduced with 8.2.0. 10 to 13 degrees to be exact. However, the difference between stock firmware and Tasmota 8.5.0 is next to nothing. 

Also, in general I find the temperature of those devices high. It might be within specs, but you can feel they are getting quite hot and that will only increase when built in small boxes with no space to breathe. So, if we can lower the temperature somehow (preferably not by disabling the PWM or increase the Sleep) that would be very welcomed. The question remains what changed in 8.2.0 wrt these devices… and is it something that might be reversed or corrected?

ascillato, 2020-09-13:
Hi, the real temperature in different Tasmota versions are the same. The CALIBRATION of Tasmota versions is different. That is why you see a difference in old Tasmota versions versus the newer versions.

The real Temperature is the same. Thr one showed in the UI is different. Please, measure the real temperature of the case (or of the board) using both versions of Tasmota and also to the stock firmware.

stefxx, 2020-09-13:
I might be able to borrow a thermal camera later this week. I will repeat the test and I hope we can come to a final solution after that. Thanks!

ascillato, 2020-09-13:
Thanks, your tests are more than welcome and if needed we can ADJUST the internal calibration to show a more exact temperature value. Thanks.

BeoQ, 2020-09-15:
I do have the same issue of a shelly-2.5 getting hot temperature reading when running tasmota (-8.4.0.1).
I'm very aware that the actual temperature of the NTC and the shelly's case temperature are two different things; the NTC will be hotter due to being closer to the heat source. 
So, measuring the actual inside temperature close to the NTC and comparing this to the reading computed by tasmota, is the important task to do. 
An thermal imaging camera pointed at an opened shelly in operation would help. Don't even try this measurement if you're not trained to work on high voltage. Caution: High Risk of Electrocution! - you have been warned...

Thinking of heat sources in a shelly:

1. The ESP8266 is likely one of the larger factors in power dissipation: 
- The hardwre design guide says average 80mA, roughly 260mW dissipation at 3.3V. but recommends a supply capable of 500mA, thats 1.65 Watt potential of dissipation.
- The actual dissipation is related to the RF energy when transmitting WiFi packets (more packets to transmit, more heat).
- The computing load of the the chip requiring CPU cycles (already known: open browser-page on tasmota, see stefxx's investigations, two days ago).
- The polling load to read sensors requires CPU cycles (disabling the ADE sensor chip)

2. The two Relay (-Coils) dissipate heat when energized. 
- The shelly does not have a circuit to reduce coil current when energized in hardware; 
- To what i could see on the PCB, it's just simple driver transistors. 
- Several hundreds of mW dissipation per energized Coil is realistic.

3. The current flowing thru the Relay Contacts and the Current-Measuring Shunts. 
- Example: Each channel is rated to 10Amps, if you have 0.1Ohm of total path resistance (contact and shunt), then you have 10Watts to dissipate, that is for each Channel when under full load (!). 
- Even though the real path resistance is likely much lower, it would increase due to wear and tear of the relay contacts over time. A Thermal protection is definitely useful here!

4. The Power-supply to power all these tightly packed components.
- Depending how shelly's engineers actually designed it, i'd guess about 2-3Watt capability; with an efficiency of about 80% (estimated) that's about 600mW to dissipate (with relays energized, and lots of computing to do)

Bottom-line(s):
- There's not much you can do about the Relay contacts and shunt. Just keep the load current low.
- At tasmota compile-time, you could remove all the bells and whistles you don't really need to reduce computing load.
- When in use, you can avoid constantly pointing your Browser at the device.
- You can try to reduce WiFi traffic to/from the device, allowing longer sleep-times of the CPU
- You could drive the relays by using PWM

Improvements:
The last line might be interesting for tasmota: if the output-pin driving the relay is capable of PWM, the firmware could energize the relay with 100% duty-cycle for some 100ms, then step back to a lower duty-cycle that is sufficient to keep the relay energized, and save power. Something between 30..50% duty-cycle might be a good starting point when trying.

Adding the thermally conductive, but electrically insulating pad between the heat sources and the case would help heat transfer. Producing the outer case of the shelly using such a material would probably be something for advanced makers or the manufacturer.

Let's see where this thread goes...

ascillato, 2020-09-16:
> You could drive the relays by using PWM

Please, don't. PWM is software PWM (the esp8266 don't have hardware PWM), so the CPU goes to 100% and the sleep time to 0. So, the amount of enegy usage goes up and the temp too. And driving a relay with PWM will reduce drastically its life.

Also, take into account that we are talking about the calibration of the temperature measurement between Tasmota versions. The fact that Tasmota is publishing a higher temperature don't means that now is actually hotter than in older versions.

Let's wait for @stefxx tests.

BeoQ, 2020-09-16:
Good point for not doing PWM to drive relays. I wasn't aware of that.
If the ESP needs to to PWM in software, the additional CPU load and heat will eat up the savings in the coil, probably making things worse.

To improve my guess on dissipation, i have unhooked and opened my shelly-2.5:

The Shunt resistor is labeled "R004" in a package similar to 1206.
- The Resistor value should then be 4mOhm
- The Power dissipation at 10Amp is then 0.4 Watt

The Relay in my unit is made by Hongfa, Type "HF32FA-G 012HSL1"
- The Datasheet says, it's 'sensitive coil' is rated 12V, needs 230mW power, and picks up (energize) at 9V.
- The Contact resistance is rated max 70mOhm (at a load of 1A)
- The Power dissipation at rated 10Amp is max 7 Watt (typically much lower, but not specified)

Mains Power Supply Chip is a "LNK304DN"
- capable of delivering 120mA 
- It's own consumption in 'self powered buck mode' is 50-80mW
- efficiency is stated as 75%
- likely set to 12V output because of the relay

There is a Chip that looks like an additional Low-Voltage Regulator with marking "WBBTA" (package SOT23-6)
- I cannot find a datasheet for it. 
- Unclear to me if it is a simple linear regulator or (more likely) a switched-capacitor buck regulator (there's no inductor visible nearby).

I'll try to also get 'external' infrared temperature measurment in the internal components, to compare them with the NTC/Tasmota-readings.

kugelkopf123, 2020-09-16:
> * designator R35, no component marking

Thanks for the Infos! 
Can you post Photos from the Board and the details?

stefxx, 2020-09-18:
I finally have a Seek Compact heat camera available to do some testing, planning to do some measurements tomorrow. Never done any heat measurements before. What should I test? How should I test? Should I remove the Shelly housing? Front and/or back? Any specific circumstances?

Just want to do it right before I waste my and possible your time with incorrect measurements.

ascillato, 2020-09-19:
Similar to https://github.com/arendst/Tasmota/issues/7991#issuecomment-621995157
So, without the casing and the 2 cases: with and without a load.

BeoQ, 2020-09-19:
Finally, I did some pictures of my opened Selly-2.5.
For reference to the issue I'll post them here. Maybe someone would rather see those in the documentation section? 

**An Overview on the Top / Component side, including the top-case where the antenna is located:** 
[image]
- The two large black boxes are the relays.
- The two things in the relay PCB and next to the screw terminal blocks, are the Shunt resistors.
- All parts in the lower area belong to the Power Supply section.
- The flat mounted cylinders are electrolytic capacitors of the power supply section.
- The round thing with some heat-shrink-tube around is an inductor for the main power supply.

**An Overview on the Solder side:**
[image]
- The antenna is connected with a very small and delicate MMCX connector. 

**Detailed view on the Solder side:**
[image]
- The Chip marked U3 is the ESP8266.
- The Chip marked U4 is the program Flash.
- The Chip marked U6 is the ADE7953.
- The NTC is marked R35, it forms a voltage divider together with R36 and clearly goes to Pin-6 of the ESP.
   Measured 0.504V reads 38.9°C in Tasmota's Webinterface.
- R13 and R14 belong to the two Switch Inputs.
- R32 and R33 belong to the voltage divider for line voltage measurement of the ADE.
- R42 is a noteworthy 1kOhm resistor to load the 12V rail, and it gets very hot (144mW @12V).

**Detailed view on the power supply section components:**
[image]
- The Chip U1 is the main power supply, it delivers 12.0V (measured)
- The Chip U2 is the low-voltage regulator, it delivers 3.34V (measured)
- The designator C7 likely belongs to the smaller brownish capacitor.
- I guess, the larger thing next to the C7 writing could be a inductor in SMT, the traces on the PCB suggest switching topology.

With these pictures it should be easier to interpret pictures from the thermal imager.

BeoQ, 2020-09-19:
I managed to get a thermal camera and take some pictures with it.
Camera set to emisivity of E=0.80, which seems to be a good value for epoxies, including typical FR4 PCB material.

In all the pictures of this post, I run tasmota-8.4 and both relays are energized. 
NTC temperature readout was done via MQTT, to avoid having a browser-session working the cpu.
Ambient temp was about 25°C, almost no airflow.

**Hot-Spots in operation:**
[image]
- The shelly was running for about an hour with relays ON, so thermal equilibrium can be assumed.
- TopLeft: the hottest area is around the ESP and the power supplies.
- Topright: the Resistor R42 is by far the hottest spot.
- LowLeft: The ESP seems to have quite a lot to compute.
- LowRight: The MQTT-readout of the NTC and the thermal camera measurement match really good on my device.

With suspicious heat coming from the power supply section, i bent away the components, and took above detail picture.
The power-on after this cooldown time, shows really good where the predominant heat sources are.

**Power Supply section warmup:**
[image]
- Left: the inductor close to C7 gets hot, and does this quite fast
- Left: the relays are energized, but there's little heat due to the large surface
- Right: The R42 gets hot almost instantly.

**Thermal conclusion:**
The guys at Shelly did a great job in building the device in the smallest possible way.
But there's room for thermal improvement:
- The NTC measures mainly the temperature of the ESP Chip, and the relays to some extent.
- Checking if R42 is really needed this way.
- A larger SMT inductor could ehelp reduce component temperature and improve it's lifetime.
- A thermally conductive PCB could help dissipate the heat in all places (expensive option).
- A thermally conductive lower case with some heat-transfer pads in the hotspots could help.
- For tasmota: the NTC reading seems OK, see how to reduce the load on the CPU and improve using power-saveing cycles.

I can imagine completely removing the R42 or using a higher value:
Many switching regulator topologies require a minimum load to work properly, but maybe the load from the low-voltage regulator supplying the chips is enough. Possibly it is required in case this low-voltage-regulator fails, although probably nothing happens then - I did not remove my R42 up to now; If you do, it's on your own risk.

If you use a thermal pad to transfer heat to the bottom case: make sure it withstands your mains voltage: 
At 230V AC you have some capacitors charged to about 325V, and safety-standards DO require far higher breakdown voltages here!

I may try to do more thermal images running older tasmota-8.1, but don't count on that.
My personal guess is: The core of this issue #7991 lies in the code. Well...

ascillato, 2020-09-20:
Thanks a lot for your measurements. Those pics are amazing!!!!

Were you able to measure the temperature with the shelly stock firmware?
Anyway, from shelly's dev measurement (https://github.com/arendst/Tasmota/issues/7991#issuecomment-621995157) your values are very close to the stock firmware.

So, in conclusion, Tasmota is measuring a value that is real, right?
And Tasmota 8.1 had a bad calibration of the NTC that is why it was reporting a lower value (not real), right?

arendst, 2020-09-20:
Thx for your effort to tackle the cause of the heat problem.

It would make sense to have the same measurements done with tasmota 8.1 as that version is presumably less hot than the above used 8.4

If indeed it is a lot cooler I could try to find the cause of the heat issue.

In the meantime users could also try to compile versions of tasmota without unneeded features and see if that lowers temp too.

stefxx, 2020-09-20:
So, I finally took some thermal pictures as well. Thanks @BeoQ your pictures look perfect. Much better than mine.

I've let the device stabilize for about an hour before measurements. Web interface off. I have 1 Shelly with Stock 1.8.3 firmware, and the other one with Tasmota. So the 8.1.0 and 8.5.0 measurements are from the same physical device. Ambient temperature 22 degrees, in a small and closed room (so no external airflow).

First the Shelly firmware, both relays activated, no load attached. Reported temperature 70 degrees, measured 73 degrees:
[image]

The Tasmota 8.5.0 device, again both relays activated, no load attached. Reported 70 degrees, measured 68:
[image]

And finally, after flashing the Tasmota device with 8.1.0, the device reported  59 degrees, actual 60 (I've let the device stabilize for 30 minutes, as the reported temperature was already stable for 15 minutes). Both relays activated, no load attached. :
[image]

So it seems there is actually a measurable difference between 8.5.0 and 8.1.0. Please let me know if I should make other measurements if that is helpful. I don't have any experience with these kind of measurements so let me know if I did anything wrong.

stefxx, 2020-09-20:
I just noticed that the picture from Shelly in this thread ([#7991 (comment)](https://github.com/arendst/Tasmota/issues/7991#issuecomment-621995157)) is not a 2.5 PM device. It looks like a dimmer.

Smandurlo, 2020-09-20:
It is not a 2.5, there is no neutral. It is something, SL version

BeoQ, 2020-09-20:
@arendst, @ascillato: Since @stefxx has done all three firmwares, i'd omit the shelly stock firmware, to avoid messing up my network security monitoring. I intend to try and measure other tasmota firmwares, but not sure when i get around to do so. Might as well be as late as next weekend.

If i get around to measure, I'll put a sonoff basic with different firmwares to the test as well. I have one that i had solderd additional I/Os to, as a sensor-node with some i2c-devices and a larger number of 1-wire temp sensors. For unknown reason, the i2c does not work with my self-compiled and stripped down tasmota-8.4. I did not get around to investigate enough to open a new issue on that; but seeing where this issue on the shelly is heading to, i suspect that my sonoff basic issues might also be related to thermal or load. 
Let's see, if this issue spreads even wider than just the shelly. You wouldn't know, unless someone happens to measure the ESP chip temp or it's power consumption in other devices...

@stefxx: your thermal images also look very good!
I did three things to improve the pictures:
- set the emissivity to something useful for the material to measure. It vastly improves the accuracy of the temperature readings.
- use the same temperature scale for all measurements. E.g manually set the Min and Max of the Temperature scale. As a result, you can easily see differences visually, even across several pictures.
- Find a good lens focus. Move the focus around, when edges in the picture look 'sharpest' you're Ok. I was already on the near end of lens, so i tried varying the distance until the picture was crisp again.

Keep it up!

BeoQ, 2020-09-23:
Here we go: 
I did a measurement series with temperature vs tasmota-versions.
I did not leave too much time for the temperature to settle, just about 5-10minutes each.
Looking at the NTC reading via MQTT showed it is enough settlement to be useful, since we don't need high accuracy on the temp reading this time. All measurement done on the same device, with no browser window open, and relays energized.

**Shelly-2.5:**
[image]
- Version 8.1 and 8.2 run rather cool on the ESP
- R42 is still the hottest point, even with low load on the ESP
- Version 8.3 and 8.4 run rather hot on the ESP
- roughly 15K temperature increase at the ESP

Something happened between the releases tasmota-8.2.0 and -8.3.0 that pushes up the shelly-2.5's temperature.

**Sonoff Basic**
For comparison: the same series on a Sonoff Basic, with selected device "sonoff basic"
[image]
- It's temperature seems to be not affected by the changes.
- hottest part is the energized relay

I hope this helps finding out the root cause.

Cheers!

[17 more replies]

closed 2020-09-25 by ascillato2