arendst/Tasmota #7995: 60 MHZ Tx 61 MHZ Rx

nilsx, 2020-03-24:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I've configured a MH Z19B CO2 sensor, this worked fine in 8.1.0. However in 8.2.0 i can't find options
60 MHZ Tx and 61 MHZ Rx anymore.

I've followed this page for setting up the sensor:
https://github.com/arendst/Tasmota/wiki/MH-Z19B

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
14:57:36 MQT: stat/co2sensor/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
14:57:36 MQT: stat/co2sensor/RESULT = {"Module":{"18":"Generic"}}
14:57:36 MQT: stat/co2sensor/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"60":"MHZ Tx"},"GPIO2":{"0":"None"},"GPIO3":{"61":"MHZ Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
14:58:23 MQT: stat/co2sensor/STATUS = {"Status":{"Module":18,"FriendlyName":["Tasmota"],"Topic":"co2sensor","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:58:23 MQT: stat/co2sensor/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"tasmotas","OtaUrl":"https://github.com/arendst/Tasmota/releases/download/v7.2.0/tasmota.bin","RestartReason":"Power on","Uptime":"2T17:54:40","StartupUTC":"2020-03-21T20:03:43","Sleep":50,"CfgHolder":4617,"BootCount":30,"SaveCount":66,"SaveAddress":"F5000"}}
14:58:23 MQT: stat/co2sensor/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:33:25","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"346/699"}}
14:58:23 MQT: stat/co2sensor/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["H369AA74F63",""],"TelePeriod":120,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
14:58:23 MQT: stat/co2sensor/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
14:58:23 MQT: stat/co2sensor/STATUS5 = {"StatusNET":{"Hostname":"co2sensor-2011","IPAddress":"192.168.2.103","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.254","Mac":"A4:CF:12:D9:87:DB","Webserver":2,"WifiConfig":4}}
14:58:23 MQT: stat/co2sensor/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.18","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D987DB","MqttUser":"openhabian","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
14:58:23 MQT: stat/co2sensor/STATUS7 = {"StatusTIM":{"UTC":"Tue Mar 24 13:58:23 2020","Local":"Tue Mar 24 14:58:23 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"06:43","Sunset":"19:09"}}
14:58:23 MQT: stat/co2sensor/STATUS10 = {"StatusSNS":{"Time":"2020-03-24T14:58:23","MHZ19B":{"Model":"B","CarbonDioxide":774,"Temperature":22.0},"TempUnit":"C"}}
14:58:23 MQT: stat/co2sensor/STATUS11 = {"StatusSTS":{"Time":"2020-03-24T14:58:23","Uptime":"2T17:54:40","UptimeSec":237280,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"H369AA74F63","BSSId":"7C:39:53:A7:4F:63","Channel":1,"RSSI":100,"Signal":-45,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Update from 8.1.0 to 8.2.0.  

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Add back the options so i can configure my device like this : 

For Sonoff Basic:

Module Type: 01 Sonoff Basic
GPIO01 60 MHZ Tx
GPIO03 61 MHZ Rx

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-03-24:
Every new release has changes and it is well documented.
Use tasmota-sensors...

blakadder, 2020-03-24:
You are using documentation for a very old version of Tasmota.
 
Expanded sensor support is moved to tasmota-sensors.bin.

closed 2020-03-24 by blakadder