arendst/Tasmota #7997: TuyaMCU weird dimming range and cannot change.

GrizzlyCH, 2020-03-24:
### PROBLEM DESCRIPTION
Hi

I just flashed the last version of tasmota on a DM_WF_MDV4 dimmer.
I read and executed much tutorials and allways the same result.

I setup the port (rx tx) and set TuyaMCU 21,2 (the onlyone who work a bit)
Wenn i set dimmerrange 0,254, my light dont change from 0% to 50% on the slide bar, and then dim from "no light" to "full light" wenn the slide bar is from 50% to 100%.
I tryed to put dimmerrange 128,254, to cut the useless first 50% and now it dim from medium to more medium light, from 30% to 60%... i cannot reach low light and full light.

I realy dont know where to search more and what to do.

EXCUSE ME FOR MY BAD ENGLISH, IT IS NOT MY FIRST LANGUAGE :-(

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): DM_WF_MDV4
- [x] Tasmota binary firmware version number used: Tasmota 8.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
17:07:11 CMD: Backlog Template; Module; GPIO 255
17:07:11 SRC: WebConsole from 172.18.200.100
17:07:11 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
17:07:11 SRC: Backlog
17:07:11 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
17:07:11 MQT: stat/Vario_Chambre/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
17:07:11 SRC: Backlog
17:07:11 CMD: Group 0, Index 1, Command "MODULE", Data ""
17:07:11 MQT: stat/Vario_Chambre/RESULT = {"Module":{"54":"Tuya MCU"}}
17:07:11 SRC: Backlog
17:07:11 CMD: Group 0, Index 1, Command "GPIO", Data "255"
17:07:11 MQT: stat/Vario_Chambre/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"107":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"108":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"52":"Led1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
17:07:16 TYA: Send "55aa00000000ff"

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
17:08:47 CMD: Status 0
17:08:47 SRC: WebConsole from 172.18.200.100
17:08:47 CMD: Group 0, Index 1, Command "STATUS", Data "0"
17:08:47 MQT: stat/Vario_Chambre/STATUS = {"Status":{"Module":54,"FriendlyName":["ChambreVario"],"Topic":"Vario_Chambre","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:08:47 MQT: stat/Vario_Chambre/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:19:29","StartupUTC":"2020-03-24T15:49:18","Sleep":50,"CfgHolder":4617,"BootCount":65,"BCResetTime":"2020-03-24T14:28:55","SaveCount":896,"SaveAddress":"F4000"}}
17:08:47 MQT: stat/Vario_Chambre/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-20T14:45:23","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"380/699"}}
17:08:47 MQT: stat/Vario_Chambre/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["NASA_SAT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00000200","00000000"]}}
17:08:47 MQT: stat/Vario_Chambre/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":23,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
17:08:47 MQT: stat/Vario_Chambre/STATUS5 = {"StatusNET":{"Hostname":"Vario_Chambre","IPAddress":"172.18.200.46","Gateway":"172.18.200.1","Subnetmask":"255.255.255.0","DNSServer":"172.18.200.1","Mac":"EC:FA:BC:68:DC:C1","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:08:47 MQT: stat/Vario_Chambre/STATUS6 = {"StatusMQT":{"MqttHost":"172.18.200.40","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_68DCC1","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:08:47 MQT: stat/Vario_Chambre/STATUS7 = {"StatusTIM":{"UTC":"2020-03-24T16:08:47","Local":"2020-03-24T17:08:47","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:43","Sunset":"19:09"}}
17:08:47 MQT: stat/Vario_Chambre/STATUS10 = {"StatusSNS":{"Time":"2020-03-24T17:08:47"}}
17:08:47 MQT: stat/Vario_Chambre/STATUS11 = {"StatusSTS":{"Time":"2020-03-24T17:08:47","Uptime":"0T00:19:29","UptimeSec":1169,"Heap":23,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"NASA_SAT","BSSId":"B0:EA:BC:A6:E3:C0","Channel":9,"RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Every installation give the same bug

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

I am missing a configuration or maybe it is a bug ?

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jontron123, 2020-03-27:
Try setting "LedTable" to "0" and see if that fixes your issue.

ascillato2, 2020-04-02:
Please, when you have time, tell us if your issue is solved. Thanks.

closed 2020-04-02 by ascillato2

bradsjm, 2020-04-12:
I had the same issue and the LEDTABLE was on, by turning it off as above it has resolved the problem, the dimmer command now sends the expected value