arendst/Tasmota #7998: DeepSleepTime and TelePeriod not working

KrisRux, 2020-03-24:
### PROBLEM DESCRIPTION
i have set DeepSleepTime 1200 and TelePeriod 300 but the ESP go to sleep after 15 seconds of firts mqtt command

`DeepSleepTime 1200
TelePeriod 300

tele/SDS011/LWT	Offline
stat/SDS011/RESULT	{"DeepSleep":{"Time":"2020-03-24T18:00:00","Epoch":1585069200}}
tele/SDS011/SENSOR	{"Time":"2020-03-24T17:39:13","AM2301":{"Temperature":19.9,"Humidity":73.5},"SDS0X1":{"PM2.5":0.0,"PM10":0.0},"CCS811":{"eCO2":400,"TVOC":0},"TempUnit":"C"}
tele/SDS011/STATE	{"Time":"2020-03-24T17:39:13","Uptime":"0T00:00:15","UptimeSec":15,"Heap":24,"SleepMode":"Dynamic","Sleep":200,"LoadAvg":56,"MqttCount":1,"Wifi":{"AP":1,"SSId":"TNCAPC48E2B","BSSId":"20:B0:01:92:F5:CB","Channel":1,"RSSI":100,"Signal":-42,"LinkCount":1,"Downtime":"0T00:00:09"}}
tele/SDS011/INFO3	{"RestartReason":"Deep-Sleep Wake"}
tele/SDS011/INFO2	{"WebServerMode":"Admin","Hostname":"SDS011-6873","IPAddress":"192.168.1.102"}
tele/SDS011/INFO1	{"Module":"Generic","Version":"8.1.0.11(6cc4f22-sensors)","FallbackTopic":"cmnd/DVES_58DAD9_fb/","GroupTopic":"cmnd/tasmotas/"}
cmnd/SDS011/POWER	
tele/SDS011/LWT	Online`

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [X ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ Node MCU ] Device used (e.g., Sonoff Basic): _____
- [8.1.0.11 ] Tasmota binary firmware version number used: _____
  - [X ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
18:04:51 CMD: STATUS 0
18:04:51 MQT: stat/SDS011/STATUS = {"Status":{"Module":18,"FriendlyName":["SDS011"],"Topic":"SDS011","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:04:51 MQT: stat/SDS011/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota-sensors.bin","RestartReason":"Power On","Uptime":"0T00:24:27","StartupUTC":"2020-03-24T16:40:24","Sleep":200,"CfgHolder":4617,"BootCount":53,"BCResetTime":"2020-03-16T10:52:42","SaveCount":88,"SaveAddress":"F9000"}}
18:04:51 MQT: stat/SDS011/STATUS2 = {"StatusFWR":{"Version":"8.1.0.11(6cc4f22-sensors)","BuildDateTime":"2020-03-15T16:02:07","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"356/699"}}
18:04:51 MQT: stat/SDS011/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TNCAPC48E2B",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000000060000005A00000000000000","00008000","00000000"]}}
18:04:51 MQT: stat/SDS011/STATUS4 = {"StatusMEM":{"ProgramSize":601,"Free":400,"Heap":20,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FDAE387","0435A3A4","B7FFBFCD","01DA9BC4","24B67CC7","00000006"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,19,20,21,22,24,26,29,32,34,37","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64"}}
18:04:51 MQT: stat/SDS011/STATUS5 = {"StatusNET":{"Hostname":"SDS011-6873","IPAddress":"192.168.1.102","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"EC:FA:BC:58:DA:D9","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
18:04:51 MQT: stat/SDS011/STATUS6 = {"StatusMQT":{"MqttHost":"m20.cloudmqtt.com","MqttPort":12409,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_58DAD9","MqttUser":"Temp_Ext","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
18:04:51 MQT: stat/SDS011/STATUS7 = {"StatusTIM":{"UTC":"2020-03-24T17:04:51","Local":"2020-03-24T18:04:51","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:43","Sunset":"19:09"}}
18:04:51 MQT: stat/SDS011/STATUS10 = {"StatusSNS":{"Time":"2020-03-24T18:04:51","AM2301":{"Temperature":21.3,"Humidity":69.9},"SDS0X1":{"PM2.5":16.1,"PM10":61.8},"CCS811":{"eCO2":415,"TVOC":2},"TempUnit":"C"}}
18:04:51 MQT: stat/SDS011/STATUS11 = {"StatusSTS":{"Time":"2020-03-24T18:04:51","Uptime":"0T00:24:27","UptimeSec":1467,"Heap":18,"SleepMode":"Dynamic","Sleep":200,"LoadAvg":11,"MqttCount":1,"Wifi":{"AP":1,"SSId":"TNCAPC48E2B","BSSId":"20:B0:01:92:F5:CB","Channel":1,"RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:09"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
set DeepSleepTime 1200 and TelePeriod 300 and see the log in a console

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2020-03-24:
@KrisRux 
From CHANGELOG.md : `

> `### 8.0.0.1 20191221
> - Add Deepsleep start delay based on Teleperiod if ``Teleperiod`` differs from 10 or 300

So please test with Teleperiod different from 10 or 300.
Apparently this change means that with 10 or 300 the ESP will go on sleep immediately after the 1st teleperiod message that follows the MQTT connection

barbudor, 2020-03-24:
@KrisRux I confirm that it is working as expected with `teleperiod 299`
So the doc needs to be updated and you can a submit a doc PR by yourself.
Please close the issue.

barbudor, 2020-03-24:
@KrisRux I confirm it is working fine with `teleperiod 299`and {deepsleeptime 600`

```teleperriod 299
21:38:23 CMD: teleperriod 299
21:38:23 MQT: stat/sns-terrasse/RESULT = {"Command":"Unknown"}
deepsleeptime 600
21:38:32 CMD: deepsleeptime 600
21:38:32 MQT: stat/sns-terrasse/RESULT = {"DeepSleepTime":600}
21:38:32 DSL: Reset wrong settings wakeup: 0, slip 10592
21:38:32 MQT: stat/sns-terrasse/RESULT = {"DeepSleep":{"Time":"2020-03-24T21:40:00","Epoch":1585082400}}
21:38:35 APP: Sleeping

00:00:00 Project tasmota sns-terrasse Version 8.1.0.7(9814468-sensors)-2_6_1
00:00:00 WIF: Connecting to AP1 BNETWLAN in mode 11N as sns-terrasse...
00:00:05 HTP: Web server active on sns-terrasse with IP address 192.168.168.120
21:40:06 MQT: Attempting connection...
21:40:06 MQT: Connected
21:40:06 MQT: tele/sns-terrasse/LWT = Online (retained)
21:40:06 MQT: cmnd/sns-terrasse/POWER = 
21:40:06 MQT: tele/sns-terrasse/INFO1 = {"Module":"sensor","Version":"8.1.0.7(9814468-sensors)","FallbackTopic":"cmnd/sns-terrasse_fb/","GroupTopic":"cmnd/tasmotas/"}
21:40:06 MQT: tele/sns-terrasse/INFO2 = {"WebServerMode":"Admin","Hostname":"sns-terrasse","IPAddress":"192.168.168.120"}
21:40:06 MQT: tele/sns-terrasse/INFO3 = {"RestartReason":"Deep-Sleep Wake"}
21:40:10 MQT: tele/sns-terrasse/STATE = {"Time":"2020-03-24T21:40:10","Uptime":"0T00:00:12","UptimeSec":12,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"BNETWLAN","BSSId":"50:D4:F7:C0:2E:B0","Channel":1,"RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:06"}}
21:40:10 MQT: tele/sns-terrasse/SENSOR = {"Time":"2020-03-24T21:40:10","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Range":4132}}
21:45:09 MQT: tele/sns-terrasse/STATE = {"Time":"2020-03-24T21:45:09","Uptime":"0T00:05:11","UptimeSec":311,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"BNETWLAN","BSSId":"50:D4:F7:C0:2E:B0","Channel":1,"RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:06"}}
21:45:09 MQT: tele/sns-terrasse/SENSOR = {"Time":"2020-03-24T21:45:09","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Range":4127}}
21:45:09 MQT: stat/sns-terrasse/RESULT = {"DeepSleep":{"Time":"2020-03-24T21:50:00","Epoch":1585083000}}
21:45:12 APP: Sleeping

00:00:00 Project tasmota sns-terrasse Version 8.1.0.7(9814468-sensors)-2_6_1
00:00:00 WIF: Connecting to AP1 BNETWLAN in mode 11N as sns-terrasse...
00:00:05 HTP: Web server active on sns-terrasse with IP address 192.168.168.120
21:50:01 MQT: Attempting connection...
21:50:01 MQT: Connected
21:50:01 MQT: tele/sns-terrasse/LWT = Online (retained)
21:50:01 MQT: cmnd/sns-terrasse/POWER = 
21:50:01 MQT: tele/sns-terrasse/INFO1 = {"Module":"sensor","Version":"8.1.0.7(9814468-sensors)","FallbackTopic":"cmnd/sns-terrasse_fb/","GroupTopic":"cmnd/tasmotas/"}
21:50:01 MQT: tele/sns-terrasse/INFO2 = {"WebServerMode":"Admin","Hostname":"sns-terrasse","IPAddress":"192.168.168.120"}
21:50:01 MQT: tele/sns-terrasse/INFO3 = {"RestartReason":"Deep-Sleep Wake"}
21:50:05 MQT: tele/sns-terrasse/STATE = {"Time":"2020-03-24T21:50:05","Uptime":"0T00:00:12","UptimeSec":12,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"BNETWLAN","BSSId":"50:D4:F7:C0:2E:B0","Channel":1,"RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:06"}}
21:50:05 MQT: tele/sns-terrasse/SENSOR = {"Time":"2020-03-24T21:50:05","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Range":4132}}
21:55:04 MQT: tele/sns-terrasse/STATE = {"Time":"2020-03-24T21:55:04","Uptime":"0T00:05:11","UptimeSec":311,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"BNETWLAN","BSSId":"50:D4:F7:C0:2E:B0","Channel":1,"RSSI":100,"Signal":-49,"LinkCount":1,"Downtime":"0T00:00:06"}}
21:55:04 MQT: tele/sns-terrasse/SENSOR = {"Time":"2020-03-24T21:55:04","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Range":4123}}
21:55:04 MQT: stat/sns-terrasse/RESULT = {"DeepSleep":{"Time":"2020-03-24T22:00:00","Epoch":1585083600}}
21:55:07 APP: Sleeping```

Jason2866, 2020-03-24:
As designed see #7298

effelle, 2020-03-25:
Closing this issue as it has been answered.

closed 2020-03-25 by effelle