arendst/Tasmota #7999: ESP restarts often when Script Option is activated

Icefox2k, 2020-03-24:
### PROBLEM DESCRIPTION
I have compiled my own .Bin with support for display & script. When i activate the script checkbox the esp starts restarting every minute. no matter if there is script input or not

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [x] Tasmota binary firmware version number used: 8.1.0.11(display)
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: PlatformIO (MS VS)
- [x] Flashing tools used: Tasmota PyFlasher
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
00:09:26 MQT: stat/tasmota/RESULT = {"NAME":"Wemos D1","GPIO":[255,255,255,255,6,5,255,255,9,4,10,255,21],"FLAG":15,"BASE":18}
00:09:26 MQT: stat/tasmota/RESULT = {"Module":{"0":"Wemos D1"}}
00:09:26 MQT: stat/tasmota/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"6":"I2C SDA"},"GPIO5":{"5":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"9":"Switch1"},"GPIO13":{"4":"DS18x20"},"GPIO14":{"10":"Switch2"},"GPIO15":{"0":"None"},"GPIO16":{"21":"Relay1"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
00:12:23 MQT: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["building_heater_circulation","Tasmota2"],"Topic":"tasmota","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:12:23 MQT: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota-minimal.bin","RestartReason":"Exception","Uptime":"0T00:18:50","StartupUTC":"2020-03-24T22:53:33","Sleep":50,"CfgHolder":4617,"BootCount":20,"BCResetTime":"2020-03-24T22:53:55","SaveCount":51,"SaveAddress":"F9000"}}
00:12:23 MQT: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.1.0.11(display)","BuildDateTime":"2020.03.24 22:55:18","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"406/699"}}
00:12:23 MQT: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SmartHome WiFi",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
00:12:23 MQT: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":516,"Free":484,"Heap":25,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000407","86A0A18F","040C9FA1","000000C1","010000C0","00004080","00000000"],"Drivers":"1,2,4,8,10,13,16","Sensors":"1,2,5,6"}}
00:12:23 MQT: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-5710","IPAddress":"10.0.20.53","Gateway":"10.0.20.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.20.1","Mac":"EC:FA:BC:C5:96:4E","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:12:23 MQT: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.17","MqttPort":1884,"MqttClientMask":"building_heater_circulation","MqttClient":"building_heater_circulation","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
00:12:23 MQT: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"2020-03-24T23:12:23","Local":"2020-03-25T00:12:23","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00"}}
00:12:23 MQT: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-03-25T00:12:23","Switch1":"ON","Switch2":"ON","DS18B20-1":{"Id":"011912E9811A","Temperature":19.7},"DS18B20-2":{"Id":"011912ED8B67","Temperature":19.6},"DS18B20-3":{"Id":"0119136E5B70","Temperature":19.6},"TempUnit":"C"}}
00:12:23 MQT: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-03-25T00:12:23","Uptime":"0T00:18:50","UptimeSec":1130,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"ON","Wifi":{"AP":1,"SSId":"SmartHome WiFi","BSSId":"44:D9:E7:FA:9B:23","Channel":6,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:06"}}}
00:12:23 MQT: stat/tasmota/STATUS12 = {"StatusSTK":{"Exception":28,"Reason":"Exception","EPC":["4023dab4","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4021d3c8","4023b20a","40222fa9","4023b300","40202f14","4023b239","40238bf0","4023b2f0","4020d248","4021fd61","4022402d","4020aa58","40209ef5","4022305b","40232cd0","40101af5"]}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Activate Script Checkbox 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-03-25:
Can not reproduce. 
You have non standard settings for flash size, You are not using latest dev or release version.
Which display? How are the display and the sensor/relais powered?
Wemos Mini clones do have a Vreg which does NOT deliver enough current on the 3.3V path to add anything.
For finding the issue disconnect anything you do not need to test. Use a different ESP board.
NodeMCU has a decent Vreg (1A).

Icefox2k, 2020-03-26:
I didn't make any changes for the flash size, but undefined Features that i dont want to use. Do u thing ia have to make more changes to compile for my Wemos D1?
I have also have compiled a Bin File from the newest release version - but same Result.

Display ist a LCD 2x16 with a I2C Adapter. All Hardware ist powered external from a Power Suply with 5V and up to 3A. Only the DS18x20 are powered by the Wemos because the 3.3V. I also had testet it with all Hardware powerd direct by USB - and had no problems Display/Sensor/Relay all worked fine, so i think we dont have a power problem - but when activating Script the problem comes up.
Yesterday I tested my Wemos with everything disconnected but same Result, finally take another Wemos flash the files - same Result. 
What is the reason for not connecting my Relais to gpio 16? I changed to gpio 2 - same Result

Thanks for your answer - keep trying

Jason2866, 2020-03-27:
Talking about the 3.3V power supply of the ESP8266 on the PCB. 
The 3.3V is generated from the extern 5V. This is done with a 3.3V Vreg on the wemos board.
This vreg is nearly on all clones to weak for reliable powering the ESP chip.
Some wemos mini clones work others do not. Lottery game. 
To rule this out, using a known to be good device (for example NodeMCU) is first step

Jason2866, 2020-03-27:
Try this build. Scripter, LCD 16x2 and DS18x20 enabled. Most other sensors disabled.

[tasmota.zip](https://github.com/arendst/Tasmota/files/4392884/tasmota.zip)

ascillato2, 2020-04-02:
Please, when you have time, tell us if your issue is solved. Thanks.

closed 2020-04-02 by ascillato2

Icefox2k, 2020-04-04:
Sorry that I have not reported earlier
yesterday i tried the bin-file from jason with the same result. when activating the checkbox for scripting the ESP starts with restarting ... and stops when unchecking.

 i can't imagine that i am the only one with this effect. but when we dont find the reason for this i will go with the rules feature for this project and take all intelligence in an iobroker script   ... not my first choice but also a possible solution