arendst/Tasmota #8001: xsns_53_sml.ino there are no negative values on some 2 way electricity meters

honikos, 2020-03-25:
**Have you looked for this feature in other issues and in the docs?**  
yes, found no solution but hints for the problem

**Is your feature request related to a problem? Please describe.**  
My electricity meter (EMH ED300 L) gives always positive values for D_TPWRCURR "Current-In/Out" but "Current Out" should give negative values

**Describe the solution you'd like**  
In dependence of a hex value, which I maybe have to identify, there could be a negative value.
In my case I could identify the hex value which shows if the power is coming in or goes out.
Would be nice to be able to make values negative if the meter does measure outgoing power.
Is it possible to have a descriptor  / variable to use for this issue?
Details see: https://forum.creationx.de/forum/index.php?thread/1095-d0-z%C3%A4hler-sml-auslesen-mit-tasmota/&postID=31606#post31606

**Describe alternatives you've considered**  
Don't see an alternative because I do not have enough programming knowledge.

**Additional context**  
This feature is necessary because graphs show false pictures an maybe I can do more logical controlling if I get the correct value vor In / Out Power  

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-03-25:
You posted in creationx forum where Gemu is active. He is the author of this driver.
Please ask there too for help.
@gemu2015  Can you take a look into?

kugelkopf123, 2020-03-25:
> You posted in creationx forum where Gemu is active. He is the author of this driver.
> Please ask there too for help.
> @gemu2015 Can you take a look into?

I had already tried to answer this question in the forum. However, the code would actually have to be adapted. 
If I understood it correctly, it is about the fact that the consumption value has a different sign depending on whether energy is being consumed or fed in. So it is not about two different values where one value stands for the consumption and the other for the feed-in but a value that covers both. 

@honikos I hope I have understood this correctly.

honikos, 2020-03-26:
> If I understood it correctly, it is about the fact that the consumption value has a different sign depending on whether energy is being consumed or fed in. 

At the moment there is no distinction between consumption or feed-in. Per definition the value shows only the energy in W. But it makes sence to have a negative value when the meter measures feed-in (s. picture). 

> So it is not about two different values where one value stands for the consumption and the other for the feed-in but a value that covers both.

As mentioned above it is one value which could give more information as only track the actual energy. It ist possible to show the direction of energy flow ( - / negative if feed-in).

@kugelkopf123  don't know if understood .... hope it helps.
[image]

gemu2015, 2020-03-27:
according to the data sheet this meter does NOT support negative values on current power!
This is not a fault of the sml decoder which of cause works with negative values if supported by the meter (e.g 2 way meter EHZ363)

matzefisi, 2020-03-27:
Hi guys, 
just found this issue and I am facing the same problem. The direction information is stored in the status part of the 108 and 208 messages. So if there is an easy way to parse the "A2" or "82" we can do the calculation on our own in the script.

77070100010800FF	640101	**A2**	01	621E 52FF 56			0001450FA101  --> Bezug
77070100010800FF	640101	**82**	01	621E 52FF 56			000145A40201  --> Einspeisung

Best regards
Matthias

gemu2015, 2020-03-27:
thanks Matthias for the hint (energy direction in status)

ok silly meter why not returning a signed current power value?

However i did put the status into a script variable  

enabled by
#define ED300L

var=sml(mn 2)  mn=meter number

returns the value in question (a2 or 82) as decimal
 
its on my fork, if tested i make a pr

files affected scripter and sml

https://github.com/gemu2015/Sonoff-Tasmota

matzefisi, 2020-03-27:
Hi gemu,
great. Thanks! I will give a try and report back.
Matthias

matzefisi, 2020-03-27:
Hi again, 
used the lunchbreak to reflash the ESP and added your config. Hope I understood the structure correct:

Config
`>D
>B
=>sensor53 r
>M 2
+1,13,s,0,9600,SML
+2,4,c,0,50,Haus
1,77070100010800ff@1000,Verbrauch,KWh,Total_in,2
1,77070100020800ff@1000,Einspeisung,KWh,Total_out,2
1,770701000F0700FF@1,Aktuell,W,current,0
1,77070100000009ff@#,Meter Nr,,Meter_number,0
2,=h==================
2,77070100010800ff@1000,Total Verbrauch,KWh,Total_in,4
2,77070100010800ff@1000,Total Einspeisung,KWh,Total_out,4
>S
print %sml("797422-5001138" 2)%
#`

But as result I get only a 0.00 in the console. I also tried meter number 1, but the result stays the same. Do I have an error in my script? 
`12:35:41 MQT: tasmota/smartmeter/tele/SENSOR = {"Time":"2020-03-27T12:35:41","SML":{"Total_in":18480.81,"Total_out":32976.35,"current":4573,"Meter_number":"797422-5001138"},"Haus":{"Total_in":0.0000,"Total_out":0.0000}}
12:35:41 0.00`

honikos, 2020-03-27:
Looks like "Haus" is not your SML meter. Should be the first one.
So maybe false meter number ?

matzefisi, 2020-03-27:
Nope, Haus can be ignored for now. I am waiting for a replacement of my second meter. I want to have it running for meter 1 (SML). That´s the EDL300.

So I also tried `print %sml(1 2)%`. But this also does not work.

gemu2015, 2020-03-27:
ok is this status only transmitted on 180 and 280 and not on the others?
then i have to decode this also, will update this afternoon.

matzefisi, 2020-03-27:
Correct. Only 180 and 280.
Thanks!

gemu2015, 2020-03-27:
next try !

matzefisi, 2020-03-27:
Unfortunately I cannot compile it. And with my poor C knowledge I cannot spot the error.
```
Sonoff-Tasmota-universal8\tasmota\xsns_53_sml.ino: In function 'uint32_t SML_Status(uint32_t)':
xsns_53_sml:2082:10: error: 'sml_status' was not declared in this scope
   return sml_status[meter];
```

gemu2015, 2020-03-27:
did you type
#define ED300L
in your user config overwrite ?

matzefisi, 2020-03-27:
Oh man, I mixed it up with the EDL300. Now it worked. Sorry. 
I get now a 162 in decimal in the console. So it seems to work. I will have another look in the afternoon when the sun goes away. Then I would expect a 130 in decimal (82 hex). Thank you very much!
Now I have to figure out how I can bring this value into the webinterface and calculate the value for MQTT, but this is well documented. I will post my solution here.

gemu2015, 2020-03-28:
new version available
the new version directly inverts current power (770701000F0700FF ) on solar feed, so you should get negative values
so this stupid meter should be fixed!

honikos, 2020-03-28:
@gemu2015 
For me it doesn't work out of the box.  
part of my user_config_override.h:

#ifndef ED300L
#define ED300L
#endif
#ifndef USE_SCRIPT
#define USE_SCRIPT  # adds about 17k flash size, variable ram size
#endif
#ifndef USE_SML_M
#define USE_SML_M
#endif
#ifdef USE_RULES
#undef USE_RULES
#endif
#define SML_MAX_VARS 5

@matzefisi 
Do I need all this stuff if I  use #define ED300L  ?
As @gemu2015  wrote, I would expect a negative output for Line
1,770701000F0700FF@1,Aktuell,W,current,0

but I get a high value  .... (think its without decimal ...)

kugelkopf123, 2020-03-28:
@honikos Had you used the version from @gemu2015  fork? 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

honikos, 2020-03-28:
@kugelkopf123    Yes I did.

gemu2015, 2020-03-29:
try new version (bug removed)

honikos, 2020-03-29:
ok, did it. Value looks good. Now waiting for the sun. Will give feedback when I have feed-in.

honikos, 2020-03-29:
perfect. it works :-)
should be merged into the base branch. hope that there are other meters with same behavior. can maybe be solved by script.

@gemu2015  Thank you!

[image]

closed 2020-03-29 by ascillato2

juergen778899, 2022-10-18:
@gemu2015  I am struggling with compiling a build for an ED300L power meter. I am using Tasmota 12.2.0 and are able to compile and upload a Tasmota light to my device. I read and tried quite some things but fail to enable support for this power meter and its strange coding of the power flow direction (it is used as "Zweirichtungszähler") in my case so flow direction matters.

I am more or less always running into error "undefined reference to `sml_hex_asci(unsigned int, char*)'" when compiling and #define USE_SML_M is in my user_config_override.h When I only use #DEFINE ED300L it does not seem to include script, at least I do not see the button in the UI under Console. 

So I am doing apprently something wrong here - any hint? How does the user_config need to look like for a Tasmota just for reading this power meter?

And do I then still need this script? Or does the Include of your ED300L code resolve the issue in the background? I am not sure reading the above...

`>D
sml1state=0
currentwatt=0
>B
=>sensor53 r

>T
currentwatt=SML#current
if sml(1 2)==162
then
sml1state=-1
else 
sml1state=1
endif
currentwatt=currentwatt*sml1state

>J
,"current:%currentwatt%

>W
==================
SML Aktuell +/- </th><td>%currentwatt% W</td>

>M 2
+1,13,s,0,9600,SML
+2,4,c,0,50,Haus

1,77070100010800ff@1000,Verbrauch,KWh,Total_in,2
1,77070100020800ff@1000,Einspeisung,KWh,Total_out,2
1,770701000F0700FF@1,Aktuell,W,current,0
1,77070100000009ff@#,Meter Nr,,Meter_number,0
2,=h==================
2,77070100010800ff@1000,Total Verbrauch,KWh,Total_in,4
2,77070100010800ff@1000,Total Einspeisung,KWh,Total_out,4
#`

Thanks a lot & great work done in making power meter readout possible - it works for all other power meters and (used to) work(s) also for the ED300L - only with the missing flow direction. :-(

bejoup, 2023-10-29:
hey there. I have the same problem with a smartmeter EMH EHZ-HW which is not sending negtive values when i produce Solar energy. I red a lot about the problem, but i didn't found a solution for that.. Even in this post, which is the same problem. Can you help me with the right script?
here are some examples:
Positive Values
00:09:16.288 : 77 07 01 00 0f 07 00 ff 01 01 62 1b 52 ff 55 00 00 0a 09 01 
00:09:16.288 : 77 07 01 00 0f 07 00 ff 01 01 62 1b 52 ff 55 00 00 0a 09 01
Negative Values - producing Energy
20:06:15.752 : 77 07 01 00 0f 07 00 ff 01 01 62 1b 52 ff 55 00 00 0e 5c 01
19:34:47.255 : 77 07 01 00 0f 07 00 ff 01 01 62 1b 52 ff 55 00 00 08 17 01 
19:34:42.917 : 77 07 01 00 0f 07 00 ff 01 01 62 1b 52 ff 55 00 00 08 c6 01 
19:34:25.521 : 77 07 01 00 0f 07 00 ff 01 01 62 1b 52 ff 55 00 00 09 d1 01

honikos, 2023-10-30:
hi @bejoup 
you need this alternative firmware.
https://github.com/gemu2015/Sonoff-Tasmota

bejoup, 2023-10-30:
> hi @bejoup you need this alternative firmware. https://github.com/gemu2015/Sonoff-Tasmota
Thank you @honikos !
Is it possible to flash on a hichi-wifi device? And afterwards the Skripts above should work?

gemu2015, 2023-10-31:
@bejoup 
you do not need my fork!  everything according SML driver is merged into tasmota.
read the docs, there are hints for this problem (https://tasmota.github.io/docs/Smart-Meter-Interface/#special-commands)
the hitchi driver should already be up to date, but just in case you may update hitchi firmware without problems
see also here for more hints: https://ottelo.jimdofree.com/stromzähler-auslesen/

honikos, 2023-10-31:
@gemu2015 , fantastic. I wasn't aware that SML drivers since version 12.x are included in tasmota. Thank you for clarification.

bejoup, 2023-10-31:
Thank you for the links and useful information.
It is Tasmota 12.3.1 preinstalled on the hichi. I red all the sides already, but it is not solving my problem.
The ED300L Problem is quite simillar, but my smart meter EHZ-HW does not transfer the right OBIS -Code 77070100020800ff. The smart meter only tranfer following:
OBIS (hex) | OBIS | Name | Wert | Einheit | Parsed
-- | -- | -- | -- | -- | --
0x0100000009ff | 0.0.9 | Geräteeinzelidentifikation |
0x0100010800ff | 1.8.0 | Zählerstand Total | 463101326 | Wh | 46310132.6Wh (Zählerstand Total)
0x0100010801ff | 1.8.1 | Zählerstand Tarif 1 | 463101326 | Wh | 46310132.6Wh (Zählerstand Tarif 1)
0x0100010802ff | 1.8.2 | Zählerstand Tarif 2 | 0 | Wh | 0.0Wh (Zählerstand Tarif 2)
0x01000f0700ff | 15.7.0 | Unbekannter Datentyp | 2569 | W | 256.9W (Unbekannter Datentyp)

The send or receive tag must be somewhere else. But I can‘t find it in the console output yet.
I will try to find the right tag an post it here. 
@gemu2015 i am not sure if i am on the right way… as far as i understand, the Bit is changing in Hex Code and it is not readable in the OBIS

gemu2015, 2023-10-31:
https://demo.volkszaehler.org/pipermail/volkszaehler-dev/2017-April/005591.html

this must be coded in special option 1, see docs again

try this , probably edit flag and or bit position

1,=so1,00010800,63,11,63,11,000f0700

bejoup, 2023-10-31:
I will try it. But its hard to guess the right flag.
The documentation says:
M,=so1 | special SML option for meters that use a bit in the status register to sign import or export like ED300L, AS2020 or DTZ541 e.g. 1,=so1,00010800,65,11,65,11,00100700 for DTZ5411. obis code that holds the direction bit, 2. Flag identifier, 3. direction bit, 4. second Flag identifier (some meters use 2 different flags), 5. second bit, 6 obis code of value to be inverted on direction bit.
-- | --

77                                
   07 01 00 01 08 00 FF 
   63 01 82  
Minus
77                                
   07 01 00 01 08 00 FF 
   63 01 A2  
Plus
I will try it when the sun comes back ;) Thank you Gemu

gemu2015, 2023-11-02:
11 is the bit number

in your meter 2 bits change  82 -> b0

63 indicates a 16 bit value (so bit 0 .. 15), so you may try bit 1 or bit 5
1,=so1,00010800,63,1,63,1,000f0700

bejoup, 2023-11-05:
Yesterday the sun was back an i tried to get a negative values, but it didnˋt work. This is my Script. Is there something wrong?
>D
>B
=>sensor53 r
>M 1
+1,3,s,1,9600,
1,77070100010800ff@1000,Zaehlerstand_Total,KWh,Total_in,2
1,770701000f0700ff@1,Verbrauch_Aktuell,W,Power_curr,2 
1,=so1,00010800,63,5,63,5,000f0700
#

@gemu2015 i tried from 1 to 6, but Not one value changed to negative. Can I somehow show, which digit is used? Like a lineout of something?

gemu2015, 2023-11-06:
i did put your raw values from above into an emulator and it works perfectly with your script from the last post, so 1,=so1,00010800,63,5,63,5,000f0700 is fine.

i hope your hitchi driver is up to date, otherwise he has updated drivers on request.

bejoup, 2023-11-06:
Wow, thank you Gemu!
How can i Check my hitchi driver Version?
I only found some more Information about the Versions in the Web gui. 
Program Version | 12.3.1(tasmota)
Builddate 2022-12-17T18:16:02
Core/SDK Version 2_7_4_9/2.2.2-dev(38a443e)
ESP Chip Id | 1451788 (ESP8266EX)

I will try it again, when the sun is back.

bejoup, 2023-11-08:
Update: no negative values yet.
I wrote to hichi, and he is not sure, but it could be that the special commands only works with tasmota 13.0.0
I updatet my hichi now. I will have a look when the sun is back.
@gemu2015 : I found a recent post about the special command-problem, where you also helped.
[https://github.com/arendst/Tasmota/discussions/17807](url)

Hopefully it works now. Would be so nice. Greetings!

bejoup, 2023-11-13:
It works! Today the first negative values. Tasmota 13.0.0 and the special command.
@gemu2015 : thanks a lot for your support. Do you know how I can add this solution for my powermeter EMH EHZ-HW to the  Tasmota Wiki? 
My script for negative Values in Tasmota 13.0.0:
>D
>B

=>sensor53 r
>M 1
+1,3,s,16,9600,
1,77070100010800ff@1000,Zaehlerstand_Total,KWh,Total_in,2
1,770701000f0700ff@1,Verbrauch_Aktuell,W,Power_curr,2 
1,=so1,00010800,63,5,63,5,000f0700
#

Have FUN!

Raphael-Vogel, 2025-02-16:
@bejoup  Thanks for providing the script. Also works on my `eHZ-HW`