arendst/Tasmota #8002: Tasmota 8.2.0.1 BME280 and DS18x20 doesn't work together.

majherek, 2020-03-25:
### PROBLEM DESCRIPTION
Using self-compiled firmware FIRMWARE_SENSORS with MQTT TLS support BME280 sensor disappear when another GPIO is set to 4 DS18x20 or 21 Relay 1. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [x] Tasmota binary firmware version number used: 8.2.0.1/9888a77
  - [x] Pre-compiled
  - [] Self-compiled
    - [] IDE / Compiler used: platformio
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
09:33:06 CMD: Backlog Template; Module; GPIO 255
09:33:07 RSL: stat/sonoff8/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
09:33:07 RSL: stat/sonoff8/RESULT = {"Module":{"18":"Generic"}}
09:33:07 RSL: stat/sonoff8/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"4":"DS18x20"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"21":"Relay1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"6":"I2C SDA"},"GPIO13":{"0":"None"},"GPIO14":{"5":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
- [x] Provide the output of this command: `Status 0`:
```
09:34:07 RSL: stat/sonoff8/STATUS = {"Status":{"Module":18,"FriendlyName":["Tasmota"],"Topic":"sonoff8","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:34:07 RSL: stat/sonoff8/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:21","StartupUTC":"2020-03-25T08:32:46","Sleep":50,"CfgHolder":4617,"BootCount":17,"BCResetTime":"2020-03-24T19:40:05","SaveCount":33,"SaveAddress":"F9000"}}
09:34:07 RSL: stat/sonoff8/STATUS2 = {"StatusFWR":{"Version":"8.2.0.1(9888a77-sensors)","BuildDateTime":"2020-03-24T15:01:28","Boot":2,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"312/699"}}
09:34:07 RSL: stat/sonoff8/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["atomix",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
09:34:07 RSL: stat/sonoff8/STATUS4 = {"StatusMEM":{"ProgramSize":601,"Free":400,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FDAA387","0415A3A4","B7FFBFCD","01DA9BC4","24367CC7","00000002"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64"}}
09:34:07 RSL: stat/sonoff8/STATUS5 = {"StatusNET":{"Hostname":"sonoff8","IPAddress":"192.168.20.8","Gateway":"192.168.20.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.199.251","Mac":"84:F3:EB:B3:92:29","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
09:34:07 RSL: stat/sonoff8/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B39229","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
09:34:07 RSL: stat/sonoff8/STATUS7 = {"StatusTIM":{"UTC":"2020-03-25T08:34:07","Local":"2020-03-25T09:34:07","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:41","Sunset":"19:10"}}
09:34:07 RSL: stat/sonoff8/STATUS10 = {"StatusSNS":{"Time":"2020-03-25T09:34:07","BME280":{"Temperature":6.6,"Humidity":41.0,"DewPoint":-5.7,"Pressure":1028.9},"PressureUnit":"hPa","TempUnit":"C"}}
09:34:07 RSL: stat/sonoff8/STATUS11 = {"StatusSTS":{"Time":"2020-03-25T09:34:07","Uptime":"0T00:01:21","UptimeSec":81,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"atomix","BSSId":"BC:26:C7:E1:66:61","Channel":11,"RSSI":100,"Signal":-34,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
00:00:00 CFG: Loaded from flash at F8, Count 34
00:00:00 QPC: Flag 0E
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota Version 8.2.0.1(9888a77-sensors)-STAGE
00:00:00 DSB: Sensors found 0
00:00:00 SHT: Sensor did not ACK command
00:00:00 I2C: BME280 found at 0x76
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 atomix in mode 11N as sonoff8...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on sonoff8 with IP address 192.168.20.8
09:34:41 NTP: UTC 2020-03-25T08:34:40, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
09:34:41 RSL: tele/sonoff8/INFO1 = {"Module":"Generic","Version":"8.2.0.1(9888a77-sensors)","FallbackTopic":"cmnd/DVES_B39229_fb/","GroupTopic":"cmnd/tasmotas/"}
09:34:41 RSL: tele/sonoff8/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff8","IPAddress":"192.168.20.8"}
09:34:41 RSL: tele/sonoff8/INFO3 = {"RestartReason":"Software/System restart"}
09:34:41 RSL: stat/sonoff8/RESULT = {"POWER":"OFF"}
09:34:41 RSL: stat/sonoff8/POWER = OFF
09:34:41 QPC: Reset
09:34:43 APP: Boot Count 18
09:34:43 CFG: Saved to flash at F7, Count 35, Bytes 4096
09:34:45 RSL: tele/sonoff8/STATE = {"Time":"2020-03-25T09:34:45","Uptime":"0T00:00:12","UptimeSec":12,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"atomix","BSSId":"BC:26:C7:E1:66:61","Channel":11,"RSSI":100,"Signal":-33,"LinkCount":1,"Downtime":"0T00:00:06"}}
[... 44 lines cut ...]
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on sonoff8 with IP address 192.168.20.8
00:00:05 HTP: Main Menu
19:50:54 NTP: UTC 2020-03-24T18:50:53, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
19:50:54 RSL: tele/sonoff8/INFO1 = {"Module":"Generic","Version":"8.2.0(sensors)","FallbackTopic":"cmnd/DVES_B39229_fb/","GroupTopic":"cmnd/tasmotas/"}
19:50:54 RSL: tele/sonoff8/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff8","IPAddress":"192.168.20.8"}
19:50:54 RSL: tele/sonoff8/INFO3 = {"RestartReason":"Software/System restart"}
19:50:54 RSL: stat/sonoff8/RESULT = {"POWER":"OFF"}
19:50:54 RSL: stat/sonoff8/POWER = OFF
19:50:54 QPC: Reset
19:50:54 HTP: Console
19:50:56 APP: Boot Count 16
19:50:56 CFG: Saved to flash at F7, Count 29, Bytes 4096
19:50:58 RSL: tele/sonoff8/STATE = {"Time":"2020-03-24T19:50:58","Uptime":"0T00:00:12","UptimeSec":12,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"atomix","BSSId":"BC:26:C7:E1:66:61","Channel":11,"RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:06"}}
19:50:58 RSL: tele/sonoff8/SENSOR = {"Time":"2020-03-24T19:50:58","DS18B20":{"Id":"051701E346FF","Temperature":17.4},"BME280":{"Temperature":-3.2,"Humidity":100.0,"DewPoint":null,"Pressure":1028.4},"PressureUnit":"hPa","TempUnit":"C"}
```

A have another problem with 8.2.0.1 and those sensors, but one issue at a time... ;-)

Jason2866, 2020-03-25:
If you self compile a good approach is using version **Tasmota** and add in user_config_override.h
only the drivers and devices you need. So you dont have anything in which can generate side effects and you have more free resources.

pepoweb, 2020-03-25:
In a downloaded version of the 8.2.0.1 firmware on a SonOff basic I loose connectivity to the DS18x20. After downgrade to 7.2 it works again.

Jason2866, 2020-03-25:
@pepoweb  different story. DS18x20 is NOT included anymore in tasmota.bin with v.8.2.x
You have to use Tasmota-sensors

pepoweb, 2020-03-25:
Ah, missed that. Thanks!

majherek, 2020-03-25:
@Jason2866 I download firmware from http://thehackbox.org/tasmota/ and http://thehackbox.org/tasmota/release/

From release it works, from latest development  - it doesn't. I didn't compile it yet ;-) Whan I would compile it I will turn on MQTT TLS ;-)  That's another story - after I use self-compiled firmware, when I turn on DS18x20 or RELAY, the reading's from BME disappear.

majherek, 2020-03-25:
@Jason2866  BTW: I changed my configuration of Tasmota and I use now only: user_config_override.h. Now I have clean git status:
```
majherek@SOKAR:~/win/Nextcloud/Dokumenty/Arduino/Tasmota$ git status
HEAD detached at v8.2.0
Untracked files:
  (use "git add <file>..." to include in what will be committed)

        MOJE/
        tasmota/my_user_config.h.orig
        tasmota/my_user_config_test.h

nothing added to commit but untracked files present (use "git add" to track)
majherek@SOKAR:~/win/Nextcloud/Dokumenty/Arduino/Tasmota$
```
I even turn _-DUSE_CONFIG_OVERRIDE_ in _[common]_ section in _platformio_override.ini_.

Jason2866, 2020-03-25:
wemos mini clone? Looks like a power issue.
Please join to Discord. More responsive.

Jason2866, 2020-03-25:
Problem solved with fewer drivers enabled in self compiled firmware.
NO Tasmota bug.

effelle, 2020-03-25:
Closing this issue as it has been answered.

closed 2020-03-25 by effelle

majherek, 2020-03-25:
No tasmota bug, pure voltage regulator on wemos clone. Thanks to @Jason2866 and others from Discord Support.