arendst/Tasmota #8007: Sp-202 power monitoring does not work any more

mali385, 2020-03-26:
### PROBLEM DESCRIPTION
My digoo sp-202 was working perfect and power monitoring also working perfect but after reset itself by the power cycle reset. Power monitoring not working at all. I am using same template before I used but power monitoring sstill not working. What can I do ?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic):  Generic
- [ ] Tasmota binary firmware version number used:  8.2.0
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: Tuya-Convert
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
RSL: stat/tasmota/RESULT = {"NAME":"Generic_SP202","GPIO":[52,0,0,131,91,134,0,0,22,17,132,21,0],"FLAG":0,"BASE":63}
stat/tasmota/RESULT = {"Module":{"0":"Generic_SP202"}}
stat/tasmota/RESULT = {"GPIO0":{"52":"Led1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"131":"HLWBL SELi"},"GPIO4":{"91":"Button2n"},"GPIO5":{"134":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"22":"Relay2"},"GPIO13":{"17":"Button1"},"GPIO14":{"132":"HLWBL CF1"},"GPIO15":{"21":"Relay1"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
12:50:05 RSL: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:50:05 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:11:53","StartupUTC":"2020-03-26T11:38:12","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2020-03-26T12:41:24","SaveCount":42,"SaveAddress":"F8000"}}
12:50:05 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-20T14:45:23","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"309/699"}}
12:50:05 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ASUSS",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
12:50:05 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
12:50:05 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-6786","IPAddress":"192.168.1.109","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"80:7D:3A:7C:9A:82","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
12:50:05 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7C9A82","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
12:50:05 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"2020-03-26T11:50:05","Local":"2020-03-26T12:50:05","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:39","Sunset":"19:12"}}
12:50:05 RSL: stat/tasmota/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
12:50:05 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-03-26T12:50:05","ENERGY":{"TotalStartTime":"2020-03-26T00:24:14","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
12:50:05 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-03-26T12:50:05","Uptime":"0T00:11:53","UptimeSec":713,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":25,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"ASUSS","BSSId":"2C:FD:A1:0D:57:40","Channel":10,"RSSI":100,"Signal":-49,"LinkCount":2,"Downtime":"0T00:00:07"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-03-26:
Try module 63 instead of user template

mali385, 2020-03-26:
It does not work for my device, when I choose module 63, it is keeping on of and after a while it reset itself. However this template was working well till it reset itself.

arendst, 2020-03-26:
In that case use your template again BUT disable seriallogging first with ``seriallog 0``

mali385, 2020-03-26:
Yes, I did.

arendst, 2020-03-26:
> Yes, I did.

Does it work now?

mali385, 2020-03-26:
No it is same.

arendst, 2020-03-26:
It might be broken. After all it did reset itself for some reason.

mali385, 2020-03-26:
Is it possible? Because everything work well except power monitoring.

arendst, 2020-03-26:
Power monitoring uses dedicated hardware. If that fails then you won't see any values popping up.

mali385, 2020-03-26:
Okay, I cannot reach the serial port of devices maybe I'll crack it and glue again. I mean if I flash firmware by using serial port, is there any chance to work. If there is a chance I can try.

arendst, 2020-03-26:
You cannot reach serialport because one of the power monitoring signals is using the serial port (See GPIO03). I don't think serial flashing will solve your problem. The code runs fine just the power monitoring hardware is not detected anymore. You could try OTA re-flash again.

Re check your template to make sure you have the correct one for your hardware.

mali385, 2020-03-26:
Okay thank you for helping, I think there is nothing to do because I checked the template several times.

closed 2020-03-27 by arendst