arendst/Tasmota #8010: pcf8574 conflicts with MCP230xx

MichiM2, 2020-03-27:
### PROBLEM DESCRIPTION
The driver xdrv_28_pcf8574.ino scans address ranges for detecting pcf8574 and pcf8574a chips. It tries to skip the address USE_MCP230xx_ADDR if it is in the Range.

Because of a bug, it nevertheless considers a detected MCP0xx-Chip erroneously as a pcf8574.

### REQUESTED INFORMATION

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Sonoff Basic
- [X] Tasmota binary firmware version number used: 8.1.0.10
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: VSC
- [ ] Flashing tools used: _____
- [ X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"5":"I2C SCL"},"GPIO2":{"0":"None"},"GPIO3":{"6":"I2C SDA"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"51":"IRrecv"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
/STATE = {"Time":"2020-03-27T12:40:46","Uptime":"0T00:30:12","UptimeSec":1812,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF","POWER9":"OFF","Wifi":{"AP":1,"SSId":"matnet","BSSId":"24:65:11:D5:B2:80","Channel":1,"RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:06"}}
/SENSOR = {"Time":"2020-03-27T12:40:46","MCP230XX":{"D0":0,"D1":0,"D2":1,"D3":0,"D4":0,"D5":0,"D6":0,"D7":0,"D8":1,"D9":1,"D10":1,"D11":1,"D12":0,"D13":0,"D14":0,"D15":0}}
/SENSOR = {"Time":"2020-03-27T12:40:46","MCP230_OUT":{"OUT_D8":"ON","OUT_D9":"ON","OUT_D10":"ON","OUT_D11":"ON","END":1}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
```
- [ ] Provide the output of this command: `Status 0`:
```
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
```

### TO REPRODUCE
Connect a MCP23017 at Address 0x20 and a PCF8574 at an address 0x21 to 0x26. The Chip at address 0x20 is detected as PCF8574 even if 
```
#define USE_MCP230xx_ADDR 0x20
```

### EXPECTED BEHAVIOUR
Chip at USE_MCP230xx_ADDR is detected as MCP230xx

### SCREENSHOTS

### ADDITIONAL CONTEXT
The bug is in the file  xdrv_28_pcf8574.ino within the function void Pcf8574Init(void) :
The Codeblock at 
```
#ifdef USE_MCP230xx_ADDR
    //AddLog_P2(LOG_LEVEL_INFO, PSTR("PCF: USE_MCP230xx_ADDR is defined!!!"));
    if (USE_MCP230xx_ADDR == pcf8574_address) {
      AddLog_P2(LOG_LEVEL_INFO, PSTR("PCF: Addr: 0x%x reserved for MCP320xx, skipping PCF8574 probe"), pcf8574_address);
      pcf8574_address++;
    }
#endif
```
is too late. It has to be placed immediately after 

```
while ((Pcf8574.max_devices < MAX_PCF8574) && (pcf8574_address < PCF8574_ADDR2 +8)) {
```

Jason2866, 2020-03-27:
Take a look in https://tasmota.github.io/docs/#/I2CDevices
for switching off the not needed driver.

MichiM2, 2020-03-27:
Yes, i know how to disable the driver. But this does not solve my problem. I need to use both drivers at the same time, which is possible and as designed.
There is already a solution in the PCF8574-driver, but on the wrong place.

Jason2866, 2020-03-27:
If you have a tested working solution a PR is welcome

arendst, 2020-03-27:
Thx. I'll fix it now.

arendst, 2020-03-27:
Pls verify

MichiM2, 2020-03-27:
Works as expected.
Tnx

closed 2020-03-27 by MichiM2