arendst/Tasmota #8011: CT Power on analog input

selahattinerdal, 2020-03-27:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Nodemcu
- [ ] Tasmota binary firmware version number used:8.2.0
  - [ ] Pre-compiled
  - [x ] Self-compiled
    - [x ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
17:44:10 CMD: PowerSet 60.0
17:44:11 MQT: stat/tasmota_BF5881/RESULT = {"Command":"Unknown"}
17:44:28 CMD: VoltageSet 225.0
17:44:29 MQT: stat/tasmota_BF5881/RESULT = {"Command":"Unknown"}
17:44:56 CMD: CurrentSet
17:44:56 MQT: stat/tasmota_BF5881/RESULT = {"Command":"Unknown"}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

I have Module Type Generic (NodeMCU) pin ADC0 set to CT Power. I want calibrate it, but receive unknow command error by PowerSet, VoltageSet, CurrentSet.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-03-27:
It is not designed to calibrate. The CT connected to the ADC is not a precise method to measure.

meingraham, 2020-03-27:
I do not recall, but there was some dialog about updating ADCParam in order to provide some ROUGH calibration capability for current measurement. You'll have to inspect the commits to see if this happened.

effelle, 2020-03-27:
Yeah, didn't add yet. Sorry.
`ADCParam 7` can control
 - ANALOG_CT_FLAGS (default 0) actually reserved for possible future use;
- ANALOG_CT_MULTIPLIER (default 2146) multiplier*100000 to convert raw ADC peak to peak range 0..1023 to RMS current in Amps. Value of 100000 corresponds to 1
- ANALOG_CT_VOLTAGE (default 2300) to convert current in Amps to apparent power in Watts using voltage in Volts*10. Value of 2200 corresponds to 220V

selahattinerdal, 2020-03-27:
ADCParam 7,0,3290,225, that's is for my SCT-013 30A. Thanks all.

selahattinerdal, 2020-03-27:
ADCParam 7,0,3290,0.225

closed 2020-03-27 by selahattinerdal