arendst/Tasmota #8013: Provide PWM / dimmable "status" LED

garci66, 2020-03-27:
**Have you looked for this feature in other issues and in the docs?**  
YES - didnt find an equivalent

**Is your feature request related to a problem? Please describe.**  
Not an operational problem - jus that the status LED is too bright at night

**Describe the solution you'd like**  
I am using some of these ESP-compatible switches: LX-WIFI-03o (or any of the hundreds of similar models). They provide a blue LED to highlight the keys at night (since there are no physical keys). They also have RED LEDs on the keys driven directly by the relay control. At night, when all lights are off, having the blue LED to find the keys is very useful, but in a dark room, the light can be quite bright. It would be great to provide a way to be able to dim the blue LEDs on command. Something like using 
`LedPower 30`
to set the Led to a 30% PWM ratio and dim the led accordingly. But I would still like to retain the ability to blink the LED when wifi is lost for example or flash with an mqtt message.

**Describe alternatives you've considered**  
My current work-around has been to configure the GPIO for the blue LED (GPIO13) as PWM1i instead of LinkLedi. This lets me control the PWM value for the blue LED and it works fine but I lose the ability for the LED to be used to indicate MQTT messages / loss of WIFI

It would be great to just be able to add the ability to set PWM brightness on the status led while keeping it as a status led instead of a full PWM port. We can keep reporting it as a "dimmer bulb" to homeassitant discovery so that it can be easily automated (for example, trigger higher brightness during the day or if a motion detector detects someone walking) - similar to what Nest does with its screen.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-03-27:
That's an interesting idea.

@arendst In theory it would just mean replacing `digitalWrite()` with `analogWrite()` and add a new setting. But I don't know the LedState code, so maybe I'm missing something. What's also nice is that `analogWrite()` at full power is equivalent to `digitalWrite(HIGH)`

arendst, 2020-03-27:
But there will be a lot of processor power (due to analogwrite) being used for PWM on a status LED. I think it's a waste.

s-hadinger, 2020-03-27:
True, but not if you set to full power. So it needs to be an educated choice.

arendst, 2020-03-27:
A user and an educated choice is a disaster meant to happen.

I will think about it. For now ledstate 30 won't work because currently there are a lot of bit mask parameters with ledstate. Needs a redesign.

garci66, 2020-03-27:
Is there such a difference between analog write and digital write? After all, PWM in the ESP is "semi-software", isnt it? so after all, we're just adjusting the PWM percentage and the timer value. Or you mean that calculating the timer value would be expensive? We could even save the "raw" (0-1024 PWM ) value between "blink" calls.. 

And we don't blink the led TOO often, no? 

I've was worried about the fact that PWM was interrupt driven and not truly done in hardware but I haven't noticed much latency on my switches when running the led at 50%.

s-hadinger, 2020-03-28:
PWM is entirely software in ESP8266. It means setting a timer at 880Hz and using interrupts to switch the output on and off with correct timing. The moment your put a PWM value different from 0 or 1023, it generates those interrupts, and consumes CPU.

In your case it wouldn't change anything since you already switched PWM on.

Jason2866, 2020-03-28:
Mhh, couldnt the wanted behaviour not be accomplished using rules?

s-hadinger, 2020-03-28:
@Jason2866 I don't think so, not for the LedLink. Or maybe catching Wifi#Disconnected/Wifi#Connected but I'm not sure how to make a PWM1 blink with rules.

I drafted a patch, it's a small code change. But if we go this route, the `LedDimmer` should apply to all leds or only to LedLink/Led1? Also keep in mind it will not be seen as a Dimmer by HA.

garci66, 2020-03-29:
im guessing internal rules? It  would still be good to have it be a dimmer in HA.. (so that's "noob-proof" automatable).

I agree that the moment I'm using PWM different than 0 / 1024 I'll get interrupts. That's the issue with soft-pwm instead of HW-PWM but the CPU seems quite fast and not really affected by it.

After all, anyone using tasmota for a LED driver would have 4 PWMs going on at the same time, no?

Thanks!

Jason2866, 2020-03-30:
Either way it is done (rules or Tasmota code change). You will NOT see as dimmer in HA.
Since the goal can be reached via Tasmota rules a code change will probably not done.
Thats what rules are made for.

s-hadinger, 2020-03-31:
What about the following: simulate the LedLink with PWM1 (or PWM1i):

`BackLog BlinkCount 0; PowerOnState 0; Dimmer 20; Power1 0`

`Rule1 ON Power1#Boot DO Power1 3 ENDON ON Wifi#Disconnected DO Power1 3 ENDON ON Wifi#Connected DO Power1 0 ENDON`

`Backlog Rule1 1; Restart 1`

If the setting is ok, you can do it in a 1-liner:

`BackLog BlinkCount 0; PowerOnState 0; Dimmer 20; Power1 0; Rule1 ON Power1#Boot DO Power1 3 ENDON ON Wifi#Disconnected DO Power1 3 ENDON ON Wifi#Connected DO Power1 0 ENDON; Rule1 1; Restart 1`

garci66, 2020-03-31:
Thanks! will give it a try. The idea of the dimmer being visible in HA is to be able to automate the brighthness change. (for example, based on a schedule or based on some other sensor, like a motion sensor). The dimmer command will set something (im guessing the first dimmer?) configured as dimmer to 20 won't it?  So this dimmer would be visible in HA. 

Sorry if im confusing things up.

I just though it would be a handy feature. Will test out using the rules but jus thought it might be more user friendly to have it "natively"

ascillato2, 2020-04-06:
Closing issue as this issue can be solved with the use of rules.

closed 2020-04-06 by ascillato2

garci66, 2020-08-10:
Just for future reference. This has now been implemented as a full feature: https://github.com/arendst/Tasmota/pull/8491