arendst/Tasmota #8015: Add a unix like command history to the Tasmota web console

scottchiefbaker, 2020-03-28:
**Have you looked for this feature in other issues and in the docs?**  

Yes. The only reference I found was issue #7483. I believe we can work around the limitations presented in that issue.

**Is your feature request related to a problem? Please describe.**  

No

**Describe the solution you'd like**  

I'm used to hitting the up arrow to scroll through the command history of my Linux terminal. The Tasmota web console doesn't support this, and it can be problematic. If I'm running `status 8` over and over to check a switch it would be very helpful to be able to just hit the up arrow and enter to run the last command.

**Describe alternatives you've considered**  

Copy and paste the same command every time... Limited in that it only stores the last command you ran, and may get overwritten if you use your system for other things.

**Additional context**  

I have written the requisite JavaScript to enable this feature on the web console. You can test this out without modifying your Tasmota installation by navigating to the Tasmota web console and copy and pasting the script below in to your browser's web console.

https://www.perturb.org/code/tasmota_command_history.js

To enable this on Tasmota would simply require including the above Javascript in the `<script>` tag on the web console page. 

Uncompressed/unminified the code is 1077 bytes. Running the code through [JSCompress](https://jscompress.com/) yields a final size of 471 bytes. I suspect I could save more bytes too if I shortened some variable names, but I have not done that yet. If this code is something you're interested in I can work on shrinking the code even further.

**(Please, remember to close the issue when the problem has been addressed)**

nagyrobi, 2020-03-28:
+1 for this

scottchiefbaker, 2020-03-28:
Just for my own education I wanted to see how small I could get the code. After renaming some variables and functions I'm able to get the code down to 310 bytes.

s-hadinger, 2020-03-28:
I believe I could use that too. Can you submit a Pull Request with the 324 bytes version?

effelle, 2020-03-28:
And this has sparked interest in me... Cool one.

scottchiefbaker, 2020-03-28:
I'm not sure how/where in the Tasmota code the Javascript would be inserted. If you can provide that starting point I can investigate.

It may just be easier for someone already familiar with the code base to add in this code.

scottchiefbaker, 2020-03-28:
I didn't see you already had a `querySelector()` alias... using that I can get the code down to 284 bytes.

effelle, 2020-03-28:
A smaller code has more probability to be included on firmware, now focus on it and reduce it to no more than 99 bytes. 😄

scottchiefbaker, 2020-03-28:
Did a little more work and got it down to 266 bytes... anything beyond that and the code won't be readable and hard to maintain. 

Before I go too much farther I'd like people to test what I have so far to see if it does what people want. Please copy/paste this [minified javascript](https://www.perturb.org/tmp/tasmota_cmd_hist.min.js) in to your browser's web console (hit F12). Then enter some commands and hit up/down to navigate the history.

effelle, 2020-03-28:
seems it remembers just the last command entered and if I press again the arrows Chrome comes in with his history cache.

scottchiefbaker, 2020-03-28:
I've tested on Firefox and it correctly remembers the last 20 commands. Anyone else able to test?

to-scho, 2020-03-29:
works fine in Safari on a Mac

arendst, 2020-03-29:
Will test too today.

arendst, 2020-03-29:
Implemented.

It has to fight with the browsers command cache but once you get used to it it works fine at least in Chrome and New Edge

closed 2020-03-29 by ascillato2

scottchiefbaker, 2020-03-29:
@arendst you can add `autocomplete="off"` to the `<input>` tag and it won't compete with the browser like you mention.

arendst, 2020-03-30:
@scottchiefbaker you don't happen to have some small bulletproof code to differentiate between tablets/phones (autocomplete="on") and PC's (autocomplete="off") do you?

scottchiefbaker, 2020-03-30:
@arendst easiest thing to do would be to check the screen with via JS. If screen width is less than 700px it's a mobile device.

I would think we'd be safe just turning autocomplete off for all platforms. I wouldn't use autocomplete on mobile, but maybe that's just me.

arendst, 2020-03-30:
It might also work if I leave the default (autocomplete on) and as soon as the up arrow or down arrow is detected set the property to off (which, in my eyes can only happen on a keyboard)

Testing...

scottchiefbaker, 2020-03-30:
@arendst that seems like a decent solution, and relatively easy to implement.