arendst/Tasmota #8017: Events with 2 # not supported in scripts

JumpmanJunior, 2020-03-28:
### PROBLEM DESCRIPTION
I am using scripts. In my T section I am assigning sensor values to variables which I then use for logging to SD card. This is working for example without problems with a connected AM2301 sensor.

```
>T
temp=AM2301#Temperature
```

But I also want to use Zigbee sensors with Zigbee2Tasmota. For a door contact sensor I have for example the following assignment:

```
>T
door=Zbreceived#0x5A85#Power
```

But this is not working. No value is assigned. I assume that this is because of the two #. These events were introduced in Tasmota lately according to the change log: rule triggers on one level deeper using syntax with two #. But this fixed it only for rules and not for scripts.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Lolin Board
- [x] Tasmota binary firmware version number used: 8.2.0.2
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: PlatformIO
- [x] Flashing tools used: PlatformIO
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
23:20:32 RSL: stat/tasmota/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
23:20:33 RSL: stat/tasmota/RESULT = {"Module":{"18":"Generic"}}
23:20:33 RSL: stat/tasmota/RESULT = {"GPIO0":{"2":"AM2301"},"GPIO1":{"166":"Zigbee Rx"},"GPIO2":{"0":"None"},"GPIO3":{"165":"Zigbee Tx"},"GPIO4":{"5":"I2C SCL"},"GPIO5":{"6":"I2C SDA"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
23:20:52 RSL: stat/tasmota/STATUS = {"Status":{"Module":18,"FriendlyName":["Kellersensor"],"Topic":"tasmota","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:20:52 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T00:00:42","StartupUTC":"2020-03-28T22:20:10","Sleep":50,"CfgHolder":4617,"BootCount":53,"BCResetTime":"2020-03-12T20:22:57","SaveCount":106,"SaveAddress":"F9000"}}
23:20:52 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.2.0.2(tasmota)","BuildDateTime":"2020.03.28 23:11:05","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"353/699"}}
23:20:52 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["LNW",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
23:20:52 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":596,"Free":404,"Heap":20,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashMode":3,"Features":["00000407","87A82187","04480121","000000CD","010017C0","C0008800","00000004"],"Drivers":"1,2,3,4,8,9,10,13,23,29,30,35,37","Sensors":"1,2,3,5,6,33"}}
23:20:52 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"Kellersensor","IPAddress":"192.168.4.119","Gateway":"192.168.4.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.4.1","Mac":"BC:DD:C2:01:8E:5B","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
23:20:52 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_018E5B","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
23:20:52 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"2020-03-28T22:20:52","Local":"2020-03-28T23:20:52","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:35","Sunset":"19:15"}}
23:20:52 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-03-28T23:20:52","AM2301":{"Temperature":21.8,"Humidity":54.4,"DewPoint":12.2},"TempUnit":"C"}}
23:20:52 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-03-28T23:20:52","Uptime":"0T00:00:42","UptimeSec":42,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"LNW","BSSId":"C8:0E:14:6B:30:54","Channel":6,"RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Try to assign a variable in the T section:
```
>T
temp=AM2301#Temperature
lq=Zbreceived#0x5A85#LinkQuality 
```

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Assignment for temp is working, but for lq is not. It should also work for lq.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2020-03-29:
did you try the >E section ?

zigbee does not report to >T

JumpmanJunior, 2020-03-29:
According to my understanding of the documentation the E section is not the correct one for this. But I tried it and moved my assignment to the E section - without positive results.

gemu2015, 2020-03-31:
two # elements ARE supported in scripter

>T is called via FUNC_JSON_APPEND which is called by most sensor drivers but NOT used by the zigbee driver.

zigbee calls XdrvRulesProcess which calls FUNC_RULES_PROCESS in scripter
which invokes section >E

since i do not have zigbee i can not track down

JumpmanJunior, 2020-04-01:
In my last test for the E section I have written ZbReceived with a small R. That's why this was not working. It is working in the E section as it should. I have created a pull request to clarify this topic in the docs: https://github.com/tasmota/docs/pull/234

closed 2020-04-01 by JumpmanJunior

martin072, 2020-12-19:
@JumpmanJunior 

I also have the same issue, with Tasmota 9.2.0 the following variable is not changed when the event happens, any idea why? I believe the spelling is correct (zigbee device is a Sonoff motion sensor)

```
>E
bm=ZbReceived#0x9821#ZoneStatusChange
```