arendst/Tasmota #8018: Additional triggers for buttons in rules

JumpmanJunior, 2020-03-28:
**Have you looked for this feature in other issues and in the docs?**  
Yes.

**Is your feature request related to a problem? Please describe.**  
In rules there are only triggers for the following button state changes: ON, OFF, TOGGLE, HOLD. There is no trigger for double press. My goal is to react to single press, double press and hold of a button which is not possible  at the moment. By default double press will trigger the relay instead. It is possible to influence this behavior with e.g. SetOption11, but it is not possible to react to all 3 events.

**Describe the solution you'd like**  
Introduce a separate trigger for double press in rules. Additionally perhaps a further SwitchMode would be possible which would support all three events (single, double, hold).

**Describe alternatives you've considered**  
The desired functionality can be achieved with scripts, but it would make the usage much easier if rules would support further triggers.

ascillato, 2020-03-29:
>There is no trigger for double press.

Sorry, Yes there is. Please check the docs at rules cookbook. There is an example using single, double and hold.

ascillato2, 2020-03-29:
Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

JumpmanJunior, 2020-03-29:
It seems my description was not clear enough. I am aware of this examples in the cookbook. As I already mentioned it is possible to modify the default behavior with SetOption11, but then you can only use the double press and not the single press in the rules.
My goal is to use all 3 events in rules and not turning the relay with any of them. As you can see in  the examples in the cookbook this is always the case at the moment.

to-scho, 2020-04-02:
If you are willing to use a workaround and if you have some dummy GPIOs left you can have that feature by configuring a dummy shutter using dummy relays 1+2 (relays in modules config):
shutterrelay1 1

Afterwards you can set assign button 1+2 (buttons in modules config) to that shutter (a single button will also work without below simultaneous button press detection):
ShutterButton1  1 - - - 0
ShutterButton1  2 - - - 0

Now there are triggers to rules:
Shutter1#Button1=1: single press of button1
Shutter1#Button1=2: double press of button1
Shutter1#Button1=3: tripple press of button1
...
Shutter1#Button1=0: hold of button1
...
Shutter1#Button2=1: single press of button2
...
Shutter1#Button0=1: simultaneous single press of button1 and button2
...
Shutter1#Button0=0: simultaneous hold of button1 and button2

Ugly, but works for me. Maybe it's with to rework the support_button.ino

effelle, 2020-04-21:
Closing this issue, new triggers for buttons were merged.

closed 2020-04-21 by effelle