arendst/Tasmota #8019: Scripter: Script no longer in the edit window after saving.

kugelkopf123, 2020-03-29:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
If a FW is compiled with the Script option and a script is saved, it is saved and apparently also executed, but it can no longer be edited. Instead, the next time the script window is called up, the input area remains empty and the script is at the top of the gray input window.
See the screenshot.
You can use a new script in the editable area, which obviously also works and is saved. However, it is shown in the gray area exactly as before as soon as you open the script window again.

Unfortunately I cannot say since when it has been so. Did nothing in the script area for about 2 months.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Wemos D1 Mini 
- [x ] Tasmota binary firmware version number used: latest Dev Branch
  - [ ] Pre-compiled
  - [x ] Self-compiled
    - [x ] IDE / Compiler used: PlatformIO on VS
- [ ] Flashing tools used: _____
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
04:10:17 RSL: RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
04:10:17 RSL: RESULT = {"Module":{"18":"Generic"}}
04:10:18 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
-

```
- [ x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
04:10:56 RSL: STATUS = {"Status":{"Module":18,"FriendlyName":["ESP-SWG2"],"Topic":"ESP-SWG2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
04:10:56 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:57","StartupUTC":"2020-03-29T02:09:59","Sleep":50,"CfgHolder":4617,"BootCount":2,"BCResetTime":"2020-03-29T04:02:38","SaveCount":9,"SaveAddress":"FB000"}}
04:10:56 RSL: STATUS2 = {"StatusFWR":{"Version":"8.2.0.2(tasmota)","BuildDateTime":"2020.03.29 04:01:58","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"337/699"}}
04:10:56 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Eiswuerstchen",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C8000100060000005A00000000000000","00000008","00000000"]}}
04:10:56 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":28,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16400B","FlashMode":3,"Features":["00000407","86002196","040803A1","00000004","000000C0","40000000","00000000"],"Drivers":"1,2,3,4,10","Sensors":"2,3"}}
04:10:56 RSL: STATUS5 = {"StatusNET":{"Hostname":"ESP-SWG2-5649","IPAddress":"192.168.1.66","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"84:0D:8E:AD:D6:11","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
04:10:56 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-03-29T02:10:56","Local":"2020-03-29T04:10:56","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99}}
04:10:56 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-03-29T04:10:56"}}
04:10:56 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-03-29T04:10:56","Uptime":"0T00:00:57","UptimeSec":57,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Eiswuerstchen","BSSId":"E0:28:6D:22:3F:BC","Channel":1,"RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
-

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Compile with:

```
#ifndef USE_SCRIPT
#define USE_SCRIPT  
#endif
#ifdef USE_RULES
#undef USE_RULES
#endif
```
Add some Code in the Script section and safe this. Than open the Script Section again.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
As It was like a few months ago that you can easily edit the script that you entered earlier.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
It has nothing to do with #define USE_SML_M
Tried it without. Just with the Script enable while compiling.

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2020-03-31:
i can not reproduce this behavior 
must be a special case of setup

kugelkopf123, 2020-03-31:
Ok, i will go on further research.

kugelkopf123, 2020-03-31:
Strange, I can't reproduce it anymore. Maybe it had something to do with a certain build. With the current version in the Dev.Branch there is no problem anymore. 
Thanks a lot for checking!

closed 2020-03-31 by kugelkopf123