arendst/Tasmota #8022: Different topics in Home Assistant

AWSb24, 2020-03-29:
**Have you looked for this feature in other issues and in the docs?**
Yes, I did  

**Is your feature request related to a problem? Please describe.**  
I use Home Assistant with MQTT to switch and monitor multiple devices in my house,
After restarting the Home Assistant it doesn't know the states of all devices until the state change is triggered. That's why I am currently using "tele/PlatsaRelais/STATE" to get a response from the device every 5 minutes. I don't want to decrease the interval to avoid flooding my network.

**Describe the solution you'd like**  
Is it possible to add an option to send the topic "tele/PlatsaRelais/STATE" at the same time as "stat/tasmota/POWER ON" or replace each other?

**Describe alternatives you've considered**  
Please suggest alternative solutions.

**Additional context**  

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-03-29:
Please, read the home assistant article in Tasmota docs. This has been addressed and solved there. You need an automation in HA.

effelle, 2020-03-29:
https://tasmota.github.io/docs/Home-Assistant/#useful-automations 
Sync Power State.

closed 2020-03-29 by effelle