arendst/Tasmota #8027: shelly 2.5 exception 9 when enabling i2c

kovaga, 2020-03-30:
### PROBLEM DESCRIPTION
Shelly 2.5 restart reason "Exception: 9" when I2C is set.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Shelly 2.5
- [ ] Tasmota binary firmware version number used:8.2.0-sensors
  - [ x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ x] Flashing tools used: WebGui
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
{"NAME":"Shelly 2.5","GPIO":[56,0,17,0,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
00:00:00 CFG: Loaded from flash at F4, Count 1032
00:00:00 APP: Some settings have been reset (6)
00:00:00 Project tasmota Terrace Light Version 8.2.0(sensors)-STAGE
00:00:00 WIF: Connecting to AP1 KVGSMH in mode 11N as shelly25_01-0099...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on shelly25_01-0099 with IP address 192.168.9.172
08:38:59 MQT: Attempting connection...
08:38:59 MQT: Connected
08:38:59 MQT: tele/shelly25_01/LWT = Online (retained)
08:38:59 MQT: cmnd/shelly25_01/POWER = 
08:38:59 MQT: tele/shelly25_01/INFO1 = {"Module":"Sonoff Basic","Version":"8.2.0(sensors)","FallbackTopic":"cmnd/DVES_692063_fb/","GroupTopic":"cmnd/tasmotas/"}
08:38:59 MQT: tele/shelly25_01/INFO2 = {"WebServerMode":"Admin","Hostname":"shelly25_01-0099","IPAddress":"192.168.9.172"}
08:38:59 MQT: tele/shelly25_01/INFO3 = {"RestartReason":{"Exception":9,"Reason":"Exception","EPC":["40105ec0","00000000","00000000"],"EXCVADDR":"00000003","DEPC":"00000000","CallChain":["4025394c","401060f1","4023ab88","402468f8","4021ae51","4021afad","40246588","402464bf","4024b788","4021c189","4023b2b3","40222a8d","40105fad","40248ffc","40222c2b","40222c5f","4020f730","4020f77c","40105e6d","40248362","4026ea68","4024900e","4026ea75","4026eaba","402458e4","40000f49","40000f49","40000e19","4010580d","40105813","4010000d"]}}
08:38:59 MQT: stat/shelly25_01/RESULT = {"POWER":"OFF"}
08:38:59 MQT: stat/shelly25_01/POWER = OFF
08:39:03 MQT: tele/shelly25_01/STATE = {"Time":"2020-03-30T08:39:03","Uptime":"0T00:00:12","UptimeSec":12,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"KVGSMH","BSSId":"0E:EC:DA:14:13:6E","Channel":1,"RSSI":44,"Signal":-78,"LinkCount":1,"Downtime":"0T00:00:06"}}
08:44:03 MQT: tele/shelly25_01/STATE = {"Time":"2020-03-30T08:44:03","Uptime":"0T00:05:12","UptimeSec":312,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"KVGSMH","BSSId":"0E:EC:DA:14:13:6E","Channel":1,"RSSI":46,"Signal":-77,"LinkCount":1,"Downtime":"0T00:00:06"}}
08:49:03 MQT: tele/shelly25_01/STATE = {"Time":"2020-03-30T08:49:03","Uptime":"0T00:10:12","UptimeSec":612,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"KVGSMH","BSSId":"0E:EC:DA:14:13:6E","Channel":1,"RSSI":46,"Signal":-77,"LinkCount":1,"Downtime":"0T00:00:06"}}
08:50:32 CMD: status 0
08:50:32 MQT: stat/shelly25_01/STATUS = {"Status":{"Module":1,"FriendlyName":["Terrace Light"],"Topic":"shelly25_01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:50:32 MQT: stat/shelly25_01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://github.com/arendst/Tasmota/releases/download/v8.1.0/tasmota-sensors.bin","RestartReason":"Exception","Uptime":"0T00:11:41","StartupUTC":"2020-03-30T07:38:51","Sleep":50,"CfgHolder":4617,"BootCount":129,"BCResetTime":"2020-03-30T08:04:37","SaveCount":1034,"SaveAddress":"FA000"}}
08:50:32 MQT: stat/shelly25_01/STATUS2 = {"StatusFWR":{"Version":"8.2.0(sensors)","BuildDateTime":"2020-03-20T14:48:25","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"414/699"}}
08:50:32 MQT: stat/shelly25_01/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":3,"LogHost":"192.168.6.15","LogPort":514,"SSId":["KVGSMH",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005AFF000000000000","00000000","00000000"]}}
08:50:32 MQT: stat/shelly25_01/STATUS4 = {"StatusMEM":{"ProgramSize":597,"Free":404,"Heap":19,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","8FDAA387","0415A3A4","B7FFBFCD","01DA9BC4","24367CC7","00000002"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64"}}
08:50:32 MQT: stat/shelly25_01/STATUS5 = {"StatusNET":{"Hostname":"shelly25_01-0099","IPAddress":"192.168.9.172","Gateway":"192.168.9.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.9.1","Mac":"2C:F4:32:69:20:63","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
08:50:32 MQT: stat/shelly25_01/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.9.12","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_692063","MqttUser":"mqttuser","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
08:50:32 MQT: stat/shelly25_01/STATUS7 = {"StatusTIM":{"UTC":"2020-03-30T07:50:32","Local":"2020-03-30T08:50:32","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:30","Sunset":"19:18"}}
08:50:32 MQT: stat/shelly25_01/STATUS10 = {"StatusSNS":{"Time":"2020-03-30T08:50:32"}}
08:50:32 MQT: stat/shelly25_01/STATUS11 = {"StatusSTS":{"Time":"2020-03-30T08:50:32","Uptime":"0T00:11:41","UptimeSec":701,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"KVGSMH","BSSId":"0E:EC:DA:14:13:6E","Channel":1,"RSSI":46,"Signal":-77,"LinkCount":1,"Downtime":"0T00:00:06"}}}
08:50:32 MQT: stat/shelly25_01/STATUS12 = {"StatusSTK":{"Exception":9,"Reason":"Exception","EPC":["40105ec0","00000000","00000000"],"EXCVADDR":"00000003","DEPC":"00000000","CallChain":["4025394c","401060f1","4023ab88","402468f8","4021ae51","4021afad","40246588","402464bf","4024b788","4021c189","4023b2b3","40222a8d","40105fad","40248ffc","40222c2b","40222c5f","4020f730","4020f77c","40105e6d","40248362","4026ea68","4024900e","4026ea75","4026eaba","402458e4","40000f49","40000f49","40000e19","4010580d","40105813","4010000d"]}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
Mar 30 09:38:59 shelly25_01-0099 ESP-MQT: tele/shelly25_01/INFO3 = {"RestartReason":{"Exception":9,"Reason":"Exception","EPC":["40105ec0","00000000","00000000"],"EXCVADDR":"00000003","DEPC":"00000000","CallChain":["4025394c","401060f1","4023ab88","402468f8","4021ae51","4021afad","40246588","402464bf","4024b788","4021c189","4023b2b3","40222a8d","40105fad","40248ffc","40222c2b","40222c5f","4020f730","4020f77c","40105e6d","40248362","4026ea68","4024900e","4026ea75","4026eaba","402458e4","40000f49","40000f49","40000e19","4010580d","40105813","4010000d"]}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Removing the I2C SDA and I2C SCL allows shelly to boot and operate.
{"NAME":"Shelly 2.5","GPIO":[56,0,17,0,21,83,0,0,0,82,0,22,156],"FLAG":2,"BASE":18}
While seting it on throws an exception
{"NAME":"Shelly 2.5","GPIO":[56,0,17,0,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18}

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Was working previously. (not sure if it is an upgrade to 8.2.0, but cant downgrade as throws an error trying to flash 8.1.0 via webgui)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-03-30:
Since Shelly is a Power Monitoring device.
**Connecting additional, not original, sensor peripherie is dangerous.** 
**Not supported and not recommended.**
Please read the documentation.

kovaga, 2020-03-30:
I haven't added any additional, not original, sensor peripherie.
The I2C bus is used to access internal ADE7953 - "Energy metering is done by a ADE7953 chip connected via I2C and IRQ on GPIO16."
At the moment, when i disable I2c only the internal Temperature is reported.

Jason2866, 2020-03-30:
Ahh, okay. Dont use tasmota-sensors version for Shelly 2.5. Please use tasmota.bin

kovaga, 2020-03-30:
I have tried both tasmota-sensors and tasmota, both with the same result. :(

Jason2866, 2020-03-30:
@arendst do you have a Shelly 2.5? I dont have...

kovaga, 2020-03-30:
@Jason2866 I have another shelly 2.5 running tasmota 6.6. Will upgrade it to 8.1.0 then to 8.2.0 and will see what happens

arendst, 2020-03-30:
Mine runs fine

[image]

arendst, 2020-03-30:
After downgrade with ``otaurl http://thehackbox.org/tasmota/release/tasmota.bin``

[image]

Still functional. I noticed you have a weird non-supported (https) otaurl. Where do your binaries come from?

kovaga, 2020-03-30:
@arendst The weird URL comes from the chrome HTTPS everywhere plugin.
But when I try to downgrade it to 8.1.0 using the correct OTAURL I get "Upgrade":"Failed Wrong HTTP Code"

Mar 30 13:42:56 shelly25_01-0099 ESP-CMD: Group 0, Index 1, Command "OTAURL", Data "http://github.com/arendst/Tasmota/releases/download/v8.1.0/tasmota.bin"
Mar 30 13:42:56 shelly25_01-0099 ESP-CFG: CR 405/699
Mar 30 13:42:56 shelly25_01-0099 ESP-MQT: stat/shelly25_01/RESULT = {"OtaUrl":"http://github.com/arendst/Tasmota/releases/download/v8.1.0/tasmota.bin"}
Mar 30 13:42:56 shelly25_01-0099 ESP-SRC: WebGui from (IP unset)
Mar 30 13:42:56 shelly25_01-0099 ESP-CMD: Group 0, Index 1, Command "UPGRADE", Data "1"
Mar 30 13:42:56 shelly25_01-0099 ESP-MQT: stat/shelly25_01/RESULT = {"Upgrade":"Version 8.2.0 from http://github.com/arendst/Tasmota/releases/download/v8.1.0/tasmota.bin"}
Mar 30 13:42:56 shelly25_01-0099 ESP-CFG: Saved to flash at FB, Count 1055, Bytes 4096
Mar 30 13:42:58 shelly25_01-0099 ESP-HTP: Web server stopped
Mar 30 13:42:58 shelly25_01-0099 ESP-UPL: http://github.com/arendst/Tasmota/releases/download/v8.1.0/tasmota.bin
Mar 30 13:42:59 shelly25_01-0099 ESP-HTP: Web server active on shelly25_01-0099 with IP address 192.168.9.172
Mar 30 13:43:00 shelly25_01-0099 ESP-HTP: Web server stopped
Mar 30 13:43:00 shelly25_01-0099 ESP-UPL: http://github.com/arendst/Tasmota/releases/download/v8.1.0/tasmota.bin
Mar 30 13:43:01 shelly25_01-0099 ESP-HTP: Web server active on shelly25_01-0099 with IP address 192.168.9.172
Mar 30 13:43:02 shelly25_01-0099 ESP-HTP: Web server stopped
Mar 30 13:43:02 shelly25_01-0099 ESP-UPL: http://github.com/arendst/Tasmota/releases/download/v8.1.0/tasmota.bin
Mar 30 13:43:03 shelly25_01-0099 ESP-HTP: Web server active on shelly25_01-0099 with IP address 192.168.9.172
Mar 30 13:43:04 shelly25_01-0099 ESP-HTP: Web server stopped
Mar 30 13:43:04 shelly25_01-0099 ESP-UPL: http://github.com/arendst/Tasmota/releases/download/v8.1.0/tasmota.bin
Mar 30 13:43:05 shelly25_01-0099 ESP-HTP: Web server active on shelly25_01-0099 with IP address 192.168.9.172
Mar 30 13:43:06 shelly25_01-0099 ESP-HTP: Web server stopped
Mar 30 13:43:07 shelly25_01-0099 ESP-HTP: Web server active on shelly25_01-0099 with IP address 192.168.9.172
Mar 30 13:43:08 shelly25_01-0099 ESP-MQT: stat/shelly25_01/UPGRADE = {"Upgrade":"Failed Wrong HTTP Code"}

If I try to manually upload the file via webgui.
Mar 30 13:44:38 shelly25_01-0099 ESP-HTP: Firmware Upgrade
Mar 30 13:44:46 shelly25_01-0099 ESP-CFG: Saved to flash at FB, Count 1056, Bytes 4096
Mar 30 13:44:46 shelly25_01-0099 ESP-UPL: File tasmota.bin ...

Mar 30 13:44:53 shelly25_01-0099 ESP-HTP: Upload done
Mar 30 13:44:54 shelly25_01-0099 ESP-MQT: Attempting connection...
Mar 30 13:44:54 shelly25_01-0099 ESP-MQT: Connected
Mar 30 13:44:54 shelly25_01-0099 ESP-MQT: tele/shelly25_01/LWT = Online (retained)
Mar 30 13:44:54 shelly25_01-0099 ESP-MQT: cmnd/shelly25_01/POWER = 
Mar 30 13:44:54 shelly25_01-0099 ESP-MQT: Subscribe to cmnd/shelly25_01/#
Mar 30 13:44:54 shelly25_01-0099 ESP-MQT: Subscribe to cmnd/tasmotas/#
Mar 30 13:44:54 shelly25_01-0099 ESP-MQT: Subscribe to cmnd/DVES_692063_fb/#

and the following message in the WebGui
Upload Failed

Upload buffer miscompare

arendst, 2020-03-30:
You are using the wrong download address. The one you use can only be used with https which is not supported by Tasmota.

The only correct otaurl is http://thehackbox.org/tasmota/release/tasmota.bin

Any versions other than the current released version you will have to download yourself from the release page and upload to your device.

See docs howto.

kovaga, 2020-03-30:
@arendst I tried manually downloading the 8.1.0 firmware and upload it via webgui, but I get "Upload buffer miscompare" error.

arendst, 2020-03-30:
Again. Read the docs.

As the downloaded binary is too big you'll need to upload a minimal version BEFORE you can upload the final 8.1.0 version.

kovaga, 2020-03-30:
@arendst Thanks for pointing it out. Have succesfully downgraded to 8.1.0 and the same error appears when I set the I2C parameters.

Mar 30 14:40:19 shelly25_01-0099 ESP-MQT: tele/shelly25_01/INFO1 = {"Module":"Sonoff Basic","Version":"8.1.0(tasmota)","FallbackTopic":"cmnd/DVES_692063_fb/","GroupTopic":"cmnd/tasmotas/"}
Mar 30 14:40:19 shelly25_01-0099 ESP-MQT: tele/shelly25_01/INFO2 = {"WebServerMode":"Admin","Hostname":"shelly25_01-0099","IPAddress":"192.168.9.172"}
Mar 30 14:40:19 shelly25_01-0099 ESP-MQT: tele/shelly25_01/INFO3 = {"RestartReason":{"Exception":9,"Reason":"Exception","EPC":["40105f18","00000000","00000000"],"EXCVADDR":"00000003","DEPC":"00000000","CallChain":["40106149","402331d0","402415ec","402169dd","40101448","402411c5","4010196e","40216b3d","402413b4","402412f3","402465b0","40218059","40230707","4021e4c1","40106005","40243ee0","4021e662","4021e697","4020e094","4024b08c","4020e0e0","4027f34c","402430b6","402683a0","40243ef2","402683ad","402683f2","40240743","40000f49","40000f49","40000f49"]}}
Mar 30 14:40:19 shelly25_01-0099 ESP-MQT: stat/shelly25_01/RESULT = {"POWER":"OFF"}
Mar 30 14:40:19 shelly25_01-0099 ESP-MQT: stat/shelly25_01/POWER = OFF
Mar 30 14:40:19 shelly25_01-0099 ESP-NTP: Drift 0, (UTC) Mon Mar 30 12:40:19 2020, (DST) Sun Mar 29 02:00:00 2020, (STD) Sun Oct 25 03:00:00 2020
Mar 30 14:40:21 shelly25_01-0099 ESP-APP: Boot Count 139
Mar 30 14:40:21 shelly25_01-0099 ESP-CFG: Saved to flash at F4, Count 1078, Bytes 4096

I have upgraded the second shelly, to 8.2 and it reports values fine. So I guess something is wrong with that device itself. The main thing is that I still can use it as a relay...

Thanks for your help guys, appreciated.

arendst, 2020-03-30:
Great you solved the issue. 

- Is the failing Shelly 2.5 a (much) newer version than your functional Shelly 2.5?
- Are you sure the failing shelly IS a shelly 2.5?

kovaga, 2020-03-30:
the failing unit was bought in August 2019 while the other one in November 2019.
I am pretty sure it is a shelly 2.5 as relay1 controlls the outside light, while relay 2 controlls the outside socket and they are integrated via MQTT to OpenHAB. :)

closed 2020-03-30 by kovaga