arendst/Tasmota #8029: Translation files for WebUI - Where are located

bovirus, 2020-03-30:
Hello.

Where can I find the localization file (ex. English and Italian) for WebUI?

Thanks.

closed 2020-03-30 by close-issue-app[bot]

arendst, 2020-03-30:
Look in repository /tasmota/language

bovirus, 2020-03-30:
@arendst 

Thanks.
I found the file.
Can I make a PR with the IT revised file for translation?

I propose you to add in repository home the info how to create/revise a language.

- Where find the file
- How to edit
- How to make the PR

About firmware localized I propose you to change the firmware file name to

**_firmware-lang-xx.bin_**

in this way all files for different languages are order neaby to the others.

arendst, 2020-03-30:
Thx for the proposals but:
- it is generally felt that people wanting to add/update a language can/will find it in the language folder.
- Doing PR's is a github excercise and does not falls within the program documentation.
- I see your file sorting issue and will keep it in mind.

So if you want to make changes to the language file just follow Github principles of making a PR.

bovirus, 2020-03-30:
@arendst 

PR was uploaded

https://github.com/arendst/Tasmota/pull/8031

Thanks.