arendst/Tasmota #8032: WS2812B- long lines

BugsonPL, 2020-03-30:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Looks like Tasmota 8.2.0 official .bin FV has problem with handling bigger amount of LED's (in my case >=180), with ~20 LED Tasmota works fine.
The same hardware works fine with WLED firmware.
I have tried Wemos D1 mini and NodeMCU boards the same results Tasmota not working, WLED ok.

Maybe it is the meter of internal pull-up pin configuration.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-03-30:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2020-04-02:
Hi, what is the output of command `Pixels` on the console?

ascillato2, 2020-04-04:
Please, when you have time, tell us if the amount of pixels was correctly defined using pixels command. Thanks.

closed 2020-04-04 by ascillato2

BugsonPL, 2020-04-14:
Yes the amount of pixels was correctly defined.