arendst/Tasmota #8034: Scheme 4 speed 1 --> very slow

zImage-de, 2020-03-30:
### PROBLEM DESCRIPTION
LED Stripe on Tasmota 8.2.0 function very well, but effect like scheme 4 ist very slow.
Even when speed 1 is set, a cycle need approx. 25 sec ... 

### REQUESTED INFORMATION
Is it possible to that effect scheme's (3,4) runs faster? 

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): MagicHome with ESP8285
- [ ] Tasmota binary firmware version number used: Tasmota 8.2.0
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
no special output

  Configuration output here:
nothing special configured 

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
no rules ...

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
{"Status":{"Module":18,"FriendlyName":["Tasmota"],"Topic":"LEDStripe","ButtonTopic":"stop","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
{"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"2T06:10:34","StartupUTC":"2020-03-28T09:56:03","Sleep":50,"CfgHolder":4617,"BootCount":30,"BCResetTime":"2020-03-24T17:42:36","SaveCount":4767,"SaveAddress":"F9000"}}
{"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-20T14:45:23","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"325/699"}}
{"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Christmann",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
{"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
{"StatusNET":{"Hostname":"LEDStripe-7940","IPAddress":"192.168.192.XXX","Gateway":"192.168.192.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.192.1","Mac":"XX:XX:XX:XX:XX:XX","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
{"StatusMQT":{"MqttHost":"192.168.192.XXX","MqttPort":1883,"MqttClientMask":"LEALED","MqttClient":"LEALED","MqttUser":"XXX","MqttCount":9,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
{"StatusTIM":{"UTC":"2020-03-30T16:06:37","Local":"2020-03-30T17:06:37","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:XX","Sunset":"19:XX"}}
{"StatusSNS":{"Time":"2020-03-30T17:06:37"}}
{"StatusSTS":{"Time":"2020-03-30T17:06:37","Uptime":"2T06:10:34","UptimeSec":195034,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":9,"POWER":"OFF","Dimmer":100,"Color":"8C00FF","HSBColor":"273,100,100","Channel":[55,0,100],"Scheme":2,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"XXXX","BSSId":"XX:XX:XX:XX:XX:XX","Channel":6,"RSSI":76,"Signal":-62,"LinkCount":9,"Downtime":"0T00:00:42"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
17:12:12 CMD: Scheme 4
17:12:12 MQT: stat/LEDStripe/RESULT = {"Scheme":4}
17:12:20 CMD: speed 1 
17:12:20 MQT: stat/LEDStripe/RESULT = {"Speed":1}

```

### TO REPRODUCE
Scheme 4
speed 1

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
no sreenshots 

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-03-31:
How fast would you like a cycle to be?

arendst, 2020-03-31:
How about these speeds:

Speed 1: 4sec, 2: 6sec, 3: 12sec, 4: 24sec, 5: 36sec, 6: 48sec, etc

zImage-de, 2020-03-31:
Thanks Arendst ... 4 sec is ok for me ... however longer cycling time than 1 or 2 minutes does not make sense for me ... it is to slow  to see changes

arendst, 2020-03-31:
Others might like it slow and it was even slower before this change.

closed 2020-03-31 by arendst

digiblur, 2020-04-02:
Miss the days of scheme 4 having the seizure speed ;)