arendst/Tasmota #8036: Tasmota v8.2.0 PIR Motion Sensors / Wemos D1 Mini / Home Assistant

Schaussi, 2020-03-31:
### PROBLEM DESCRIPTION
Testing Tasmota 8.2.0, I see that buttons/switches/motion sensors attached to Wemos D1 Mini are not detected by Home Assistant any more, so I had to revert to 8.1.0 for the moment where it works fine.
As I read in the changelog, some new HA API feature is used for buttons, so it seems like switches do not work the way described in the [wiki](https://github.com/arendst/Tasmota/wiki/PIR-Motion-Sensors) any more.
I see the button change event within Tasmota console, but it is not recognized by HA as an entity at all (using latest HA release, MQTT integration, Mosquitto broker).
Tested with MQTT Explorer; it seems like v8.2.0 uses another topic structure, e.g. 8.2.0 sensor appears under "tele/", 8.1.0 appears under "myname/", as far as I remember.

### REQUESTED INFORMATION
Please update the Wiki with the necessary steps for e.g. using an SR501 motion sensor.

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [x] Tasmota binary firmware version number used: tasmota.bin 8.2.0 and 8.1.0 via Github or Tasmotizer
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
(Sorry don't have it at the moment, but don't remember any errors; I guess it can be reproduced with any button on any device)

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
(no errors; rule used like described in wiki)

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Tasmotizer -> Firmware tasmota.bin 8.2.0 to Wemos D1 Mini.
Send default config, access web interface, configure e.g. pin D1 as Switch(9).
SetOption19 1 for HA discovery.
Rule like described on PIR wiki page

### EXPECTED BEHAVIOUR
Button/switch/motion sensor detection within HA, like with v8.1.0.

### SCREENSHOTS

### ADDITIONAL CONTEXT

nmweizi, 2020-03-31:
i am actually having the same issue

effelle, 2020-03-31:
Buttons and Switches have changed on 8.2 discovery.
Is in the docs: https://tasmota.github.io/docs/Home-Assistant/

Schaussi, 2020-03-31:
Thank you, I haven't seen this part of doc yet, will try that in a few days.
I guess it will work then.
Maybe you could delete or redirect the apparently old wiki page which contains the "old" 8.1.0 config way to avoid confusion: https://github.com/arendst/Tasmota/wiki/PIR-Motion-Sensors

closed 2020-03-31 by Schaussi

effelle, 2020-03-31:
The wiki was there for legacy versions, is why we added  a comment in bold: **ALL INFORMATION APPLIES ONLY TO VERSIONS PRIOR TO TASMOTA V7**. 
For future reference , the correct link is on the docs pages: https://tasmota.github.io/docs/PIR-Motion-Sensors/.
Now, if you look at it you will see is virtually the same of the old wiki, because is not wrong and the example doesn't consider the use of the discovery at all.
What you did not get well yet is how discovery was working on 8.1 and how is working now.
Using the rule on a 8.1 and on a 8.2 will lead to the same thing, the only difference is on 8.2 there is not the mqtt enforcement anymore, hence your topic will start with /tele or /cmnd instead of the name of the device. 
On 8.1 The discovery was creating a binary sensor (a switch in your case) still connected to the relay. In that case the binary sensor wasn't that helpful without a rule. Adding the rule of the example will simply work as expected. 
When you use a switch in a rule you are detaching the relay from the switch because rules have the priority so when a switch changes state in a rule the relay will not follow the switch.
You said was working fine on 8.1 and that is true, but was not operating has you think. 
The switch was and still is not listening to the topic you add on the rule (stat/%topic%/PIR1) but was simply reflect what Tasmota is publishing on `stat/%topic%/SWITCH1 = {"STATE":"ON"}` when discovery is enabled.  Is why you were able to see the change on HA.
On 8.2, things have changed:  the switch is still a binary sensor but can be also a trigger depending of the config, see docs. 
To have the entity/trigger under HA you need to change the `switchtopic` to a custom name.
Doing so, you are passing the control of the switch directly to HA and the rule will not work anymore.  The switch will be separated from the relay (or power control in general) and will work just for HA automation.

nmweizi, 2020-04-03:
is fine. thank you.