arendst/Tasmota #8038: Tasmota 8.2 fails to properly provision a RGBCW light using Home Assistant MQTT auto-discovery.

bhaonvashon, 2020-03-31:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Tasmota 8.1 properly provisions a RGBCW (5 channel) light for use with Home Assistant using the MQTT auto-discovery mechanism.  When the device is provisioned using MQTT auto-discovery, HA offers a `more_info` popup for the device that includes a `color wheel` and `brightness` and `color temperature` controls.

Tasmota 8.2 appears to break the capability to provision a RGBCW light using MQTT auto-discovery.  When the device is provisioned using MQTT auto-discovery, HA no offers a `more_info` popup for the device that includes a `color wheel` and `brightness` control but does *not* offer a `color temperature` control.  When provision using MQTT auto-discovery, HA does not have the ability to control the device via color temperature.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Teckin SB50
- [ ] Tasmota binary firmware version number used: 8.2.0.3
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: TasmoAdmin
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

05:34:10 CMD: backlog template;module;gpio 255
05:34:10 MQT: devices/lights/teckin-sb50-ef82d0/stat/RESULT = {"NAME":"Teckin SB50v3","GPIO":[0,0,0,0,37,40,0,0,38,41,39,0,0],"FLAG":0,"BASE":18}
05:34:11 MQT: devices/lights/teckin-sb50-ef82d0/stat/RESULT = {"Module":{"0":"Teckin SB50v3"}}
05:34:11 MQT: devices/lights/teckin-sb50-ef82d0/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"37":"PWM1"},"GPIO5":{"40":"PWM4"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"38":"PWM2"},"GPIO13":{"41":"PWM5"},"GPIO14":{"39":"PWM3"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
0

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
05:36:45 CMD: status0
05:36:45 MQT: devices/lights/teckin-sb50-ef82d0/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["teckin_sb50_ef82d0"],"Topic":"teckin-sb50-ef82d0","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
05:36:49 MQT: devices/lights/teckin-sb50-ef82d0/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["teckin_sb50_ef82d0"],"Topic":"teckin-sb50-ef82d0","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
05:36:49 MQT: devices/lights/teckin-sb50-ef82d0/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://banderson-tp.local:5555/data/firmwares/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:04:23","StartupUTC":"2020-03-31T09:32:26","Sleep":50,"CfgHolder":4617,"BootCount":500,"BCResetTime":"2020-03-28T21:16:49","SaveCount":673,"SaveAddress":"F8000"}}
05:36:49 MQT: devices/lights/teckin-sb50-ef82d0/stat/STATUS2 = {"StatusFWR":{"Version":"8.2.0.3(290ec88-tasmota)","BuildDateTime":"2020-03-30T19:00:15","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"401/699"}}
05:36:49 MQT: devices/lights/teckin-sb50-ef82d0/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Jia Ting",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00000228","00000000"]}}
05:36:49 MQT: devices/lights/teckin-sb50-ef82d0/stat/STATUS4 = {"StatusMEM":{"ProgramSize":578,"Free":424,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14400B","FlashMode":0,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
05:36:49 MQT: devices/lights/teckin-sb50-ef82d0/stat/STATUS5 = {"StatusNET":{"Hostname":"tasmota-ef82d0","IPAddress":"10.0.1.16","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.1","Mac":"CC:50:E3:EF:82:D0","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
05:36:49 MQT: devices/lights/teckin-sb50-ef82d0/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.201","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_EF82D0","MqttUser":"device","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
05:36:49 MQT: devices/lights/teckin-sb50-ef82d0/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-03-31T09:36:49","Local":"2020-03-31T05:36:49","StartDST":"2020-03-08T02:00:00","EndDST":"2020-11-01T02:00:00","Timezone":99,"Sunrise":"01:28","Sunset":"14:19"}}
05:36:49 MQT: devices/lights/teckin-sb50-ef82d0/stat/STATUS10 = {"StatusSNS":{"Time":"2020-03-31T05:36:49"}}
05:36:49 MQT: devices/lights/teckin-sb50-ef82d0/stat/STATUS11 = {"StatusSTS":{"Time":"2020-03-31T05:36:49","Uptime":"0T00:04:23","UptimeSec":263,"Heap":22,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":42,"Color":"0,0,0,46,107","HSBColor":"0,0,0","White":42,"CT":395,"Channel":[0,0,0,18,42],"Scheme":0,"Fade":"ON","Speed":2,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Jia Ting","BSSId":"E0:63:DA:8E:23:56","Channel":11,"RSSI":100,"Signal":-23,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

  * Flash with Tasmota 8.2.0.3
  * Setup and enable HA auto-discovery.
  * Use HA to control device using the `more_info` dialog.
  * Note that HA does not offer a `color temperature` control and cannot in general control the device using color temperature (and thus the cool and warm white LEDs of the device).

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
[... 156 lines cut ...]
POWER
Dimmer = 42
CT = 395
```

RESULT:

```json
{
  "POWER": "ON",
  "Dimmer": 42,
  "Color": "0,0,0,32,75",
  "HSBColor": "316,29,0",
  "Channel": [
    0,
    0,
    0,
    13,
    29
  ],
  "CT": 395
}
```

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2020-03-31:
On 8.2 we changed a lot of things and the most important was removing the mqtt enforcement, meaning there is no more the need to invert `%topic%` and `%prefix%`. 
If you look closely to Tasmota logs you will see an error in logs:
```
HASS: MQTT discovery failed due to too long topic or friendly name. 
Please shorten topic and friendly name. Failed to format(xxx/xxx):
```
You have a very long topic, `devices/lights/teckin-sb50-ef82d0/%prefix%`, if you do a test shortening the name a bit you will have all in place. 
For example try with  `devices/lights/teckin-sb50/%prefix%`:

```json
{
"name":"teckin_sb50_ef82d0",
"stat_t":"devices/lights/tele/teckin-sb50/STATE",
"avty_t":"devices/lights/tele/teckin-sb50/LWT",
"pl_avail":"Online",
"pl_not_avail":"Offline",
"cmd_t":"devices/lights/cmnd/teckin-sb50/POWER",
"val_tpl":"{{value_json.POWER}}",
"pl_off":"OFF",
"pl_on":"ON",
"uniq_id":"471078_LI_1",
"dev":{
"ids":[
"471078"
]
},
"bri_cmd_t":"devices/lights/cmnd/teckin-sb50/Dimmer",
"bri_stat_t":"devices/lights/tele/teckin-sb50/STATE",
"bri_scl":100,
"on_cmd_type":"brightness",
"bri_val_tpl":"{{value_json.Dimmer}}",
"rgb_cmd_t":"devices/lights/cmnd/teckin-sb50/Color2",
"rgb_stat_t":"devices/lights/tele/teckin-sb50/STATE",
"rgb_val_tpl":"{{value_json.Color.split(',')[0:3]|join(',')}}",
"fx_cmd_t":"devices/lights/cmnd/teckin-sb50/Scheme",
"fx_stat_t":"devices/lights/tele/teckin-sb50/STATE",
"fx_val_tpl":"{{value_json.Scheme}}",
"fx_list":[
"0",
"1",
"2",
"3",
"4"
],
"clr_temp_cmd_t":"devices/lights/cmnd/teckin-sb50/CT",
"clr_temp_stat_t":"devices/lights/tele/teckin-sb50/STATE",
"clr_temp_val_tpl":"{{value_json.CT}}"
}
```

Now you question will be ... Why? 
Because since we removed the MQTT enforcement the length of the various topics can't be shortened anymore as we did on older versions. `RGBCW` and `RGBWW` entities are always near to the length limit. 
In other words, you only solution is to reduce the topic and/or the name of the device.

bhaonvashon, 2020-03-31:
Thanks for the prompt reply!  You are so right.  Learned a bit more about Tasmota and logging from this.  Eventually found the message you referenced.  Shortening fulltopic fixed the issue.

Closing as this is really not an issue, but a mis-configuration on my part.

closed 2020-03-31 by bhaonvashon

effelle, 2020-03-31:
You are welcome!

shaymdev, 2021-01-23:
@effelle Would it be difficult or against convention to make that HASS discovery failure more apparent... like maybe so that the default `weblog 2` would show that message?

It took me quite some time to figure out that this is what was plaguing _some_ of my devices.

arendst, 2021-01-23:
The current log level for this message is 1 (ERROR) so you must have seen it with ``weblog 2``...

shaymdev, 2021-01-23:
Maybe I'm doing something wrong. Here's the output when I had `weblog 2` and then I `setoption19 1` but don't see any error (though the device is incorrectly setup in HA). Then I `setoption19 0`, set `weblog 4` then do `setoption19 1` again and this time I see the error in the console. 

`00:00:58 CMD: weblog
00:00:58 MQT: stat/bulbOffice2_longNameHereForTest/RESULT = {"WebLog":2}
00:01:06 CMD: so19 1
00:01:06 MQT: stat/bulbOffice2_longNameHereForTest/RESULT = {"SetOption19":"ON"}
00:01:06 MQT: homeassistant/light/E24C9D_LI_1/config = {"name":"Office Lamp 2","stat_t":"tele/bulbOffice2_longNameHereForTest/STATE","avty_t":"tele/bulbOffice2_longNameHereForTest/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"cmnd/bulbOffice2_longNameHereForTest/POWER","val_tpl":"{{value_json.POWER}}","pl_off":"OFF","pl_on":"ON","uniq_id":"E24C9D_LI_1","dev":{"ids":["E24C9D"]},"bri_cmd_t":"cmnd/bulbOffice2_longNameHereForTest/Dimmer","bri_stat_t":"tele/bulbOffice2_longNameHereForTest/STATE","bri_scl":100,"on_cmd_type":"brightness","bri_val_tpl":"{{value_json.Dimmer}}","rgb_cmd_t":"cmnd/bulbOffice2_longNameHereForTest/Color2","rgb_stat_t":"tele/bulbOffice2_longNameHereForT ... (retained)
00:01:06 MQT: homeassistant/sensor/E24C9D_status/config = {"name":"Office Lamp 2 status","stat_t":"tele/bulbOffice2_longNameHereForTest/HASS_STATE","avty_t":"tele/bulbOffice2_longNameHereForTest/LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attr_t":"tele/bulbOffice2_longNameHereForTest/HASS_STATE","unit_of_meas":"%","val_tpl":"{{value_json['RSSI']}}","ic":"mdi:information-outline","uniq_id":"E24C9D_status","dev":{"ids":["E24C9D"],"name":"Office Lamp 2","mdl":"Lohas RGBWW","sw":"9.2.0(tasmota)","mf":"Tasmota"}} (retained)
00:01:50 MQT: tele/bulbOffice2_longNameHereForTest/STATE = {"Time":"1970-01-01T00:01:50","Uptime":"0T00:01:50","UptimeSec":110,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":20,"MqttCount":1,"POWER":"ON","Dimmer":100,"Color":"36,69,255,0,255","HSBColor":"231,86,100","White":100,"CT":500,"Channel":[14,27,100,0,100],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"ShadyRest_IoT","BSSId":"78:8A:20:DA:F8:E7","Channel":11,"RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:03"}}
00:01:50 MQT: stat/bulbOffice2_longNameHereForTest/RESULT = {"POWER":"ON","Dimmer":100,"Color":"36,69,255,0,255","HSBColor":"231,86,100","White":100,"CT":500,"Channel":[14,27,100,0,100]}
00:02:36 CMD: so19 0
00:02:36 MQT: stat/bulbOffice2_longNameHereForTest/RESULT = {"SetOption19":"OFF"}
00:02:37 LOG: Home Assistant MQTT Discovery disabled.
00:04:06 CMD: weblog 4
00:04:06 MQT: stat/bulbOffice2_longNameHereForTest/RESULT = {"WebLog":4}
00:04:06 CFG: Saved to flash at FA, Count 201, Bytes 4096
00:04:06 WIF: Checking connection...
00:04:24 CMD: so19 1
00:04:24 SRC: WebConsole from 192.168.1.53
00:04:24 CMD: Group 0, Index 19, Command "SO", Data "1"
00:04:24 MQT: stat/bulbOffice2_longNameHereForTest/RESULT = {"SetOption19":"ON"}
00:04:24 MQT: homeassistant/binary_sensor/E24C9D_SW_6/config =  (retained)
00:04:24 MQT: homeassistant/device_automation/E24C9D_SW_7_TOGGLE/config =  (retained)
00:04:24 MQT: homeassistant/device_automation/E24C9D_SW_7_HOLD/config =  (retained)
00:04:24 MQT: homeassistant/binary_sensor/E24C9D_SW_7/config =  (retained)
00:04:24 MQT: homeassistant/device_automation/E24C9D_SW_8_TOGGLE/config =  (retained)
00:04:24 MQT: homeassistant/device_automation/E24C9D_SW_8_HOLD/config =  (retained)
00:04:24 MQT: homeassistant/binary_sensor/E24C9D_SW_8/config =  (retained)
00:04:24 MQT: homeassistant/cover/E24C9D_SHT_1/config =  (retained)
00:04:24 MQT: homeassistant/cover/E24C9D_SHT_2/config =  (retained)
00:04:24 MQT: homeassistant/cover/E24C9D_SHT_3/config =  (retained)
00:04:24 MQT: homeassistant/cover/E24C9D_SHT_4/config =  (retained)
00:04:24 MQT: homeassistant/switch/E24C9D_RL_1/config =  (retained)
00:04:24 HASS: MQTT discovery failed due to too long topic or device/friendly name. Please shorten topic and/or device/friendly name. Failed to format (161/147):
00:04:24 ,"clr_temp_cmd_t":"cmnd/bulbOffice2_longNameHereForTest/CT","clr_temp_stat_t":"tele/bulbOffice2_longNameHereForTest/STATE","clr_temp_val_tpl":"{{value_json.CT}}"
00:04:24 HASS: MQTT discovery failed due to too long topic or device/friendly name. Please shorten topic and/or device/friendly name. Failed to format (182/147):
00:04:24 ,"whit_val_cmd_t":"cmnd/bulbOffice2_longNameHereForTest/White","whit_val_stat_t":"tele/bulbOffice2_longNameHereForTest/STATE","whit_val_scl":100,"whit_val_tpl":"{{value_json.White}}"
00:04:24 MQT: homeassistant/light/E24C9D_LI_1/config = {"name":"Office Lamp 2","stat_t":"tele/bulbOffice2_longNameHereForTest/STATE","avty_t":"tele/bulbOffice2_longNameHereForTest/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"cmnd/bulbOffice2_longNameHereForTest/POWER","val_tpl":"{{value_json.POWER}}","pl_off":"OFF","pl_on":"ON","uniq_id":"E24C9D_LI_1","dev":{"ids":["E24C9D"]},"bri_cmd_t":"cmnd/bulbOffice2_longNameHereForTest/Dimmer","bri_stat_t":"tele/bulbOffice2_longNameHereForTest/STATE","bri_scl":100,"on_cmd_type":"brightness","bri_val_tpl":"{{value_json.Dimmer}}","rgb_cmd_t":"cmnd/bulbOffice2_longNameHereForTest/Color2","rgb_stat_t":"tele/bulbOffice2_longNameHereForT ... (retained)
00:04:24 MQT: homeassistant/switch/E24C9D_RL_2/config =  (retained)
00:04:24 MQT: homeassistant/light/E24C9D_LI_2/config =  (retained)
00:04:24 MQT: homeassistant/switch/E24C9D_RL_3/config =  (retained)
[... cut ...]
00:04:24 MQT: homeassistant/sensor/E24C9D_status/config = {"name":"Office Lamp 2 status","stat_t":"tele/bulbOffice2_longNameHereForTest/HASS_STATE","avty_t":"tele/bulbOffice2_longNameHereForTest/LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attr_t":"tele/bulbOffice2_longNameHereForTest/HASS_STATE","unit_of_meas":"%","val_tpl":"{{value_json['RSSI']}}","ic":"mdi:information-outline","uniq_id":"E24C9D_status","dev":{"ids":["E24C9D"],"name":"Office Lamp 2","mdl":"Lohas RGBWW","sw":"9.2.0(tasmota)","mf":"Tasmota"}} (retained)
00:04:24 MQT: tasmota/discovery/D8F15BE24C9D/config =  (retained)
00:04:24 CFG: Saved to flash at F9, Count 202, Bytes 4096`