arendst/Tasmota #8040: Apparent regression after changes in the scope of PR #7506

teixeluis, 2020-03-31:
### PROBLEM DESCRIPTION

In the SerialReceived message, even if the received data is not a a JSON, it is not being surrounded by quotation marks. This broke the previous behaviour. For example:

Before:

```
{"SerialReceived":"FE01080927090053230000236401000050000088120300364900000487030000080000F2"}
```

After:

```
{"SerialReceived":FE01080927090053230000236401000050000088120300364900000487030000080000F2}
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): _____
- [X] Tasmota binary firmware version number used: 8.2.0
  - [X] Pre-compiled

arendst, 2020-03-31:
Try to enable mqtt receive raw by sending something  with ``SerialSend5`` once.

closed 2020-03-31 by close-issue-app[bot]

teixeluis, 2020-03-31:
Hi @arendst,

Thanks for the feedback. I haven't yet tested myself, but according to another user, SerialSend5 is being used, and still something like:

```
{"SerialReceived":FE01080927090053230000236401000050000088120300364900000487030000080000F2}
```
is received.

Does this makes sense?

Cheers

arendst, 2020-04-01:
Pls retry.

teixeluis, 2020-04-01:
> Pls retry.

Thanks.