arendst/Tasmota #8041: sht30 doesn't show temperature and humidity

mtasilva, 2020-03-31:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I have tasmota with sht30 but I can't see the temperature and humidity

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

14:21:34 CMD: Grupo 0, Indíce 1, Comando "TEMPLATE", Dados ""
14:21:34 RSL: stat/tasmota_DCEB60/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,0,0,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}

14:22:58 RSL: stat/tasmota_DCEB60/RESULT = {"Module":{"18":"Generic"}}

14:23:30 CMD: Grupo 0, Indíce 1, Comando "GPIO", Dados ""
14:23:30 RSL: stat/tasmota_DCEB60/RESULT = {"GPIO0":{"0":"Nenhum"},"GPIO1":{"0":"Nenhum"},"GPIO2":{"0":"Nenhum"},"GPIO3":{"0":"Nenhum"},"GPIO4":{"5":"I2C SCL"},"GPIO5":{"6":"I2C SDA"},"GPIO12":{"0":"Nenhum"},"GPIO13":{"0":"Nenhum"},"GPIO14":{"0":"Nenhum"},"GPIO15":{"0":"Nenhum"},"GPIO16":{"0":"Nenhum"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

14:24:05 RSL: stat/tasmota_DCEB60/STATUS = {"Status":{"Module":18,"FriendlyName":["Tasmota"],"Topic":"tasmota_DCEB60","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:24:05 RSL: stat/tasmota_DCEB60/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:02:44","StartupUTC":"2020-03-31T13:21:21","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2020-03-31T12:49:31","SaveCount":27,"SaveAddress":"F9000"}}
14:24:05 RSL: stat/tasmota_DCEB60/STATUS2 = {"StatusFWR":{"Version":"8.2.0.3(tasmota)","BuildDateTime":"2020-03-31T12:47:19","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"322/699"}}
14:24:05 RSL: stat/tasmota_DCEB60/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["CasaEsc4G",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
14:24:05 RSL: stat/tasmota_DCEB60/STATUS4 = {"StatusMEM":{"ProgramSize":579,"Free":424,"Heap":26,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashMode":3,"Features":["00000816","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
14:24:05 RSL: stat/tasmota_DCEB60/STATUS5 = {"StatusNET":{"Hostname":"tasmota_DCEB60-2912","IPAddress":"192.168.1.167","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"84:F3:EB:DC:EB:60","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
14:24:05 RSL: stat/tasmota_DCEB60/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DCEB60","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
14:24:05 RSL: stat/tasmota_DCEB60/STATUS7 = {"StatusTIM":{"UTC":"2020-03-31T13:24:05","Local":"2020-03-31T14:24:05","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:28","Sunset":"19:19"}}
14:24:05 RSL: stat/tasmota_DCEB60/STATUS10 = {"StatusSNS":{"Time":"2020-03-31T14:24:05"}}
14:24:05 RSL: stat/tasmota_DCEB60/STATUS11 = {"StatusSTS":{"Time":"2020-03-31T14:24:05","Uptime":"0T00:02:44","UptimeSec":164,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"xxx","BSSId":"00:06:91:75:BF:70","Channel":6,"RSSI":52,"Signal":-74,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

Grupo 0, Indíce 1, Comando "WEBLOG", Dados "4"
14:24:47 RSL: stat/tasmota_DCEB60/RESULT = {"WebLog":4}
14:24:48 WIF: A verificar ligação...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2020-03-31:
Wrong [build](https://github.com/arendst/Tasmota/blob/development/BUILDS.md). Starting from 8.2 you must use `tasmota-sensor` to have your SH30 working.

closed 2020-03-31 by effelle

mtasilva, 2020-03-31:
Please i understand your time.
My version is 8.2, what, hown can I use tasmota-sensor?
I download from :https://github.com/arendst/Tasmota , and after flash with visual code.
Can you give me a direction?

effelle, 2020-03-31:
You can try to do an OTA update directly from Tasmota webui on firmware upgrade--> OTA Url using this path:
http://thehackbox.org/tasmota/release/tasmota-sensors.bin.
Or flash it again using visual code with the very same bin file.

mtasilva, 2020-03-31:
You are great, it's work. What is the difference between tasmota.bin and tasmota-sensors.bin.
Thought that there was only this version tasmota.bin. Is tasmota-sensors mostly used for sensors?

effelle, 2020-03-31:
Yes. the only difference is Tasmota-Sensors has many more sensor enabled by default.

mtasilva, 2020-04-01:
Sorry my last post,  tasmota-sensors suport rcwl 0516 on esp12F?