arendst/Tasmota #8042: Driver for ZH03B Serial Laser Particles Sensor similar to PMS7003

ShaggyDog18, 2020-03-31:
**Have you looked for this feature in other issues and in the docs?**  
Yes, found a similar sensor PMS7003  with UART interface which is supported.
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
looking for driver for ZH03B Serial Laser Particles Sensor which is similar to PMS7003
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

gururise, 2020-04-12:
Looks useful enough I may implement this if I can get my hands on a ZH03B anytime soon.

ShaggyDog18, 2020-04-13:
Excellent! Thank you. Please, refer to datasheet, it is detailed and contains all necessary information!
I am expecting my ZH03 module to arrive in couple of weeks. So, will be able to test soon...

ShaggyDog18, 2020-05-02:
Just received the module and tested it. Works fine. So, will be able to test if implemented.

ShaggyDog18, 2020-05-29:
> Looks useful enough I may implement this if I can get my hands on a ZH03B anytime soon.

@gururise, are you still willing to create driver for the module?

ShaggyDog18, 2020-06-05:
found out ZH03B  is similar to PMS3003 / PMS5003...

ShaggyDog18, 2020-07-03:
I've just developed a nice and neat library for the module: [SD_ZH03B](https://github.com/ShaggyDog18/SD_ZH03B)
Works fine. Would that make possible deployment the module to Tasmota firmware?

closed 2020-08-03 by stale[bot]