arendst/Tasmota #8044: make tasmota less chatty on battery operated devices

brjhaverkamp, 2020-03-31:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**  

I am experimenting with Tasmota 8.2 on a Earykong TYMC-1 Door Window Sensor
(https://templates.blakadder.com/TYMC-1.html)
After installing it successfully I monitored it in node red and noticed that on every event (door open or door close) the device emits 13 messages.  See the example below
3/31/2020, 6:03:54 PMnode: da8b81ba.a7494tele/tasmota-0138/LWT : msg.payload : string[6]"Online"
3/31/2020, 6:03:54 PMnode: da8b81ba.a7494cmnd/tasmota-0138/POWER : msg.payload : string[0]""
3/31/2020, 6:03:54 PMnode: da8b81ba.a7494tele/tasmota-0138/INFO1 : msg.payload : string[117]"{"Module":"Tuya MCU","Version":"8.1.0(tasmota)","FallbackTopic":"cmnd/DVES_60E08A_fb/","GroupTopic":"cmnd/tasmotas/"}"
3/31/2020, 6:03:54 PMnode: da8b81ba.a7494tele/tasmota-0138/INFO2 : msg.payload : string[80]"{"WebServerMode":"Admin","Hostname":"tasmota-0138","IPAddress":"192.168.71.237"}"
3/31/2020, 6:03:54 PMnode: da8b81ba.a7494tele/tasmota-0138/INFO3 : msg.payload : string[28]"{"RestartReason":"Power on"}"
3/31/2020, 6:03:54 PMnode: da8b81ba.a7494stat/tasmota-0138/RESULT : msg.payload : string[15]"{"POWER":"OFF"}"
3/31/2020, 6:03:54 PMnode: da8b81ba.a7494stat/tasmota-0138/POWER : msg.payload : string[3]"OFF"
3/31/2020, 6:03:54 PMnode: da8b81ba.a7494tele/tasmota-0138/RESULT : msg.payload : string[85]"{"TuyaReceived":{"Data":"55AA0005000501010001010D","Cmnd":5,"CmndData":"0101000101"}}"
3/31/2020, 6:03:54 PMnode: da8b81ba.a7494stat/tasmota-0138/RESULT : msg.payload : string[14]"{"POWER":"ON"}"
3/31/2020, 6:03:54 PMnode: da8b81ba.a7494stat/tasmota-0138/POWER : msg.payload : string[2]"ON"
3/31/2020, 6:03:54 PMnode: da8b81ba.a7494tele/tasmota-0138/RESULT : msg.payload : string[85]"{"TuyaReceived":{"Data":"55AA00050005030400010213","Cmnd":5,"CmndData":"0304000102"}}"
3/31/2020, 6:03:54 PMnode: da8b81ba.a7494stat/tasmota-0138/BATT : msg.payload : string[4]"high"
3/31/2020, 6:03:54 PMnode: da8b81ba.a7494tele/tasmota-0138/RESULT : msg.payload : string[51]"{"TuyaReceived":{"Data":"55AA0002000001","Cmnd":2}}

Many of these messages are not needed in normal operation. Basically only the stat/topic/POWER messages would suffice. 
The extra messages make the device less reliable, as it needs to transmit in a limited time, before it goes to sleep again. Moreover, it is an extra battery drain (however limited it might be)
It would be great if there is a mode (battery mode) that switches off these excess messages automaticaly, or a console command to do so manually

Kind regards,

Bert Haverkamp

effelle, 2020-03-31:
Did you tried to play with logs settingsto see if you can reduce the traffic? 
You can use a lower level logging to the `WebLog` and `SerialLog` both default to `2`, `MqttLog` default is already `0`. 
A word of warning: your device communicate trough serial to the Tuya MCU, setting serial log to zero is probably not a good idea.

brjhaverkamp, 2020-04-01:
That was a bit quick to close. Could you please reopen? 
Yes, I have set all log levels to 0 even. This has no effect.
I also made sure I closed the web interface, as I read somewhere that having the webpage open can have an effect on the messages. But also this has no effect.
The minimum I have is 0. The different power messages might be a problem with the way the device reports the changes, so I am looking into that seperately. 
But the INFO and RESULT messages are superfluous in my opinion.
On the chat, blakadder mentioned this indeed a known fact but at the moment not configurable without digging into the source

arendst, 2020-04-01:
> 3/31/2020, 6:03:54 PMnode: da8b81ba.a7494tele/tasmota-0138/LWT : msg.payload : string[6]"Online"

Functional to tell the broker we're back in business

> 3/31/2020, 6:03:54 PMnode: da8b81ba.a7494cmnd/tasmota-0138/POWER : msg.payload : string[0]""

This messages could be disabled by a yet to define SetOption X

> 3/31/2020, 6:03:54 PMnode: da8b81ba.a7494tele/tasmota-0138/INFO1 : msg.payload : string[117]"{"Module":"Tuya MCU","Version":"8.1.0(tasmota)","FallbackTopic":"cmnd/DVES_60E08A_fb/","GroupTopic":"cmnd/tasmotas/"}"
3/31/2020, 6:03:54 PMnode: da8b81ba.a7494tele/tasmota-0138/INFO2 : msg.payload : string[80]"{"WebServerMode":"Admin","Hostname":"tasmota-0138","IPAddress":"192.168.71.237"}"
3/31/2020, 6:03:54 PMnode: da8b81ba.a7494tele/tasmota-0138/INFO3 : msg.payload : string[28]"{"RestartReason":"Power on"}"

These messages could be disabled by a yet to define SetOption Y

> 3/31/2020, 6:03:54 PMnode: da8b81ba.a7494stat/tasmota-0138/RESULT : msg.payload : string[15]"{"POWER":"OFF"}"

Functional

> 3/31/2020, 6:03:54 PMnode: da8b81ba.a7494stat/tasmota-0138/POWER : msg.payload : string[3]"OFF"

This messages could be disabled by a yet to define SetOption X

> 3/31/2020, 6:03:54 PMnode: da8b81ba.a7494tele/tasmota-0138/RESULT : msg.payload : string[85]"{"TuyaReceived":{"Data":"55AA0005000501010001010D","Cmnd":5,"CmndData":"0101000101"}}"

Can already be disabled with ``SetOption66 0``

> 3/31/2020, 6:03:54 PMnode: da8b81ba.a7494stat/tasmota-0138/RESULT : msg.payload : string[14]"{"POWER":"ON"}"

Functional

> 3/31/2020, 6:03:54 PMnode: da8b81ba.a7494stat/tasmota-0138/POWER : msg.payload : string[2]"ON"

This messages could be disabled by a yet to define SetOption X

> 3/31/2020, 6:03:54 PMnode: da8b81ba.a7494tele/tasmota-0138/RESULT : msg.payload : string[85]"{"TuyaReceived":{"Data":"55AA00050005030400010213","Cmnd":5,"CmndData":"0304000102"}}"

Can already be disabled with ``SetOption66 0``

> 3/31/2020, 6:03:54 PMnode: da8b81ba.a7494stat/tasmota-0138/BATT : msg.payload : string[4]"high"

Functional

> 3/31/2020, 6:03:54 PMnode: da8b81ba.a7494tele/tasmota-0138/RESULT : msg.payload : string[51]"{"TuyaReceived":{"Data":"55AA0002000001","Cmnd":2}}

Can already be disabled with ``SetOption66 0``

brjhaverkamp, 2020-04-01:
Hello Theo,

Thanks for picking this up. And even providing the fix at such short notice! Much appreciated.
Greets/Groeten uit Limburg!:-)

closed 2020-04-01 by ascillato2