arendst/Tasmota #8046: BW-LT20 blinks when turned on

stofakiller, 2020-03-31:
### PROBLEM DESCRIPTION
I just upgraded from 8.1 to 8.2 on all my devices, approx 20, but on my BW-LT20, it turns off after 2-3 seconds, and turns back on again, annoying

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
19:15:43 MQT: stat/repo/STATUS = {"Status":{"Module":18,"FriendlyName":["Repo"],"Topic":"repo","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:15:43 MQT: stat/repo/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:19:42","StartupUTC":"2020-03-31T17:56:01","Sleep":50,"CfgHolder":4617,"BootCount":16,"BCResetTime":"2020-03-31T18:23:37","SaveCount":347,"SaveAddress":"F9000"}}
19:15:43 MQT: stat/repo/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-20T14:45:23","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"355/699"}}
19:15:43 MQT: stat/repo/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["st","stofakiller"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
19:15:43 MQT: stat/repo/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":0,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
19:15:43 MQT: stat/repo/STATUS5 = {"StatusNET":{"Hostname":"repo-7064","IPAddress":"192.168.1.218","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"DC:4F:22:71:9B:98","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
19:15:43 MQT: stat/repo/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.176","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_719B98","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:15:43 MQT: stat/repo/STATUS7 = {"StatusTIM":{"UTC":"2020-03-31T18:15:43","Local":"2020-03-31T19:15:43","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:28","Sunset":"19:19"}}
19:15:43 MQT: stat/repo/STATUS10 = {"StatusSNS":{"Time":"2020-03-31T19:15:43"}}
19:15:43 MQT: stat/repo/STATUS11 = {"StatusSTS":{"Time":"2020-03-31T19:15:43","Uptime":"0T00:19:42","UptimeSec":1182,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":50,"Color":"0080","White":50,"CT":500,"Channel":[0,50],"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":2,"SSId":"stofakiller","BSSId":"44:F0:34:41:42:03","Channel":1,"RSSI":44,"Signal":-78,"LinkCount":1,"Downtime":"0T00:00:17"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
19:16:35 CMD: weblog 4
19:16:35 MQT: stat/repo/RESULT = {"WebLog":4}
19:16:36 CFG: Saved to flash at F8, Count 348, Bytes 4096

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stofakiller, 2020-03-31:
[image]

ascillato, 2020-03-31:
Please, show us what the console says when this happens. Thanks

stofakiller, 2020-04-01:
00:00:00 CFG: Loaded from flash at F8, Count 56
00:00:00 Project tasmota Toilet Version 8.2.0(tasmota)-STAGE
00:00:00 WIF: Connecting to AP2 stofakiller in mode 11N as toilet-6212...
00:00:09 WIF: Connect failed with AP timeout
00:00:09 WIF: Connecting to AP1 stofakiller in mode 11N as toilet-6212...
00:00:16 WIF: Connected
00:00:16 HTP: Web server active on toilet-6212 with IP address 192.168.1.209
07:29:02 MQT: Attempting connection...
07:29:02 MQT: Connected
07:29:02 MQT: tele/toilet/LWT = Online (retained)
07:29:02 MQT: cmnd/toilet/POWER = 
07:29:02 MQT: tele/toilet/INFO1 = {"Module":"Generic","Version":"8.2.0(tasmota)","FallbackTopic":"cmnd/DVES_719844_fb/","GroupTopic":"cmnd/tasmotas/"}
07:29:02 MQT: tele/toilet/INFO2 = {"WebServerMode":"Admin","Hostname":"toilet-6212","IPAddress":"192.168.1.209"}
07:29:02 MQT: tele/toilet/INFO3 = {"RestartReason":"External System"}
07:29:02 MQT: stat/toilet/RESULT = {"POWER":"ON"}
07:29:02 MQT: stat/toilet/POWER = ON
07:29:06 MQT: tele/toilet/STATE = {"Time":"2020-04-01T07:29:06","Uptime":"0T00:00:23","UptimeSec":23,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":51,"Color":"0082","White":51,"CT":500,"Channel":[0,51],"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"stofakiller","BSSId":"44:F0:34:41:42:03","Channel":1,"RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:17"}}

stofakiller, 2020-04-02:
What is URL for 8.1.0?
http://thehackbox.org/tasmota/release/tasmota.bin is only the latest

ascillato, 2020-04-02:
Tasmota 8.1.0 is on github release page: https://github.com/arendst/Tasmota/releases/tag/v8.1.0

But please, before that, try latest Tasmota from development branch: http://thehackbox.org/tasmota/tasmota.bin

stofakiller, 2020-04-03:
I want to go back to the firmware that worked, but when i try i get a Upload buffer miscompare. How can i downgrade Tasmota?

effelle, 2020-04-03:
If you are not flashing directly using serial you have to install a minimal version first.
Please follow the instructions on [documentation](https://tasmota.github.io/docs/Upgrading/#upgrade-by-file-upload). 
Still, I enforce what @ascillato proposed you: test the latest dev in one of your devices first. You will help us to catch the issue.

stofakiller, 2020-04-04:
Ok, i did the development

Program Version | 8.2.0.3(c066b0c-tasmota)
-- | --
2020-04-03T19:00:50
STAGE/2.2.2-dev(38a443e)

Exactly the same issue. I think it blink when connecting to wifi

stofakiller, 2020-04-04:
I put the weblog to 4.
Here is from 8.2.0.3
00:00:00 CFG: Loaded from flash at F5, Count 182
00:00:00 QPC: Flag 0E
00:00:00 CFG: CR 377/699
00:00:00 SRC: Restart
00:00:00 Project tasmota Toilet Version 8.2.0.3(c066b0c-tasmota)-STAGE
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 stofakiller in mode 11N as toilet-6212...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Attempting connection...
00:00:06 QPC: Reset
00:00:07 WIF: Checking connection...
00:00:07 WIF: Attempting connection...
00:00:08 WIF: Checking connection...
00:00:08 WIF: Attempting connection...
00:00:08 APP: Boot Count 69
00:00:08 CFG: Saved to flash at F4, Count 183, Bytes 4096
00:00:09 WIF: Checking connection...
00:00:09 WIF: Connected
00:00:09 HTP: Web server active on toilet-6212 with IP address 192.168.1.209
11:18:25 NTP: UTC 2020-04-04T10:18:24, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
11:18:25 MQT: Attempting connection...
11:18:25 MQT: Connected
11:18:25 MQT: tele/toilet/LWT = Online (retained)
11:18:25 MQT: cmnd/toilet/POWER = 
11:18:25 MQT: Subscribe to cmnd/toilet/#
11:18:25 MQT: Subscribe to cmnd/tasmotas/#
11:18:25 MQT: Subscribe to cmnd/DVES_719844_fb/#
11:18:25 MQT: tele/toilet/INFO1 = {"Module":"Generic","Version":"8.2.0.3(c066b0c-tasmota)","FallbackTopic":"cmnd/DVES_719844_fb/","GroupTopic":"cmnd/tasmotas/"}
11:18:25 MQT: tele/toilet/INFO2 = {"WebServerMode":"Admin","Hostname":"toilet-6212","IPAddress":"192.168.1.209"}
11:18:25 MQT: tele/toilet/INFO3 = {"RestartReason":"Power On"}
11:18:25 MQT: stat/toilet/RESULT = {"POWER":"ON"}
11:18:25 MQT: stat/toilet/POWER = ON
11:18:29 MQT: tele/toilet/STATE = {"Time":"2020-04-04T11:18:29","Uptime":"0T00:00:16","UptimeSec":16,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":51,"Color":"0082","White":51,"CT":500,"Channel":[0,51],"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"stofakiller","BSSId":"AC:84:C6:B5:89:20","Channel":11,"RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:10"}}
11:18:44 WIF: Checking connection...

And here is from 8.1.0
00:00:00 CFG: Loaded from flash at F5, Count 182
00:00:00 QPC: Flag 0E
00:00:00 CFG: CR 377/699
00:00:00 SRC: Restart
00:00:00 Project tasmota Toilet Version 8.2.0.3(c066b0c-tasmota)-STAGE
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 stofakiller in mode 11N as toilet-6212...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Attempting connection...
00:00:06 QPC: Reset
00:00:07 WIF: Checking connection...
00:00:07 WIF: Attempting connection...
00:00:08 WIF: Checking connection...
00:00:08 WIF: Attempting connection...
00:00:08 APP: Boot Count 69
00:00:08 CFG: Saved to flash at F4, Count 183, Bytes 4096
00:00:09 WIF: Checking connection...
00:00:09 WIF: Connected
00:00:09 HTP: Web server active on toilet-6212 with IP address 192.168.1.209
11:18:25 NTP: UTC 2020-04-04T10:18:24, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
11:18:25 MQT: Attempting connection...
11:18:25 MQT: Connected
11:18:25 MQT: tele/toilet/LWT = Online (retained)
11:18:25 MQT: cmnd/toilet/POWER = 
11:18:25 MQT: Subscribe to cmnd/toilet/#
11:18:25 MQT: Subscribe to cmnd/tasmotas/#
11:18:25 MQT: Subscribe to cmnd/DVES_719844_fb/#
11:18:25 MQT: tele/toilet/INFO1 = {"Module":"Generic","Version":"8.2.0.3(c066b0c-tasmota)","FallbackTopic":"cmnd/DVES_719844_fb/","GroupTopic":"cmnd/tasmotas/"}
11:18:25 MQT: tele/toilet/INFO2 = {"WebServerMode":"Admin","Hostname":"toilet-6212","IPAddress":"192.168.1.209"}
11:18:25 MQT: tele/toilet/INFO3 = {"RestartReason":"Power On"}
11:18:25 MQT: stat/toilet/RESULT = {"POWER":"ON"}
11:18:25 MQT: stat/toilet/POWER = ON
11:18:29 MQT: tele/toilet/STATE = {"Time":"2020-04-04T11:18:29","Uptime":"0T00:00:16","UptimeSec":16,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":51,"Color":"0082","White":51,"CT":500,"Channel":[0,51],"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"stofakiller","BSSId":"AC:84:C6:B5:89:20","Channel":11,"RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:10"}}
11:18:44 WIF: Checking connection...

Jason2866, 2020-04-06:
Please try provided firmware. It is build with active GPIO blink fix. 
Build on Arduino Core commit e64cb619f7 and latest development Tasmota
[tasmota.bin.gz](https://github.com/arendst/Tasmota/files/4437443/tasmota.bin.gz)

TimelessNL, 2020-04-06:
**+1**

v8.1.0 (sdk 2.6.3) = Normal but small dip in brightness after about 3sec.
v8.2.0 (sdk 2.6.3) = ON->OFF->ON after about 3sec.

arendst, 2020-04-06:
I have LSC bulbs. Will dig em out the trash can and have a look.

TimelessNL, 2020-04-06:
Haha trash can? You threw them away because of the noise they make?

arendst, 2020-04-06:
I had no real use for them as the wife didn't liked them. 

Just tested and LSC E14 bulb which runs fine but as it uses an SM2135 PWM controller this is expected to run OK.

The LSC E27 filament with two PWM channels indeed goes ON-OFF-ON after four seconds when a restart is performed.

I'll check what a core 2.4.2 does...

arendst, 2020-04-06:
And as expected, core 2.4.2 runs just fine.

So we need to investigate the massive changes to PWM between core versions. Nice challenge.

TimelessNL, 2020-04-06:
But Arduino Core 2.6.3 and Tasmota v8.1.0 also works fine. So it's not only core related?

arendst, 2020-04-06:
With both Core 2.6.1 and 2.6.3 and Tasmota 8.1.0 I see some dimming after four seconds. Not a complete Off but still there is something going on.

TimelessNL, 2020-04-06:
True, but wasn't that because of WiFi interupts and software PWM? 2.4.2 has no dimming issue even?

s-hadinger, 2020-04-06:
@arendst I see this dimming too with MagicHome Led driver.

The dimming is caused by 'something' interfering with the PWM interrupt handler and changing slightly timings. The PWM settings are unchanged during this period.

I guess it's caused by Wifi connection, but that's only a guess.

s-hadinger, 2020-04-06:
@TimelessNL Can you self compile a version, and uncomment in `xdrv_04_light.ino` lines 2027-2028:
```
char msg[24];
AddLog_P2(LOG_LEVEL_DEBUG, PSTR("LGT: Channels %s"), ToHex_P((const unsigned char *)cur_col_10, 10, msg, sizeof(msg)));
```

Then enable `WebLog 4` and capture the logs after a restart. We should know whether it is a PWM Interrupt issue or some command/rule/mqtt that turn them Off/On.

arendst, 2020-04-06:
Just compiled 8.2.0.3 with core feature_stage AND disabled the file ``tasmota/core_esp8266_waveform.cpp``. This solves the ON-OFF-ON issue and returns to the slightly dim I noticed before.

s-hadinger, 2020-04-06:
Wow, that's strange. Theo, would you be kind enough and uncomment line 277 `tasmota/core_esp8266_waveform.cpp` and try again. That's a safe-guard that was supposed not to be useful., but who knows.

```
          while (-cyclesToGo > wave->nextTimeHighCycles + wave->nextTimeLowCycles) { cyclesToGo += wave->nextTimeHighCycles + wave->nextTimeLowCycles)};

```

arendst, 2020-04-06:
As requested by enabling lines 2027-2028 and enabling the local waveform:

```
00:00:00 CFG: Loaded from flash at FB, Count 83
00:00:00 QPC: Reset
00:00:00 SRC: Restart
00:00:00 LGT: Channels 3B002800000000000000
00:00:00 Project tasmota SF klein Version 8.2.0.3(theo)-STAGE
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:01 WIF: Network (re)scan started...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Network 0, AP2, SSId indebuurt_IoT, Channel 1, BSSId 74:83:C2:2A:F1:AC, RSSI -84, Encryption 1
00:00:04 WIF: Network 1, AP2, SSId indebuurt_IoT, Channel 11, BSSId 18:E8:29:CA:17:C1, RSSI -38, Encryption 1
00:00:04 WIF: Connecting to AP2 indebuurt_IoT Channel 11 BSSId 18:E8:29:CA:17:C1 in mode 11N as lscsf3...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on lscsf3 with IP address 192.168.2.158
15:08:20 NTP: UTC 2020-04-06T13:08:19, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
15:08:21 MQT: Attempting connection...
15:08:21 MQT: Connected
15:08:21 MQT: tele/lscsf3/LWT = Online (retained)
15:08:21 MQT: cmnd/lscsf3/POWER = 
15:08:21 MQT: Subscribe to cmnd/lscsf3/#
15:08:21 MQT: Subscribe to cmnd/lights/#
15:08:21 MQT: Subscribe to cmnd/DVES_D55D97_fb/#
15:08:21 MQT: tele/lscsf3/INFO1 = {"Module":"LSC-Filam","Version":"8.2.0.3(theo)","FallbackTopic":"cmnd/DVES_D55D97_fb/","GroupTopic":"cmnd/lights/"}
15:08:21 MQT: tele/lscsf3/INFO2 = {"WebServerMode":"Admin","Hostname":"lscsf3","IPAddress":"192.168.2.158"}
15:08:21 MQT: tele/lscsf3/INFO3 = {"RestartReason":"Software/System restart"}
15:08:21 MQT: stat/lscsf3/RESULT = {"POWER":"ON"}
15:08:21 MQT: stat/lscsf3/POWER = ON
15:08:22 APP: Boot Count 28
15:08:22 CFG: Saved to flash at FA, Count 84, Bytes 4096
15:08:25 MQT: tele/lscsf3/STATE = {"Time":"2020-04-06T15:08:25","Uptime":"0T00:00:13","UptimeSec":13,"Heap":29,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":70,"MqttCount":1,"POWER":"ON","Dimmer":26,"Color":"422C","White":26,"CT":292,"Channel":[26,17],"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":2,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"RSSI":100,"Signal":-42,"LinkCount":1,"Downtime":"0T00:00:07"}}
15:08:39 WIF: Checkin
```
will now try your latest request.

s-hadinger, 2020-04-06:
Thanks, nothing in the logs sor the issue is definitely in the PWM interrupt handler.

arendst, 2020-04-06:
Did the change as you requested. Got compile error. Solved the code as follows:
```
//          while (-cyclesToGo > wave->nextTimeHighCycles + wave->nextTimeLowCycles) { cyclesToGo += wave->nextTimeHighCycles + wave->nextTimeLowCycles)};
          while (-cyclesToGo > wave->nextTimeHighCycles + wave->nextTimeLowCycles) { cyclesToGo += wave->nextTimeHighCycles + wave->nextTimeLowCycles; }
```
but it still does an ON-OFF-ON after a restart (in addition to the planned OFF-ON as a result of the reset off PCM's at start of restart but that has nothing to do with the ON-OFF-ON issue).

arendst, 2020-04-06:
I'll compare our/your waveform.cpp with the core one...

arendst, 2020-04-06:
So the only active difference is below.

The current core one (with slight dim):
```
        if (cyclesToGo < 0) {
          waveformState ^= mask;
          if (waveformState & mask) {
            if (i == 16) {
              GP16O |= 1; // GPIO16 write slow as it's RMW
            } else {
              SetGPIO(mask);
            }
            wave->nextServiceCycle = now + wave->nextTimeHighCycles;
            nextEventCycles = min_u32(nextEventCycles, wave->nextTimeHighCycles);
          } else {
            if (i == 16) {
              GP16O &= ~1; // GPIO16 write slow as it's RMW
            } else {
              ClearGPIO(mask);
            }
            wave->nextServiceCycle = now + wave->nextTimeLowCycles;
            nextEventCycles = min_u32(nextEventCycles, wave->nextTimeLowCycles);
          }
        } else {
          uint32_t deltaCycles = wave->nextServiceCycle - now;
          nextEventCycles = min_u32(nextEventCycles, deltaCycles);
        }
      }
```
The current tasmota with ON-OFF-ON:
```
        if (cyclesToGo < 0) {
          // See #7057
          // The following is a no-op unless we have overshot by an entire waveform cycle.
          // As modulus is an expensive operation, this code is removed for now:
          // cyclesToGo = -((-cyclesToGo) % (wave->nextTimeHighCycles + wave->nextTimeLowCycles));
          //
          // Alternative version with lower CPU impact:
          // while (-cyclesToGo > wave->nextTimeHighCycles + wave->nextTimeLowCycles) { cyclesToGo += wave->nextTimeHighCycles + wave->nextTimeLowCycles; }
          waveformState ^= mask;
          if (waveformState & mask) {
            if (i == 16) {
              GP16O |= 1; // GPIO16 write slow as it's RMW
            } else {
              SetGPIO(mask);
            }
            wave->nextServiceCycle += wave->nextTimeHighCycles;
            nextEventCycles = min_u32(nextEventCycles, max_32(wave->nextTimeHighCycles + cyclesToGo, microsecondsToClockCycles(1)));
          } else {
            if (i == 16) {
              GP16O &= ~1; // GPIO16 write slow as it's RMW
            } else {
              ClearGPIO(mask);
            }
            wave->nextServiceCycle += wave->nextTimeLowCycles;
            nextEventCycles = min_u32(nextEventCycles, max_32(wave->nextTimeLowCycles + cyclesToGo, microsecondsToClockCycles(1)));
          }
        } else {
          uint32_t deltaCycles = wave->nextServiceCycle - now;
          nextEventCycles = min_u32(nextEventCycles, deltaCycles);
        }
      }
```
Was there any reason to get rid of variable now in wave->nextServiceCycles?

arendst, 2020-04-06:
In fact #7851 seems to be culprit

arendst, 2020-04-06:
While experimenting with the core code I noticed it also does ON-OFF-ON with low dimmer values below 10% so...

s-hadinger, 2020-04-06:
I created #7851 to remove ongoing flickering due to phase changes in PWM. I'm don't understand why it would create this On-Off-On effect. Is it happening during wifi connection?

arendst, 2020-04-06:
Yes, around that time.

s-hadinger, 2020-04-07:
I was able to reproduce on MagicHome but the effect depends a lot on the color levels. For some levels it's only a slight change, for others it almost turns the light off.

My test is now: `Color BFA906` with `LedTable 1`. It is normally bright yellow, and goes down to a dark blue during less than a second.

I tried with current code, and two other variants from Arduino Core, but they all have the same visual effect
https://github.com/esp8266/Arduino/pull/7022
https://github.com/esp8266/Arduino/pull/7192

I will get my scope in the following days and try to understand what is happening.

s-hadinger, 2020-04-07:
I think I understand what's happening.

Current core: the PWM interrupt handler tries to keep each PWM level constant, and sacrifices phase between channels, which leads to slight flickering in the long run.

Current Tasmota with patch https://github.com/esp8266/Arduino/pull/7057: interrupt handlers tries to keep phase between channels and sacrifies PWM levels. Under normal circumstances it works very well, but under high interrupt pressure, as it happens at wifi connection, it gives bad results.

I will try to detect if the interrupt pressure is too high and revert to sacrificing phases during those periods. Stay tuned.

s-hadinger, 2020-04-07:
This should fix it. On my MagicHome the flickering during wifi connection is minimal (slight change in colors).

closed 2020-04-08 by arendst

dok-net, 2020-04-10:
@arendst @s-hadinger I am working on PR https://github.com/esp8266/Arduino/pull/7022 , and I am sure I have made great progress recently. My own testing can only go as far as checking the audio, watching the built-in LED, listening to servos, and taking in-detail measurements by oscilloscope.
That said, without feedback to exact commits I am fairly out on my own. I think your use case is of actual relevance, so I would like to ask you if you could checkout the "latest" commit of my PR, give it a run, and report back your results. What's VERY IMPORTANT, is that you state the commitish of the revision you are testing with! Thank you.

s-hadinger, 2020-04-22:
@TimelessNL Changing the Arduino PWM implementation in #8246 and setting `PwmFrequency 223` gives almost perfect results, even during the initial wifi connection.

PR #8094 (2020-04-08): Fix PWM flickering during wifi connection (#8046)
## Description:

Fix #7851 allowed the PWM interrupt handler to keep phases constant between channels and avoid flickering.

Unfortunatel, under heavy interrupt load, at wifi connection, the interrupt handler struggles to keep PWM at correct levels. In case of an overshoot of the PWM cycle by more than 25%, the interrupt handler reverts to the previous behavior and favors PWM level instead of phases.
This minimizes flickering at wifi connection.

**Related issue (if applicable):** fixes #8046 

## Checklist:
  - [x] The pull request is done aga [... cut]