arendst/Tasmota #8051: Cannot control device (Devola Panel Heater) at all after successful flash

EdwardJB, 2020-04-01:
### PROBLEM DESCRIPTION
After a successful flash I'm unable to control the device (a [Devola DVPW1500B Panel Heater](https://www.devola.co.uk/products/devola-designer-1-5kw-lot-20-compliant-glass-panel-heater-wifi-compatible-black-dvpw1500b)) in any way. There was no template, so I tried https://tasmota.github.io/docs/Configuration-Procedure-for-New-Devices/. But each of the three "ID Components" template toggles had no effect on my device. I also set `weblog 3` and pressed hardware buttons. The buttons work on the device itself, but I don't see anything in the console.

I also tried some speculative MCU commands, but there are no `dpId` entries in the log and other things error.

```
11:48:20 CMD: TuyaMCU 11,1
11:48:20 SRC: WebConsole from 192.168.0.18
11:48:20 CMD: Group 0, Index 1, Command "TUYAMCU", Data "11,1"
11:48:20 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): [Devola DVPW1500B Panel Heater](https://www.devola.co.uk/products/devola-designer-1-5kw-lot-20-compliant-glass-panel-heater-wifi-compatible-black-dvpw1500b)
- [x] Tasmota binary firmware version number used: 8.2.0.3(a096adb-tasmota)
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tuya Convert
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
11:30:02 CMD: Backlog Template; Module; GPIO 255
11:30:02 SRC: WebConsole from 192.168.0.18
11:30:02 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
11:30:02 SRC: Backlog
11:30:02 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
11:30:02 RSL: stat/tasmota/RESULT = {"NAME":"ID Components","GPIO":[21,22,23,24,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
11:30:02 SRC: Backlog
11:30:02 CMD: Group 0, Index 1, Command "MODULE", Data ""
11:30:02 RSL: stat/tasmota/RESULT = {"Module":{"0":"ID Components"}}
11:30:02 SRC: Backlog
11:30:02 CMD: Group 0, Index 1, Command "GPIO", Data "255"
11:30:02 RSL: stat/tasmota/RESULT = {"GPIO0":{"21":"Relay1"},"GPIO1":{"22":"Relay2"},"GPIO2":{"23":"Relay3"},"GPIO3":{"24":"Relay4"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
11:31:02 CMD: Group 0, Index 1, Command "STATUS", Data "0"
11:31:02 RSL: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4"],"Topic":"tasmota","ButtonTopic":"0","Power":9,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:31:02 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T01:09:33","StartupUTC":"2020-04-01T09:21:29","Sleep":50,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2020-04-01T10:21:37","SaveCount":25,"SaveAddress":"F9000"}}
11:31:02 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.2.0.3(a096adb-tasmota)","BuildDateTime":"2020-03-31T19:00:11","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"311/699"}}
11:31:02 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["EBW",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008008","2805C8000100060000005A00000000000000","00000000","00000000"]}}
11:31:02 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":578,"Free":424,"Heap":26,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":0,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
11:31:02 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-4155","IPAddress":"192.168.0.67","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"194.168.4.100","Mac":"EC:FA:BC:4A:10:3B","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
11:31:02 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4A103B","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:31:02 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"2020-04-01T10:31:02","Local":"2020-04-01T11:31:02","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:26","Sunset":"19:21"}}
11:31:02 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-04-01T11:31:02"}}
11:31:02 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-04-01T11:31:02","Uptime":"0T01:09:33","UptimeSec":4173,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":0,"POWER1":"ON","POWER2":"OFF","POWER3":"OFF","POWER4":"ON","Wifi":{"AP":1,"SSId":"EBW","BSSId":"48:D3:43:0C:2E:E1","Channel":6,"RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
11:08:56 WIF: Checking connection...
11:09:16 WIF: Checking connection...
11:09:36 WIF: Checking connection...
11:09:56 WIF: Checking connection...
11:10:16 WIF: Checking connection...
```

### TO REPRODUCE
Flash the same device and try controlling it.

### EXPECTED BEHAVIOUR
I expected the ID Components template to control the device.

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
Note: I'm very new to Tasmota, so might be doing something very wrong!

Jason2866, 2020-04-01:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and the [Support](https://github.com/arendst/Tasmota#support-information).

Thanks.

### Support Information
See [Documentation](http://tasmota.com/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-04-01 by ascillato2

davidgurr, 2021-01-12:
@EdwardJB did you ever get anywhere with this? I have the same heater and similar problems

EdwardJB, 2021-01-12:
> @EdwardJB did you ever get anywhere with this? I have the same heater and similar problems

@davidgurr In short, no: I changed to IR heating.

Longer story: It turns out the heater does not use the default Tuya communication protocol between the MCU and the ESP chip. I deciphered most of the protocol they do use (included below, if it helps) and was able to control the device by sending the right packets via a serial connection. Even so I gave up on the project as I couldn't get Tasmota to read the non-Tuya, raw serial messages at all. Had this been possible (though I might be and I just failed at it) I could have perhaps written Tasmota scripts to decode the packets, or sent them wholesale back to Home Assistant for processing. So the only solution would have been to write custom Tasmota drivers just for one heater, which I didn't fancy.

|Identifier |FIELD2|Power Status|FIELD4    |Power Setting|Target Temp|FIELD7|Current Temp|FIELD9 |Checksum|Terminator|
|-----------|------|------------|----------|-------------|-----------|------|------------|-------|--------|----------|
|F2F2  (MCU)|2100  |1  On       |2020019000|2  Low       |14 (20c)   |1     |17 (23c)    |1000001|60      |7E        |
|F1F1  (ESP)|      |2  Off      |          |4 Anti-Frost |           |      |            |       |        |          |

Note, the Checksum is just the sum of all previous bits in the packet.

davidgurr, 2021-01-12:
Did you manage to get it working again when you flashed the backup of the original firmware? I was expecting it to just rejoin the network with the same settings, but it doesn't.
On resetting and trying to repair, it then fails every time.

EdwardJB, 2021-01-12:
You won't see a heartbeat in any event cos its dependent on the default Tuya protocol. How are you trying to flash back the stock firmware - and what do you mean when you say "trying to repair"?

davidgurr, 2021-01-12:
I used Tasmotizer to reflash the backup that I took before I loaded Tasmota.

Sorry, that should be "re-pair", as in trying to re-pair the heater with the SmartLife app.

EdwardJB, 2021-01-12:
I seem to remember trying to flash back stock and not getting it to work (though I didn't try hard).

Even so, I still have some of these heaters that I never flashed, plus my own backup of the stock firmware, so perhaps between us we can fix it.

It might be easier if you joined the Tasmota discord. We had some back-and-forth about this heater that you also might find useful.

davidgurr, 2021-01-12:
Yep, I did join the Discord and saw your comments on there.

davidgurr, 2021-01-13:
Just tried using raw esptool.py with -fm dout and -ff 26m and it's still not booting. Slighting alarmed by seeing that esptool is reporting the flash size as 2M when Tasmotizer (which I used to take he backup and flash tasmota) only created a 1M backup ...

What size is your backup?

EdwardJB, 2021-01-13:
> Just tried using raw esptool.py with -fm dout and -ff 26m and it's still not booting. Slighting alarmed by seeing that esptool is reporting the flash size as 2M when Tasmotizer (which I used to take he backup and flash tasmota) only created a 1M backup ...
> 
> What size is your backup?

It's 1MB too. Let me know if you need me to dump an original chip.

davidgurr, 2021-01-13:
Yes please - can you dump at 2MB?

`esptool.py -p PORT read_flash 0 0x200000 flash_contents.bin`

EdwardJB, 2021-01-13:
@davidgurr I didn't have `esptool` handy so used Tasmotizer set to 2MB. [backup_20210113_181343.zip](https://github.com/arendst/Tasmota/files/5810405/backup_20210113_181343.zip)

Jason2866, 2021-01-13:
Will probably not work since every firmware is unique and device dependent

davidgurr, 2021-01-14:
It's the same spec device. But if it's keyed to something unique (like MAC address) then I see your point.

Tested it (every which way) and it still doesn't work. Possible failure reasons are:

- Original Tasmotizer backup won't re-flash because flash size is 2MB and Tasmotizer only downloaded 1MB (may not be relevant if firmware is inside 1MB)
- Full 2MB backup from @EdwardJB fails possibly if each firmware is keyed to the specific hardware, in which case there's no way to reflash from a different physical device
- Possibility that the MCU is actually fried (though I can't see how that could have happened)
- Some weirdness that Tuya deliberately do to ensure you just can't reflash with the original firmware

I'm leaning towards the last option.

In any case, it looks like my only option for recovering Smart functionality is to make it work in Tasmota. 

@EdwardJB what was the specific problem you had in reading the MCU to ESP data packet? Is it that the SerialReceived rule trigger only supports ASCII data rather than Hex?

EdwardJB, 2021-01-14:
I didn't even get that far. I just tried a number of baud rates, pin settings (TuyaRx or Serial) etc. and couldn't get any data showing in Tasmota no matter how many times I mashed buttons on the device.

But I'm a dummy when it comes to this low level stuff so might have missed something basic.

Obviously I did manage to analyse traffic from a PC so it deffo works.

davidgurr, 2021-01-14:
OK, so I've got further then. I configured Tasmota as a Generic module and then GPIO13 as SerBr Rx and GPIO15 as SerBr Tx

After rebooting, I went into the console and issued:
```
baudrate 9600
serialsend5 x
```
(You have to give SerialSend some data to send).

Then, when I hold down the power button on the heater for 3s (which you do to put the original firmware into Wifi pairing mode) I get the following on the console:
`18:02:22 RSL: tele/tasmota_B7AE5F/RESULT = {"SSerialReceived":"~"}
`
So I can definitely receive data!

Is the data packet you outlined used in both directions? Or is that just what the MCU sends to the ESP?

EdwardJB, 2021-01-14:
Cool! It's in both directions. See the Identifier column for which is which.

davidgurr, 2021-01-14:
So if I manually turn on and off the heater, I get this in the console:
`18:15:56 RSL: tele/tasmota_B7AE5F/RESULT = {"SSerialReceived":"F2F202100102020019000214000116010000015F7E"}
18:22:44 RSL: tele/tasmota_B7AE5F/RESULT = {"SSerialReceived":"F2F202100202020019000100010016010000014B7E"}
`

More digging around and pressing buttons seems to give:

- F2F20210 header
- 01 (on) 02 (off)
- 01 (childlock on) 02 (childlock off)
- 02001900 padding (presumably meaning something)
- 01 (off) 02 (low) 03 (high - I have the 2000W version of the heater) 04 (anti-frost)
- 14 (setpoint - 20 decimal) when turned on
- 0001 padding when on, 0100 padding when off
- 16 (temperature - 22 decimal) 
- 01000001 padding
- 5F/4B checksum
- 7E terminator

Also, pressing and holding the power key (used to put the original firmware in pairing mode, leaving the Wifi indicator on the heater blinking once per sec) gives:
18:45:15 RSL: tele/tasmota_B7AE5F/RESULT = {"SSerialReceived":"F2F20600067E"}

So I'm guessing we have:

- F2F2 header
- 0600 some sort of Wifi status indicator
- 06 checksum
- 7E terminator

It would be great to know what the stock firmware sends back when pairing so that we can set the Wifi indicator on the heater accordingly - when Tasmota is connected or isn't. It could even be used to reset Tasmota so that resets back to the standard AP mode.

EdwardJB, 2021-01-14:
OK, let's meet in the chat and see what we can work out.

davidgurr, 2021-01-14:
What chat?

EdwardJB, 2021-01-14:
Tasmota support #lounge. Same place you found my messages last time.