arendst/Tasmota #8052: Make Counter<x> value available on Rules

lalo-uy, 2020-04-01:
**Have you looked for this feature in other issues and in the docs?**  

Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
Tested on fw 8.2
conuter#c1 value is only available if it self is the trigger or on telperiod.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

Have Counter value as you have the sensors value, like:

ON event#queryCont DO Var1 = %counter#C1% ENDONE
 
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

ON counter#C1 DO Var1 = %value% ENDONE

Goes all the time, it could be if only trigged on C1 update,
but seems better to only assign the value on request by other trigger.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**  

Thanks in advance

ascillato2, 2020-04-09:
Hi, what is the use case of this? Why you need that?

stefanbode, 2020-04-10:
This is by far not easy to implement because the counter may change quite quick and you cannot call all rules after a change of the counter. This will overload the firmware. Anyhow I think it should be possible to set teleperiod =10 and check the counter on every teleperiod. If this is not enough maybe we can go a more complicated way and do a counter evaluation every second and call the rule engine if there is a change

lalo-uy, 2020-04-10:
I have a flowmeter sending pulses to the counter and an electrovalve in
relay1
On command event pour=1300 the relay goes on and waits for 1300 pulses
before go off

Rule1 "on event#Pour do backlog Var1 %value% + counter#c1 ; rule2 1 endon"}
Rule2 "on counter#c1>%Var1% do backlog power 0 ; Rule2 0 ; endon"}

I got it working with a 3th rule:

{"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":455,"Rules":"on
event#pour do backlog Var1 %value% ; rule2 1 endon"}
{"Rule2":"OFF","Once":"ON","StopOnError":"OFF","Free":439,"Rules":"on
counter#c1 do backlog ADD1 %value%;  power 1; rule3 1 ; rule2 0 endon"}
{"Rule3":"OFF","Once":"ON","StopOnError":"OFF","Free":454,"Rules":"on
counter#c1>%Var1% do backlog power 0 ; Rule3 0 ; endon"}

closed 2020-05-13 by stale[bot]