arendst/Tasmota #8054: Accumulated Time value for relay/output being active

DavidWicks, 2020-04-02:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

Using sonoff pow2 as an example it would be great to have  registers to display the accumulated time an output/relay has be turned on. Daily & Total.  Similar to the way the power consumption is shown for Today, Yesterday, Total.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
None
**Additional context**  
_Add any other context or screenshots about the feature request here._
No
**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-04-17:
Closing this request as nobody is working on it, and also because this feature is already implemented and enabled by default on any automation software where Tasmota connects to.

If anyone is interested on working on this, please ask to reopen or provide a Pull Request.

Thanks for sharing your ideas.

closed 2020-04-17 by ascillato2

DavidWicks, 2020-04-18:
Hi, asciallato2, 

Thanks for looking at this.  

I dont think the duplicate tag is appropriate as this request is at a hardware level, not software.

If a software command is issued and for some reason not received by the hardware, a true time cannot be calculated. I believe this could be a very helpful feature.

:)