arendst/Tasmota #8055: Support for Shelly EM

ilgrank, 2020-04-03:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
It seems to me that Shelly EM isn't currently supported, even tho it should be using the same ADE7953 chip as the Shelly 2.5

**Describe the solution you'd like**  
Add support for shelly EM?

Many thanks!

ascillato, 2020-04-03:
Hi,

As you are the one that owns that device, please, open it and post photos of the board.

ascillato, 2020-04-03:
Then, if it has an ESP8266 chip and an ADE7953, you should backup the stock firmware (as explained in the docs) and then flash Tasmota. After that, you should test the GPIO outputs in order to make a new template (as explained in the docs). These steps are explained very well on the docs, but if you need further help you can ask here or in the Tasmota Support Chat.

effelle, 2020-04-03:
@digiblur has/had one. 
He tested it with the stock firmware but I don't know if he changed the firmware.
On his [review](https://www.youtube.com/watch?v=pdo0IzpEEMI
) you can see the internals too.

digiblur, 2020-04-03:
I did flash it later as well with Tasmota. I had some issues with the little relay if I remember right but I don't use it anyways. Let me see if I can find the template.

ilgrank, 2020-04-03:
I have one indeed, but it is currently hooked up in a remote location, and so I cannot open it. I have read several times in the shelly support forum it should be using the same chip as the 2.5.

ascillato2, 2020-04-09:
@ilgrank 

Hi, any news on this? Were you able to open it and takes some photos of the board?

ilgrank, 2020-04-10:
@ascillato2 : we're in lockdown for Covid-19 here, but since there's photos of it it here https://www.youtube.com/watch?v=pdo0IzpEEMI
as @effelle had pointed out, I was thinking it was enough..

djbx83, 2020-04-12:
Photos attacched.
[image]
[image]

ilgrank, 2020-04-12:
Thanks @djbx83

ascillato, 2020-04-12:
Great! Thanks.

So, seems that the only missing is just someone that has this device to flash it and test the GPIOs and make a template for it.

digiblur, 2020-04-12:
Try this one.  Pinned it out a while back and got it working for the most part except the relay as it was acting weird from what I remember.

{"NAME":"Shelly EM","GPIO":[0,0,0,0,0,0,0,0,6,156,5,21,0],"FLAG":2,"BASE":18}

djbx83, 2020-04-13:
ShellyEM doesn't use standard pins but they are much smaller... I can't understard what cable I've to use to connect my usb-ftdi... help me!

digiblur, 2020-04-13:
Breadboard jumpers work fine.

djbx83, 2020-04-13:
Sorry digiblur, but breadbord jumpers are slightly wider, and I can't insert wire in socket

digiblur, 2020-04-13:
That's what I used to flash mine, mine were smaller than dupont male jumpers.

ilgrank, 2020-04-14:
I have seen photos of people using stripped ethernet wires in that holes

ascillato2, 2020-04-17:
Hi, Any news on this? Could you try the template from @digiblur ?

ilgrank, 2020-04-18:
@ascillato2 : we're in lockdown for Covid-19 here, and so I can't move from home and as I wrote before, the EM is in a remote location.
As far as I know, there's no way of remotely flashing Tasmota to a Shelly,but please correct me if I'm wrong.
Will test it as soon as lockdown is lifted
Thanks!

ascillato, 2020-04-18:
@ilgrank 
Hi, sorry, I forgot to tag. That message was intended to @djbx83

@djbx83 
Hi, Any news on this? Could you try the template from @digiblur ?

djbx83, 2020-04-18:
I've used a small piece of ethernet cable @ascillato2 @ascillato as suggested by @ilgrank 
But..... unfortunatly I've bricked shelly and ftdi programmer... ooops!

ilgrank, 2020-04-19:
@djbx83 : you tried to  flash it while it was mains powered?

djbx83, 2020-04-19:
Yes @ilgrank .... LOL

ilgrank, 2020-04-20:
@djbx83 whoops :(
You're lucky you didn't fry your USB ports then..

ascillato2, 2020-05-06:
@djbx83 

> unfortunatly I've bricked shelly and ftdi programmer

Hi,

Your device is burnt? you couldn't reflash it? The only way to brick an ESP8266 is by applying more than 3.3V to VCC and GND pins.

Hope you weren't flashing with AC connected.

ilgrank, 2020-05-07:
@ascillato2 : yes, he was flashing with mains connected, that's what he replied me on his last message

Frigorio, 2020-05-23:
> 
> 
> Try this one. Pinned it out a while back and got it working for the most part except the relay as it was acting weird from what I remember.
> 
> {"NAME":"Shelly EM","GPIO":[0,0,0,0,0,0,0,0,6,156,5,21,0],"FLAG":2,"BASE":18}

Mmmm no, It doesn't look good to me...
[image]

Frigorio, 2020-05-23:
Ok, changed analog to user (15) and now **if the relay is on** the meter is working...

[image]

ascillato, 2020-05-23:
@Frigorio 

Great!!! Thank you very much for testing and sharing the template.

Can I ask you one little favour? Can you please add this template to the Template repository? https://templates.blakadder.com/new.html

Thanks.

Frigorio, 2020-05-24:
> Ok, changed analog to user (15) and now **if the relay is on** the meter is working...

Stupid me... SetOption21 was OFF 🤣 

> Can I ask you one little favour? Can you please add this template to the Template repository?

Sure!

ilgrank, 2020-05-27:
@Frigorio 
 
> > Ok, changed analog to user (15) and now **if the relay is on** the meter is working...
> Stupid me... SetOption21 was OFF 🤣

So now the meter is working anyways (even if relay off), right?
Also, to fix the temperature misreading you used a temperature offset?
Thanks!

Frigorio, 2020-05-27:
> So now the meter is working anyways (even if relay off), right?

Yes, meter is working.

> Also, to fix the temperature misreading you used a temperature offset?

Nope, to enable the template to work I had to set <analog> to User (15) so no temperature is read...

ilgrank, 2020-05-27:
@ascillato so while the basic functionality is there, there's still some data missing

ascillato2, 2020-06-18:
https://templates.blakadder.com/shelly_EM.html

closed 2020-06-18 by ascillato2

ilgrank, 2020-06-19:
@ascillato2 : so the issue @Frigorio was having (no temperature read) has been fixed?

Frigorio, 2020-06-19:
Nope. I had to change ADC0 to User (15) otherwise the relay "was acting
weird" as @digiblur <https://github.com/digiblur> said.

mugnier102, 2020-06-20:
Do you have 2 channel ? I flashed mine and has got only 1 measure. Is it normal ?
Does that mean only channel 1 or both channel ?
Could someone change comment on template to indicate ADC0 to User(15) and SetOption21 ?

Frigorio, 2020-06-20:
> Do you have 2 channel ? I flashed mine and has got only 1 measure. Is it normal ?
I only have 1 channel connected, but as you can see in my screenshot the measure displayed is dual...

ilgrank, 2020-06-20:
@Frigorio : and so you're lacking temperature monitoring, right?

Sonnenberg, 2020-07-17:
Hello

I also got a ShellyEM and flashed with Tasmota.
so far so good and everything went according to plan

I connected the Shelly but I just don't get any values. does anyone know why?

I took the template as described here
[image]
[image]
[image]
[image]
[image]

ilgrank, 2020-07-17:
@Sonnenberg this issue is closed, you better ask @ascillato to re-open it, since the current template seem to
a) not be working correctly (alas, I can't test it until I get to my EM location)
b) missing temperature data anyways.

Frigorio, 2020-07-17:
@Sonnenberg You have to set command "SetOption21 1" to enable energy
monitoring when power is off.

Sonnenberg, 2020-07-17:
hi i'm not sure i had tried it with setoption21 on. I will test it again later and then report

ilgrank, 2020-07-17:
@Frigorio: but still lacking temperature, right?

Sonnenberg, 2020-07-17:
no, according to my memory, the temperature shows 152 degrees. however, this is switched off with ADC0 user (15). it makes no difference whether the relay is switched on or off

ilgrank, 2020-07-17:
@Sonnenberg it's a safety measure, you should know if your equipment is overheating :)

Sonnenberg, 2020-07-17:
I switched SetOption21 to 1.
with this template:
{"NAME": "Shelly EM", "GPIO": [0,0,0,0,0,0,0,0,6,156,5,21,0], "FLAG": 15, "BASE": 18}

[image]
[image]

with this template: {"NAME": "Shelly EM", "GPIO": [0,0,0,0,0,0,0,0,6,156,5,21,0], "FLAG": 2, "BASE": 18}
I have the following result.

[image]

I tried both templates with SetOption21 1, but I still don't get any values in watts & volts.
I do not understand this because a post with the same Tasmota version gets at least the values displayed

seamus65, 2020-07-19:
Are you getting no reading because you are clamped over the whole cable and not just one tail?
The live and neutral will cancel each other out if you put the clamp iver both.

Sonnenberg, 2020-07-19:
Yes of course
you are absolutely right, I could have thought of that on my own;)

[image]

sometimes you can't see the forest for the trees ^^
okay that works
now we only have the "problem" with the relay

best regards and thanks so far for your help =)
@ilgrank , do you have the right template at work?

Frigorio, 2020-07-19:
@Sonnenberg which "problem" do you have with the switch?

Sonnenberg, 2020-07-19:
I have now turned "SetOption21" off
the relay is working properly

but the temperature is still missing,
if this is switched on, the relay switches off again immediately

the temperature problem has already been mentioned above.

arendst, 2020-07-19:
Guys, if the temperature switches the relay off it means it either detects and overtemp or the tmeperature calculation is wrong.

In this case it's obvious the calculation is wrong. You'll have to change ADCPARAMs according to the installed NTC resistor bridge.

Sonnenberg, 2020-07-19:
exactly, the temperature is so high
that the Shelly or Tasmota turns off the device
a security function that is super but I'm a little afraid to destroy the Shelly by testing it.

I have installed Tasmota on all my devices and I am very satisfied with it, but I was probably a bit hasty with the ShellyEM, because I probably can't use Tasmota as I would have liked

zbuh, 2020-07-26:
I've flashed with Tasmota 8.3.1, loaded the template and SetOption21 to On.
But I can only see the relay. No power information. Any clue of what may I'm doing wrong?

zbuh, 2020-07-26:
a new erase and flash fixed!

ilgrank, 2020-07-27:
zbuh: meaning you also see temperature?

zbuh, 2020-07-27:
no. no temperature

mugnier102, 2020-07-29:
@Sonneberg:
You have missed the most important physical part.
YOu must have only 1 wire in the shelf. You're picture show that you put red and blue wire in the shelf.
 Move your shelf to have only the red wire (or the blue if you prefer...).
Putting both make the blue one to cancel the red one..... and so the measure is 0.

Frigorio, 2021-03-31:
@ilgrank wrote:
> so the issue @Frigorio was having (no temperature read) has been fixed?
> @Frigorio : and so you're lacking temperature monitoring, right?

@arendst wrote:
> Guys, if the temperature switches the relay off it means it either detects and overtemp or the tmeperature calculation is wrong.
> In this case it's obvious the calculation is wrong. You'll have to change ADCPARAMs according to the installed NTC resistor bridge.

I can confirm that ADCPARAMs are wrong, I asked Dimitar Dimitrov for ShellyEM's NTC parameters and I'm waiting for a feedback ([permalink](https://www.facebook.com/groups/ShellyIoTCommunitySupport/permalink/3601666286599376)).

Frigorio, 2021-03-31:
> I can confirm that ADCPARAMs are wrong, I asked Dimitar Dimitrov for ShellyEM's NTC parameters and I'm waiting for a feedback ([permalink](https://www.facebook.com/groups/ShellyIoTCommunitySupport/permalink/3601666286599376)).
OK. Dimitar Dimitrov said that _"Shelly EM do not have NTC"_
So, if Shelly EM has no temperature sensor, what the hack is A0 used for?

kaciker, 2021-05-26:
Hello, I just flashed Tasmota to a EM that worked during 2 years with Shelly firm. After check all templates that I see, the device show only one chanel (I had fisically two) and the power showed is wrong, less than real.

I tested changing septoption21 and no changes in 0/1

Can someone help me ?

This is with https://templates.blakadder.com/shelly_EM.html

[image]

[image]

[4 more replies]