arendst/Tasmota #8057: BH1750 setting sensitivity bigger than the 69 default maxes lux reading

lhurt, 2020-04-03:
### PROBLEM DESCRIPTION
Setting the measurement time higher than 69 maxes the lux output.
The calculation doesn't take the factors required for other reading modes than BH1750_CONTINUOUS_HIGH_RES_MODE into account.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: 8.2.0.3
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: platformIO
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
18:43:12 MQT: openhab/gang-sideboard-led/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}
18:43:12 MQT: openhab/gang-sideboard-led/stat/RESULT = {"Module":{"20":"H801"}}
18:43:12 MQT: openhab/gang-sideboard-led/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"52":"Led1"},"GPIO2":{"6":"I2C SDA"},"GPIO3":{"5":"I2C SCL"},"GPIO4":{"41":"PWM5"},"GPIO5":{"57":"Led2i"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"39":"PWM3"},"GPIO13":{"38":"PWM2"},"GPIO14":{"40":"PWM4"},"GPIO15":{"37":"PWM1"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
18:44:15 MQT: openhab/gang-sideboard-led/stat/STATUS = {"Status":{"Module":20,"FriendlyName":["gang-sideboard-led"],"Topic":"gang-sideboard-led","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:44:15 MQT: openhab/gang-sideboard-led/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T08:53:07","StartupUTC":"2020-04-03T08:51:08","Sleep":50,"CfgHolder":1709,"BootCount":9,"BCResetTime":"2020-04-03T09:33:40","SaveCount":29,"SaveAddress":"3FA000"}}
18:44:15 MQT: openhab/gang-sideboard-led/stat/STATUS2 = {"StatusFWR":{"Version":"8.2.0.3(sensors)","BuildDateTime":"2020-04-02T19:34:55","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"400/699"}}
18:44:15 MQT: openhab/gang-sideboard-led/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":1,"LogHost":"192.168.1.250","LogPort":514,"SSId":["domum nostrae",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005AFF000000000000","00000020","00000000"]}}
18:44:15 MQT: openhab/gang-sideboard-led/stat/STATUS4 = {"StatusMEM":{"ProgramSize":602,"Free":2468,"Heap":19,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FDAA387","0415A3A5","B7FFBFCD","01DA9BC4","24367CC7","00000002"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64"}}
18:44:15 MQT: openhab/gang-sideboard-led/stat/STATUS5 = {"StatusNET":{"Hostname":"gang-dimmer","IPAddress":"192.168.1.192","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.250","Mac":"5C:CF:7F:B9:49:19","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
18:44:15 MQT: openhab/gang-sideboard-led/stat/STATUS6 = {"StatusMQT":{"MqttHost":"automation.hurt.local","MqttPort":1883,"MqttClientMask":"gang-sideboard-led","MqttClient":"gang-sideboard-led","MqttUser":"gang-sideboard-led","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
18:44:15 MQT: openhab/gang-sideboard-led/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-04-03T17:44:15","Local":"2020-04-03T18:44:15","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:22","Sunset":"19:24"}}
18:44:15 MQT: openhab/gang-sideboard-led/stat/STATUS10 = {"StatusSNS":{"Time":"2020-04-03T18:44:15","BH1750":{"Illuminance":0}}}
18:44:15 MQT: openhab/gang-sideboard-led/stat/STATUS11 = {"StatusSTS":{"Time":"2020-04-03T18:44:15","Uptime":"0T08:53:07","UptimeSec":31987,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":0,"Color":"0000000000","HSBColor":"0,100,0","White":0,"CT":153,"Channel":[0,0,0,0,0],"Scheme":0,"Fade":"ON","Speed":5,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"domum nostrae","BSSId":"44:4E:6D:61:79:E0","Channel":6,"RSSI":100,"Signal":-37,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
18:48:46 MQT: openhab/gang-sideboard-led/tele/SENSOR = {"Time":"2020-04-03T18:48:46","BH1750":{"Illuminance":0}}
18:49:19 CMD: Sensor10 
18:49:19 MQT: openhab/gang-sideboard-led/stat/RESULT = {"Sensor10":{"Resolution":0,"MTime":69}}
18:49:34 CMD: Sensor10 70
18:49:34 MQT: openhab/gang-sideboard-led/stat/RESULT = {"Sensor10":{"Resolution":0,"MTime":70}}
18:49:46 MQT: openhab/gang-sideboard-led/tele/STATE = {"Time":"2020-04-03T18:49:46","Uptime":"0T08:58:38","UptimeSec":32318,"Heap":22,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":75,"Color":"BF00000000","HSBColor":"0,100,75","White":0,"CT":153,"Channel":[75,0,0,0,0],"Scheme":0,"Fade":"ON","Speed":5,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"domum nostrae","BSSId":"44:4E:6D:61:79:E0","Channel":6,"RSSI":100,"Signal":-38,"LinkCount":1,"Downtime":"0T00:00:06"}}
18:49:46 MQT: openhab/gang-sideboard-led/tele/SENSOR = {"Time":"2020-04-03T18:49:46","BH1750":{"Illuminance":65535}}
18:50:15 CMD: Sensor10 69
18:50:15 MQT: openhab/gang-sideboard-led/stat/RESULT = {"Sensor10":{"Resolution":0,"MTime":69}}
18:50:46 MQT: openhab/gang-sideboard-led/tele/STATE = {"Time":"2020-04-03T18:50:46","Uptime":"0T08:59:38","UptimeSec":32378,"Heap":22,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":75,"Color":"BF00000000","HSBColor":"0,100,75","White":0,"CT":153,"Channel":[75,0,0,0,0],"Scheme":0,"Fade":"ON","Speed":5,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"domum nostrae","BSSId":"44:4E:6D:61:79:E0","Channel":6,"RSSI":100,"Signal":-41,"LinkCount":1,"Downtime":"0T00:00:06"}}
18:50:46 MQT: openhab/gang-sideboard-led/tele/SENSOR = {"Time":"2020-04-03T18:50:46","BH1750":{"Illuminance":0}}
18:50:50 CMD: Sensor10 1
18:50:50 MQT: openhab/gang-sideboard-led/stat/RESULT = {"Sensor10":{"Resolution":1,"MTime":69}}
18:51:46 MQT: openhab/gang-sideboard-led/tele/STATE = {"Time":"2020-04-03T18:51:46","Uptime":"0T09:00:38","UptimeSec":32438,"Heap":22,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":75,"Color":"BF00000000","HSBColor":"0,100,75","White":0,"CT":153,"Channel":[75,0,0,0,0],"Scheme":0,"Fade":"ON","Speed":5,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"domum nostrae","BSSId":"44:4E:6D:61:79:E0","Channel":6,"RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:06"}}
18:51:46 MQT: openhab/gang-sideboard-led/tele/SENSOR = {"Time":"2020-04-03T18:51:46","BH1750":{"Illuminance":0}}
18:52:03 CMD: Sensor10 70
18:52:03 MQT: openhab/gang-sideboard-led/stat/RESULT = {"Sensor10":{"Resolution":1,"MTime":70}}
18:52:46 MQT: openhab/gang-sideboard-led/tele/STATE = {"Time":"2020-04-03T18:52:46","Uptime":"0T09:01:38","UptimeSec":32498,"Heap":22,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":75,"Color":"BF00000000","HSBColor":"0,100,75","White":0,"CT":153,"Channel":[75,0,0,0,0],"Scheme":0,"Fade":"ON","Speed":5,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"domum nostrae","BSSId":"44:4E:6D:61:79:E0","Channel":6,"RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:06"}}
[... cut ...]
In the test the room was dark and the reading of 0 expected. As the light situation didn't change throughout the test a value of 0 or near 0 would have been expected

### SCREENSHOTS
not applicable

### ADDITIONAL CONTEXT
The software is on a H801 light controller upgraded to 4MB flash. The BH1750 is hooked up to the I2C bus additionally.

May be it's a matter of the lux calculation that btw definately doesn't take the different factors into account when using modes other than the default BH1750_CONTINUOUS_HIGH_RES_MODE mode.

ascillato, 2020-04-03:
See https://github.com/arendst/Tasmota/issues/8016

lhurt, 2020-04-04:
@ascillato 
Thanks for the hint. Unfortunately I didn't find an explanation in #8016. May be I just don't get it.

_One more example_
1. MTReg at 69 (default
2. MTReg at 37
3. MTReg at 140
```09:31:06 MQT: openhab/gang-sideboard-led/stat/RESULT = {"Sensor10":{"Resolution":0,"MTime":69}}
09:31:11 MQT: openhab/gang-sideboard-led/tele/STATE = {"Time":"2020-04-04T09:31:11","Uptime":"0T00:02:12","UptimeSec":132,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":0,"Color":"0000000000","HSBColor":"0,100,0","White":0,"CT":153,"Channel":[0,0,0,0,0],"Scheme":0,"Fade":"ON","Speed":5,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"domum nostrae","BSSId":"44:4E:6D:61:79:E0","Channel":6,"RSSI":100,"Signal":-32,"LinkCount":1,"Downtime":"0T00:00:06"}}
09:31:11 MQT: openhab/gang-sideboard-led/tele/SENSOR = {"Time":"2020-04-04T09:31:11","BH1750":{"Illuminance":35}}
09:31:17 CMD: Sensor10 35
09:31:17 MQT: openhab/gang-sideboard-led/stat/RESULT = {"Sensor10":{"Resolution":0,"MTime":35}}
09:32:11 MQT: openhab/gang-sideboard-led/tele/STATE = {"Time":"2020-04-04T09:32:11","Uptime":"0T00:03:12","UptimeSec":192,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":1,"POWER":"OFF","Dimmer":0,"Color":"0000000000","HSBColor":"0,100,0","White":0,"CT":153,"Channel":[0,0,0,0,0],"Scheme":0,"Fade":"ON","Speed":5,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"domum nostrae","BSSId":"44:4E:6D:61:79:E0","Channel":6,"RSSI":100,"Signal":-32,"LinkCount":1,"Downtime":"0T00:00:06"}}
09:32:11 MQT: openhab/gang-sideboard-led/tele/SENSOR = {"Time":"2020-04-04T09:32:11","BH1750":{"Illuminance":17}}
09:49:37 CMD: Sensor10 140
09:49:37 MQT: openhab/gang-sideboard-led/stat/RESULT = {"Sensor10":{"Resolution":0,"MTime":140}}
09:50:27 MQT: openhab/gang-sideboard-led/tele/STATE = {"Time":"2020-04-04T09:50:27","Uptime":"0T00:01:12","UptimeSec":72,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":0,"Color":"0000000000","HSBColor":"0,100,0","White":0,"CT":153,"Channel":[0,0,0,0,0],"Scheme":0,"Fade":"ON","Speed":5,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"domum nostrae","BSSId":"44:4E:6D:61:79:E0","Channel":6,"RSSI":100,"Signal":-36,"LinkCount":1,"Downtime":"0T00:00:06"}}
09:50:27 MQT: openhab/gang-sideboard-led/tele/SENSOR = {"Time":"2020-04-04T09:50:27","BH1750":{"Illuminance":65535}}
```
Lux calculation for Mode 0:
raw_reading/ (1.2 * (69/MTReg))
 --> **raw_value** from BH1750 during the following test is around **42**

With the default settings (Mode 0, MTreg: 69) I get a reading of 35lux.
Internal calculation:
42/( 1.2 * 69/69 )

After halving the MTReg value to 35 the reading says 17 as expected.
Internal calculation
42/( 1.2 * 69/35 ) --> 17

Doubling the MTReg value to 140 should produce a reading of 70lux
42/( 1.2 * 69/140 ) --> 70

But it reads 65535 lx

And I don't know why this happens. May be I just don't see the reason, but I didn't find answers in #8016 about that.

I'd be glad to receive a hint.

ascillato2, 2020-04-05:
Hi, please update to latest Tasmota version from development branch and test it. Theo has pushed a fix for this. Thanks

lhurt, 2020-04-05:
Hi @ascillato2,
thanks for the information. That seemed to do the trick.

Closing issue.
Have a nice sunday and stay healthy.

closed 2020-04-05 by lhurt