arendst/Tasmota #8059: Shutter command to update the position value based on external remote control

he-so, 2020-04-03:
**Have you looked for this feature in other issues and in the docs?**  
yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
I am using the the shutter driver with USE_SHUTTER enabled. When changing the the relay status, the shutter position value is updated accordingly. In my situation, I run a RF controled shutter controled by tasmota with the keeloq driver. When the shutter driver initiates a movement (ralay up, down, stop has changed), I use a Rule to send the corresponding keeloq code. The problem is: Currently, in tasmota, the shutter movement is not tracked and the position value is not updated when controling the shutter **with the external remote control**.

**Describe the solution you'd like**  
I'd like to have the tasmota firmware to receive the RF commands sent by the original external remote control and update the tasmota shutter position value accordingly. For receiving the RF commands, I have code to be implemented in the keeloq driver. This is fine. The received codes (up, down, stop) can trigger a Rule. But, I how can I use these events to update the shutter position value and prevent tasmota from sending the same command again? When just updating the relay outputs (code up received -> relay up on), the Rule will trigger again and send keeloq codes again, which will lead to problems.

My idea and a possible solution would be, to have a kind of dummy command, that will update the shutter position value, but not trigger any relay outputs.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

Maybe variables can be used, to suppress following commands, when a remote control code was received. But this sounds like a hack.

**Additional context**  
_Add any other context or screenshots about the feature request here._

stefanbode, 2020-04-10:
Hi, I’m not sure why this does not work. Whenever you change the shutter through a relay it whatever the current version reports the correct shutter position during and after the movement. Can you please send in a logfile with loglevel 4 when you execute the rf command. I want to see how and if the action is received by the shutter driver.

stefanbode, 2020-04-10:
Regarding the double send of commands: what happens if you delete all rules. Does the rf directly switch on Relay1 or relay2?  If so, i just have see why the shutter driver does not get the information. If you want to have a rule controlling the shutter you must ensure that the rf does not trigger the relays on top. This anyhow is my recommended solution. Try never to trigger relays, but send shutter up/down/stop to the shutter driver.

he-so, 2020-04-10:
> Try never to trigger relays, but send shutter up/down/stop to the shutter driver.

Exactly, my current solution Implements a dependency like this:

Shutter command -> toggles relay -> sends rf commands to physical shutter.

So, when I use the received commands from the external physical remote to update the shutter driver position value (while the physical shutter has already started to move), the shutter driver will toggle the dummy relay and the above mentioned rule cascade will also trigger sending the rf commands again, which is superfluous an should not happen.

stefanbode, 2020-04-10:
Ok this will never wok because you send the rf signal directly To the shutter. This is a sidath and will always create loops and incorrect positions as unexpected behavior. This is the same like people operate shutter with the old buttons and wonder about he result. I will not support his. If you find a workaround for you that’s fine. All other users have to go the normal way

stefanbode, 2020-04-10:
@arendst . Please close.

closed 2020-04-10 by arendst