arendst/Tasmota #8063: shutter position is always 100 in MQTT

ihavenonick, 2020-04-04:
### PROBLEM DESCRIPTION
_shutter position is always 100_

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
11:40:14 MQT: stat/tasmota/RESULT = {"NAME":"Shelly 2.5","GPIO":[56,255,17,255,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18}
11:40:14 MQT: stat/tasmota/RESULT = {"Module":{"0":"Shelly 2.5"}}
11:40:15 MQT: stat/tasmota/RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"0":"None"},"GPIO2":{"17":"Button1"},"GPIO3":{"0":"None"},"GPIO4":{"21":"Relay1"},"GPIO5":{"83":"Switch2n"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"6":"I2C SDA"},"GPIO13":{"82":"Switch1n"},"GPIO14":{"5":"I2C SCL"},"GPIO15":{"22":"Relay2"},"GPIO16":{"156":"ADE7953 IRQ"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
11:40:57 MQT: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["Bad","Tasmota2"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[3,3,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:40:57 MQT: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T20:16:55","StartupUTC":"2020-04-03T13:24:02","Sleep":100,"CfgHolder":4617,"BootCount":3,"BCResetTime":"2020-03-17T15:37:56","SaveCount":222,"SaveAddress":"1F6000"}}
11:40:57 MQT: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.2.0.3(tasmota)","BuildDateTime":"2020.04.01 18:17:38","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"395/699"}}
11:40:57 MQT: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FRITZ!","FRITZ!"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C8000100060000005A00000000000000","40000001","00000000"]}}
11:40:57 MQT: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":484,"Free":1540,"Heap":25,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000407","0F002196","040003A1","00000004","010000C0","00000900","00000000"],"Drivers":"1,2,3,4,9,10,27,29","Sensors":"2,3"}}
11:40:57 MQT: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"Rollladen-Bad","IPAddress":"192.168.100.2","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.1","Mac":"80:7D:3A:74:58:46","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
11:40:57 MQT: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.100.10","MqttPort":1883,"MqttClientMask":"Rollladen-Bad","MqttClient":"Rollladen-Bad","MqttUser":"Jan","MqttCount":4,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:40:57 MQT: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"2020-04-04T09:40:57","Local":"2020-04-04T11:40:57","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"06:54","Sunset":"20:00"}}
11:40:57 MQT: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-04-04T11:40:57","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":53.4},"ENERGY":{"TotalStartTime":"2020-03-14T14:33:24","Total":0.026,"Yesterday":0.001,"Today":0.001,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":0,"Voltage":0,"Current":[0.000,0.000]},"Shutter1":{"Position":80,"Direction":0,"Target":80},"TempUnit":"C"}}
11:40:57 MQT: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-04-04T11:40:57","Uptime":"0T20:16:55","UptimeSec":73015,"Heap":25,"SleepMode":"Dynamic","Sleep":100,"LoadAvg":12,"MqttCount":4,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":2,"SSId":"FRITZ!","BSSId":"44:4E:6D:58:11:01","Channel":12,"RSSI":64,"Signal":-68,"LinkCount":2,"Downtime":"0T00:00:11"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
11:42:13 ADE: 6109828, 4473, [1787, 33, 650, -25], [1773, 1, 645, -4]
11:42:13 ADE: U 6109828, C 4473, I 0 + 0 = 0, P 0 + 0 = 0
11:42:14 ADE: 6109637, 4472, [1787, -35, 650, 17], [1773, -27, 645, -17]
11:42:14 ADE: U 6109637, C 4472, I 0 + 0 = 0, P 0 + 0 = 0
11:42:15 ADE: 6111076, 4473, [1787, -58, 650, -44], [1773, 1, 645, 11]
11:42:15 ADE: U 6111076, C 4473, I 0 + 0 = 0, P 0 + 0 = 0
11:42:16 ADE: 6110290, 4473, [1778, 16, 647, 11], [1773, -16, 645, 13]
11:42:16 ADE: U 6110290, C 4473, I 0 + 0 = 0, P 0 + 0 = 0
11:42:17 ADE: 6112515, 4472, [1778, -70, 647, 8], [1773, 7, 645, -10]
11:42:17 ADE: U 6112515, C 4472, I 0 + 0 = 0, P 0 + 0 = 0
11:42:18 ADE: 6113602, 4473, [1778, 9, 647, -11], [1773, 5, 645, -14]
11:42:18 ADE: U 6113602, C 4473, I 0 + 0 = 0, P 0 + 0 = 0
11:42:19 ADE: 6112521, 4474, [1778, -90, 647, 60], [1773, 6, 645, 18]
11:42:19 ADE: U 6112521, C 4474, I 0 + 0 = 0, P 0 + 0 = 0
11:42:20 ADE: 6123909, 4473, [1778, 60, 648, 53], [1773, 1, 646, -7]
11:42:20 ADE: U 6123909, C 4473, I 0 + 0 = 0, P 0 + 0 = 0
11:42:21 ADE: 6125896, 4474, [1778, -32, 648, 38], [1773, -5, 646, 0]
11:42:21 ADE: U 6125896, C 4474, I 0 + 0 = 0, P 0 + 0 = 0
11:42:22 ADE: 6126532, 4474, [1778, -11, 648, 50], [1773, -17, 647, 23]
11:42:22 ADE: U 6126532, C 4474, I 0 + 0 = 0, P 0 + 0 = 0
11:42:23 ADE: 6126080, 4473, [1778, 43, 648, 59], [1773, 7, 647, 4]
[... 132 lines cut ...]

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
[image]
11:46:43 MQT: stat/tasmota/RESULT = {"POWER2":"ON"}
11:46:43 MQT: stat/tasmota/POWER2 = ON
11:46:43 MQT: stat/tasmota/RESULT = {"ShutterPosition1":79}
11:46:43 SHT: Shutter1 Real 27600, Start 28000, Stop 22120, Dir -1, Delay 0, Rtc 0.20 [s], Freq 0
11:46:43 MQT: stat/tasmota/RESULT = {"Shutter1":{"Position":99,"Direction":-1,"Target":79}}
11:46:44 SHT: Shutter1 Real 25400, Start 28000, Stop 22120, Dir -1, Delay 0, Rtc 1.30 [s], Freq 0
11:46:44 MQT: stat/tasmota/RESULT = {"Shutter1":{"Position":91,"Direction":-1,"Target":79}}
11:46:45 SHT: Shutter1 Real 23500, Start 28000, Stop 22120, Dir -1, Delay 0, Rtc 2.25 [s], Freq 0
11:46:45 MQT: stat/tasmota/RESULT = {"Shutter1":{"Position":84,"Direction":-1,"Target":79}}
11:46:46 MQT: stat/tasmota/RESULT = {"POWER2":"OFF"}
11:46:46 MQT: stat/tasmota/POWER2 = OFF
11:46:46 SHT: Shutter1 Real 21900, Start 28000, Stop 22120, Dir -1, Delay 0, Rtc 3.05 [s], Freq 0
11:46:46 MQT: stat/tasmota/SHUTTER1 = 78
11:46:46 MQT: stat/tasmota/RESULT = {"Shutter1":{"Position":78,"Direction":0,"Target":79}}

[image]

**(Please, remember to close the issue when the problem has been addressed)**

schrej-zz, 2020-04-04:
No,
payload SHUTTER1 and RESULT both show other positions than 100 in your log.

ascillato2, 2020-04-04:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-04-04 by ascillato2

ihavenonick, 2020-04-04:
I checked it again with other shutters and it is not a Tasmota problem. The other shutters are OK.
I therefore close the issues.