arendst/Tasmota #8069: Discovery on tasmota for homeassistant with different type

ricain59, 2020-04-04:
**Have you looked for this feature in other issues and in the docs?**  
yes

**Is your feature request related to a problem? Please describe.**  
no

**Describe the solution you'd like**  
I view it's possible all power as light or switch, but i search possibility have diferente type.

**Describe alternatives you've considered**  
It's possible on mode discovery have power1 as light and power2 and 3 as switch?

**Additional context**  
Discovery for home assistant

**(Please, remember to close the issue when the problem has been addressed)**

ricain59, 2020-04-04:
I have receive response on first topic.

closed 2020-04-04 by ricain59

ascillato, 2020-04-04:
Already supported. See commands in the docs. Thanks.