arendst/Tasmota #8073: Fade setting ignored when mains power is applied

TimelessNL, 2020-04-04:
### PROBLEM DESCRIPTION
When mains power is applied, `FADE 1` and `SPEED 15` settings are ignored and instead turns on at dimmer level brightness immediately.
### REQUESTED INFORMATION
- ✅ Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- ✅ Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- ✅ Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- ✅ Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- ✅ Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- ✅ Device used (e.g., Sonoff Basic): Any device with at least 1 PWM channel
- ✅ Tasmota binary firmware version number used: 8.2.0
  - ⛔️ Pre-compiled
  - ✅ Self-compiled
    - ✅ IDE / Compiler used: ArduinoIDE SDK 2.6.3
- ✅ Flashing tools used: Web OTA
- ✅ Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
23:51:33 CMD: Backlog Template; Module; GPIO 255
23:51:33 RSL: RESULT = {"NAME":"LSC RGBW LED","GPIO":[0,255,0,255,0,0,0,0,181,0,180,0,0],"FLAG":15,"BASE":18}
23:51:34 RSL: RESULT = {"Module":{"0":"LSC RGBW LED"}}
23:51:34 RSL: RESULT = {"GPIO0":{"0":"Geen"},"GPIO1":{"0":"Geen"},"GPIO2":{"0":"Geen"},"GPIO3":{"0":"Geen"},"GPIO4":{"0":"Geen"},"GPIO5":{"0":"Geen"},"GPIO9":{"0":"Geen"},"GPIO10":{"0":"Geen"},"GPIO12":{"181":"SM2135 Dat"},"GPIO13":{"0":"Geen"},"GPIO14":{"180":"SM2135 Clk"},"GPIO15":{"0":"Geen"},"GPIO16":{"0":"Geen"}}
```
- ⛔️ If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
- ⛔️ Provide the output of this command: `Status 0`:
- ⛔️ Provide the output of the Console log output when you experience your issue; if applicable:

### TO REPRODUCE
1. Flash Tasmota 8.2.0
2. Command `reset 5`
3. Configure device to have atleast 1 PWM channel.
4. Command `backlog fade 1; speed 15; restart 1;`
5. Unplug power.
6. Apply power.
7. See that it turns on at configured brightness without fading.

### EXPECTED BEHAVIOUR
Just like `FADE` is used during soft ON/OFF commands, I expect the device to 'wakeup' and uses `FADE` when power is applied. This worked in v8.1.0.

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
N/A

s-hadinger, 2020-04-05:
This is on purpose, Fade does not work well when you restart the Tasmota device. The fading would start then freeze after a fraction of second (wifi scan?) and restarts again. So by default it is disabled to avoid annoying behavior.

You can simulate this with rules, and start the fading right after wifi scan. To see what it looks like you can try `PowerOnState 0` and `Rule1 ON Power1#Boot DO Power On ENDON` and `Rule1 1`. Then you can adjust as you wish.

TimelessNL, 2020-04-05:
Ah ok, maybe I've missed it but it would have been nice to read it somewhere on the WIKI or Commit history that the behavior has changed.

As for me, I always liked the slow start feature so it mimics a fluorescent bulb. Even if there is a small hickup in the animation. Would it be possible to make it a `setoption` setting so people can decide if thet want fade on startup or not?

Thanks for the rule suggestion, only downside is that it causes 1 additional flashwrite since the `powerstate` is toggled every boot. Except when you set `setoption0` off

TimelessNL, 2020-04-05:
> You can simulate this with rules, and start the fading right after wifi scan. To see what it looks like you can try PowerOnState 0 and Rule1 ON Power1#Boot DO Power On ENDON and Rule1 1

I tried the your given rule, but my SM2135 devices seem to fade-in just fine, but on all my PWM driven bulbs it does not, probably because the SM2135 is not so quick to respond. PWM bulbs do wake but when the `FADE` is finished it briefly turns completely `OFF` and then `ON` again after which it remains stable. So I think the `POWERONSTATE 0` and `POWER ON` are conflicting somehow, yet I see no mention of this and in the console there is no `OFF` reference to be found.

Jason2866, 2020-04-05:
If you compile yourself (not so easy possibel with Arduino IDE) and use 
Tasmota feature stage core (Arduino commit e64cb619f...) the blinking on off on
is maybe gone. Tasmota development has a code change for this.
ONLY possible since the Arduino commit written before.
Your used Arduino Core 2.6.3 is too old.

TimelessNL, 2020-04-05:
> If you compile yourself (not so easy possibel with Arduino IDE) and use
> Tasmota feature stage core (Arduino commit e64cb619f...) the blinking on off on
> is maybe gone. Tasmota development has a code change for this.
> ONLY possible since the Arduino commit written before.

Do you mean [this](https://github.com/esp8266/Arduino/commit/e64cb619f79a3c888dd56b957d8f48ce74f35735) commit? But shouldn't that be happening very early at boot? Because the ON->OFF->ON behavior I'm experiencing is more like 3sec after boot.

> Your used Arduino Core 2.6.3 is too old.

But the v8.2.0 public release is build against v2.6.3 (372a3ec) and is thus also affected by this bug?

Jason2866, 2020-04-06:
Yes, Tasmota release version is build on commit 372a3ec. The [GPIO blink fix](https://github.com/esp8266/Arduino/commit/e64cb619f79a3c888dd56b957d8f48ce74f35735l)
is NOT in the used commit for the release version 8.2.0 or in any precompiled bin.
You have to compile yourself (easy with Platformio using Platformio_override.ini) or copy manually the needed files in Arduino IDE from [commit](https://github.com/esp8266/Arduino.git#e64cb619f79a3c888dd56b957d8f48ce74f35735)

s-hadinger, 2020-04-06:
@TimelessNL It seems easy to add a SetOption to decide whether Fade is used at startup or not. Is it still useful?

TimelessNL, 2020-04-06:
Yeah that means `PowerOnState` is still used and  not overruled by the given rules. This makes it easier to understand in my opinion.

TimelessNL, 2020-04-06:
> Yes, Tasmota release version is build on commit 372a3ec. The [GPIO blink fix](https://github.com/esp8266/Arduino/commit/e64cb619f79a3c888dd56b957d8f48ce74f35735l)

But this commit fixes an "early boot" GPIO initialization issue. This is not what I'm seeing my ON->OFF-ON bug happens +/-3sec after boot, I don't think my issue is fixed by this commit?

s-hadinger, 2020-04-06:
I do see a change in colors briefly at power on for PWM, because the interrupt storm of wifi connection disturbs the PWM interrupts. But it's not a complete ON-OFF-ON.

In the meantime, you can force fade at startup by just changing one line in code:
in file `xdrv_04_light.ino` line 274, change to `bool     fade_initialized = true;` (instead of false)

s-hadinger, 2020-04-06:
PR submitted.

TimelessNL, 2020-04-06:
Wow! fast response, thanks!

TimelessNL, 2020-04-06:
> I do see a change in colors briefly at power on for PWM, because the interrupt storm of wifi connection disturbs the PWM interrupts. But it's not a complete ON-OFF-ON.
>
> In the meantime, you can [...]

May I ask what hardware you're using? I've run tests on a H801 and LSC Filament bulb, and they both share this ON->OFF->ON issue. I did test a NodeMCU with one of its onboard LEDs (GPIO16) as PWM, and where my H801 and LSC devices show this behavior every boot, the NodeMCU only shows it once it a while... very strange.

closed 2020-04-06 by arendst

s-hadinger, 2020-04-06:
I'm testing with MagicHome led driver. I also have E14 bulbs RGBW driven by PWM. I've never seen a full ON-OFF-ON.

Can you enable `WebLog 4` and share the logs?

arendst, 2020-04-06:
I think with regards to the power ON-OFF-ON this is the same issue as #8046 

If I look sharp at my LC01 with three channel PWM I do see this phenomenon in a mild way too. After about 4 seconds, when the wifi tries to connect, I see a slight increase in brightness which resumes soon to the the correct one.

I've tried without MQTT, power cycle detection (which writes to flash) all to no avail. I thing, as @s-hadinger  also suggested it is the wifi storm which stalls the PWM cycles. Depending on how wifi performs this may result in complete OFF as some users experience.

TimelessNL, 2020-04-06:
> I do see a change in colors briefly at power on for PWM, because the interrupt storm of wifi connection disturbs the PWM interrupts.

Yeah this seems to be the same for me on <= v8.1.0, it was never a problem for me just a side effect of software PWM. But v8.2.0 seems to have made things worse, it almost seems some value is reinitialized and therefor switches OFF->ON 🤔 

> But it's not a complete ON-OFF-ON.

May I ask what hardware you're using? I've run tests on a H801 and LSC Filament bulb, and they both share this ON->OFF->ON issue. I did test a NodeMCU with one of its onboard LEDs (GPIO16) as PWM, and where my H801 and LSC devices show this behavior every boot, the NodeMCU only shows it once it a while... very strange.

> Depending on how wifi performs this may result in complete OFF as some users experience.

@arendst To my knowledge you also have the SM2135 type LSC bulbs, do you by any chance also have the Filament style E14/E27? Since all the Filament style LSC bulbs I tested go ON->OFF->ON on v8.2.0. Because I also saw this small dip in brightness on v8.1.0 and below, but v8.2.0 made things worse (_same SDK 2.6.3_)

TimelessNL, 2020-04-06:
But regarding to the OP, this issue has been fixed by @s-hadinger .
So better continue the ON->OFF->ON issue in #8046?

arendst, 2020-04-06:
Let's continue in #8046

PR #8079 (2020-04-06): Add command ``SetOption91 1`` to enable fading at startup / power on
## Description:

Fading at startup (power on or reboot) is disabled by default because it cannot be rendered smoothly (short freeze maybe due to wiffi scan). Option91 allows to force fade at start-up.

**Related issue (if applicable):** fixes #8073

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core Tasmota_core_stage
  - [x] The code change pass travis tests. ** [... cut]