arendst/Tasmota #8074: unable to disable mqtt retained messages

s1nbad, 2020-04-05:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
after issuing switchretain 0 and buttonretain 0 the device still publishes LWT retained message

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
00:00:00 CFG: Loaded from flash at F8, Count 111
00:00:00 Project tasmota SWdual131 Version 8.2.0(tasmota)-STAGE
00:00:00 WIF: Connecting to AP1 PKHA in mode 11N as SWdual131...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on SWdual131 with IP address 192.168.1.131
08:38:02 MQT: Attempting connection...
08:38:02 MQT: Connected
NOTE: publish of retained message
08:38:02 MQT: tele/SWdual131/LWT = Online (retained)
NOTE: confirmation of status of switchretain and buttonretain
08:42:44 CMD: switchretain
08:42:44 MQT: stat/SWdual131/RESULT = {"SwitchRetain":"OFF"}
08:42:52 CMD: buttonretain
08:42:52 MQT: stat/SWdual131/RESULT = {"ButtonRetain":"OFF"}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
NOTE: savestate = 0
08:57:51 CMD: status 0
08:57:51 MQT: stat/SWdual131/STATUS = {"Status":{"Module":39,"FriendlyName":["SWdual131","SWdual131"],"Topic":"SWdual131","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:57:51 MQT: stat/SWdual131/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:56","StartupUTC":"2020-04-05T07:56:55","Sleep":50,"CfgHolder":4617,"BootCount":26,"BCResetTime":"2020-04-05T08:32:52","SaveCount":117,"SaveAddress":"FA000"}}
08:57:51 MQT: stat/SWdual131/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-20T14:45:23","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"313/699"}}
08:57:51 MQT: stat/SWdual131/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["PKHA",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008008","2805C8000100060000005A00000000000000","00000000","00000000"]}}
08:57:51 MQT: stat/SWdual131/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
08:57:51 MQT: stat/SWdual131/STATUS5 = {"StatusNET":{"Hostname":"SWdual131","IPAddress":"192.168.1.131","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"60:01:94:CA:27:78","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
08:57:51 MQT: stat/SWdual131/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.4","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CA2778","MqttUser":"mqttUser","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
08:57:51 MQT: stat/SWdual131/STATUS7 = {"StatusTIM":{"UTC":"2020-04-05T07:57:51","Local":"2020-04-05T08:57:51","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:18","Sunset":"19:27"}}
08:57:51 MQT: stat/SWdual131/STATUS10 = {"StatusSNS":{"Time":"2020-04-05T08:57:51","Switch1":"ON","Switch2":"ON"}}
08:57:51 MQT: stat/SWdual131/STATUS11 = {"StatusSTS":{"Time":"2020-04-05T08:57:51","Uptime":"0T00:00:56","UptimeSec":56,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"PKHA","BSSId":"D0:17:C2:E8:6B:50","Channel":11,"RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I want to turn off retained messages from being published

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-04-05:
Why no retained on LWT? See https://www.hivemq.com/blog/mqtt-essentials-part-9-last-will-and-testament/

s1nbad, 2020-04-05:
Thx, I understand the usefulness of LWT but in my circumstances I wish to disable them.
Am I missing a command in which I am able to disable them ?
I have tried powerretain 0 switchretain 0 powerstate 0 savestate 0

arendst, 2020-04-05:
There is no command to disable it. You'll have to edit the code and re-compile yourselves

s-hadinger, 2020-04-05:
This is a compile time option `#define MQTT_NO_RETAIN`

arendst, 2020-04-05:
How could I forget.

s-hadinger, 2020-04-05:
It was added by #7744

s1nbad, 2020-04-05:
Understood - thanks for your help

closed 2020-04-05 by s1nbad

s1nbad, 2020-06-07:
Hope you can help - are you able to confirm this command is still working
#define MQTT_NO_RETAIN 

I have compiled with this option set and still the LWT messages are retained it seems to have no effect

I have also tried #define MQTT_NO_RETAIN    true    -  same result

It is rainy season in Asia with constant power cuts and the retained messages are causing havoc - I simply want to disable the retain flag on all mqtt messages including LWT

s1nbad, 2020-06-09:
The behaviour I desire is to disable the retain flag being placed on ANY tele messages that are sent by the Tasmota software including LWT messages.

I had been advised that  #define MQTT_NO_RETAIN  was the solution but it does not seem to work - hence the question.

Not sure I understand why the offline and online messages of the broker won't work when doing this.  Surely, the only time a broker ever sends messages is when it sends from its database of stored messages (retained) - by disabling the retain flag it will not store those messages and therefore not send any messages and only retransmit client messages, which is exactly the behaviour I desire.

ascillato, 2020-06-09:
LWT messages are part of the MQTT standard. Disabling retain on LWT will make your device to show up as ONLINE and READY to receive commands in your home automation software while it is OFFLINE too, or it may not be controllable (depending on how LWT messages are configured on your home automation software). Please, check Tasmota MQTT Docs ( https://tasmota.github.io/docs/MQTT/ ) and the MQTT standard ( http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/mqtt-v3.1.1.html at **_3.1.2.5 Will Flag_** ).

Anyway, if you don't want to use the standard and you don't care about knowing if a device is ONLINE or NOT (may be hard to troubleshoot issues - that is why we don't recommend this change), you are able to compile by yourself and disable all LWT messages in my_user_config.h using the keys:

MQTT_LWT_MESSAGE and MQTT_NO_RETAIN

s1nbad, 2020-06-09:
ascillato - thanks for your help - appreciated

I do understand mqtt - I live in Asia and this time of year we experience VERY unstable power and with 200 devices the retained messages, for me, present an issue.  I am not asking for any changes to the standard the Will message can still optionally be sent with or without the retain flag.
As you say, anyway...

I have tried MQTT_NO_RETAIN and it does not seem to be working

It is used within   xdrv_02_mqtt.ino (line 317)
#if defined(USE_MQTT_TLS) && defined(USE_MQTT_AWS_IOT) || defined(MQTT_NO_RETAIN)
however the definition does not seem to make any difference to the #if statement

Even when I hardcode "#define MQTT_NO_RETAIN" immediately before the #if statement which allocates "retained=false" I still see no change in behaviour. 

I will take a look at MQTT_LWT_MESSAGE

ascillato, 2020-06-09:
> for me, present an issue.

Why a LWT message is an issue with VERY unstable power?

s1nbad, 2020-06-09:
It is a timing issue.  When the storms descend the power fluctuates with the devices, broker and servers coming on and off line within seconds repeatedly.  The broker sends the retained messages out which NodeRed then picks up, however, the devices are offline by the time my automation completes.

I quite understand that this is not a priority, as you say best practice dictates that this option should not used. I just thought I might be missing something as I cant get the definition MQTT_NO_RETAIN to make the change in behaviour documented.

To be fair I can resolve this by disabling the retention within the broker itself.  I was looking for a quick fix rather than rewrite my automations.  Completely understand if you want to leave the issue closed.

ascillato, 2020-06-09:
> the power fluctuates with the devices, broker and servers coming on and off line within seconds repeatedly. 

So, your server should be on an UPS to prevent this. If not, your server life spam is going to be very short.

> The broker sends the retained messages out which NodeRed then picks up.

So, you are using the LWT messages in your automations?

If so, they are Ok, to be sent. If a device is OFFLINE, and your broker starts, It will send the OFFLINE message to your home automation software. If your Tasmota goes ONLINE, your broker then will send ONLINE. (the ONLINE message is NOT retained) The MQTT broker is the one that sends an OFFLINE to the LWT topic only when your device stop responding, OR when your broker starts, OR when your home automation software subscribe to those topics.

The retained messages that should make issues to you, are the POWER ON and OFF messages. By default they are NOT retained and we recommend not to use retain on them.

>  I cant get the definition MQTT_NO_RETAIN to make the change in behaviour documented.

Have you deleted your retained messages of your broker? That needs to be done manually in the broker or by sending a retained empty message to the topic you want to delete.

s1nbad, 2020-06-09:
Really, I understand I can get round this, I am aware of what I should have and I will relook at my automations. Of course I have deleted the retained messages.

My question was whether there was any reason why MQTT_NO_RETAIN does not behave as documented and whether this is a change.  

In - xdrv_02_mqtt.ino (line 317)
#if defined(USE_MQTT_TLS) && defined(USE_MQTT_AWS_IOT) || defined(MQTT_NO_RETAIN)
the OR statement is quite clear BUT the definition MQTT_NO_RETAIN within my_user_config does not reach this .ino file.  the variable "retained" remains True.

I know this because if I hardcode #define MQTT_NO_RETAIN immediately before the above #if statement the "retained" variable changes to False.

BUT, neither of the above turns the retain flag OFF.
(I know my changes are being compiled as I change a non critical text field each time to check)

This behaviour is not as documented and you should be aware - unless there is something I am missing, hence the question.

arendst, 2020-06-09:
I've tested this today and it works as designed. Easily seen if you enable logging where it states (retained) if retain is on like used with command sensorretain 1. With the define enabled it does not show the (retained) message.

ascillato, 2020-06-09:
@s1nbad 

Peter,

The retain of LWT message is not being controlled by that key. The key MQTT_NO_RETAIN is only for sensor and power messages.

See:

https://github.com/arendst/Tasmota/blob/5f9b5c377dcd049cd562be806f5ff449f30f9e10/tasmota/xdrv_02_mqtt.ino#L504-L506

It is hardcoded as TRUE for LWT.

Anyway, I think it isn't good to disable retain for LWT. That would be a non-standard way of using the MQTT broker, so expect weird behaviours from your broker and tasmota.

I think your automations should be different and your server should have UPS. Disabling a part of LWT I think is not a solution.

s1nbad, 2020-06-09:
seems you came to the same conclusion as i did - it has no effect on LWT messages
glad the dos have been udated - thanks for your help

PR #8649 (2020-06-09): Update comment on MQTT Retained messages at my_user_config.h file
## Description:

Update comment on MQTT Retained messages at my_user_config.h file

**Related issue (if applicable):** fixes #8074

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core ESP8266 V.2.7.1
  - [x] The code change is tested and works on core ESP32 V.1.12.2
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contrib [... cut]