arendst/Tasmota #8076: PCF8574 has limited addresses. Why?

stefanbode, 2020-04-05:
### PROBLEM DESCRIPTION
With the change of Arend at Nov8th 2019 0x27 and 0x38 was removed. I do not see a reason for this, if we change a bit the detection phase.
 * I2C Address: PCF8574  = 0x20 .. 0x27 (0x27 is not supported),
 *              PCF8574A = 0x39 .. 0x3F (0x38 is not supported)

Proposed change:
Line 102-105:
```
    if ((PCF8574_ADDR1 +8) == pcf8574_address) {  // Support I2C addresses 0x20 to 0x27 and 0x38 to 0x3F
      pcf8574_address = PCF8574_ADDR2 ;
    }
  }
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): _____
- [ x] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ x] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
does not recognize on port 0x38

### EXPECTED BEHAVIOUR
recognize on 0x38

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2020-04-05:
Seems the change was made when the new version of i2c drivers was implemented. 
Stefan, are the information on I2CDEVICES.md and my_user_config correct?

```
2 | USE_PCF8574 | xdrv_28 | PCF8574 | 0x20 - 0x26 | 8-bit I/O expander
2 | USE_PCF8574 | xdrv_28 | PCF8574A | 0x39 - 0x3F | 8-bit I/O expander
```
```
//  #define USE_PCF8574                            
// [I2cDriver2] Enable PCF8574 I/O Expander (I2C addresses 0x20 - 0x26 and 0x39 - 0x3F) (+1k9 code)
```

stefanbode, 2020-04-05:
No, address range should include 27 and 38. I originally created the driver and now had to update one device. 0x38 is the default start port.

stefanbode, 2020-04-05:
Page 2:  http://www.ti.com/lit/ml/scyb031/scyb031.pdf

effelle, 2020-04-05:
Ok, if you don't mind I prefer wait @arendst before changing it because I really don't know if there was something blocking the addresses.

arendst, 2020-04-05:
Do not change. These changes are made to support lcd and any other device using a pcf to communicate.

Just select another supported address and both you and any lcd user are happy.

closed 2020-04-05 by effelle