arendst/Tasmota #8078: Failure with Rules Trigger for Switch State

Icefox2k, 2020-04-05:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I think there is a little mistake with the VALUE when using Switch1#State Trigger. When Switchmode is set to 1 it will puplish a value of 0 when switch input is on and a 1 when input is off. Same thing when Switchmode is set to 0 then it will puplish a 2 on every change of the Input 

In the past i was using the other Trigger Feature for Rules like Switch1#State=1 that function isn't available anymore so i tried with the new Switch1#State Trigger

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [X] Tasmota binary firmware version number used: 8.2.0.1
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: PlatformIO (MS VS)
- [X] Flashing tools used: Tasmota PyFlasher
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
`23:06:41 CMD: Gruppe 0, Index 1, Befehl "TEMPLATE", Daten ""
23:06:41 MQT: stat/tasmota/RESULT = {"NAME":"Wemos D1","GPIO":[255,255,21,255,6,5,255,255,9,4,10,255,255],"FLAG":15,"BASE":18}
23:06:41 SRC: Backlog
23:06:41 CMD: Gruppe 0, Index 1, Befehl "MODULE", Daten ""
23:06:41 MQT: stat/tasmota/RESULT = {"Module":{"0":"Wemos D1"}}
23:06:42 SRC: Backlog
23:06:42 CMD: Gruppe 0, Index 1, Befehl "GPIO", Daten "255"
23:06:42 MQT: stat/tasmota/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"21":"Relay1"},"GPIO3":{"0":"None"},"GPIO4":{"6":"I2C SDA"},"GPIO5":{"5":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"9":"Switch1"},"GPIO13":{"4":"DS18x20"},"GPIO14":{"10":"Switch2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}`

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
`23:07:53 CMD: Gruppe 0, Index 1, Befehl "RULE", Daten ""
23:07:53 MQT: stat/tasmota/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":449,"Rules":"on Switch1#state do Publish tele/tasmota/Switch1 %value% endon"}
23:07:53 SRC: Backlog
23:07:53 CMD: Gruppe 0, Index 2, Befehl "RULE", Daten ""
23:07:53 MQT: stat/tasmota/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
23:07:53 SRC: Backlog
23:07:53 CMD: Gruppe 0, Index 3, Befehl "RULE", Daten ""
23:07:53 MQT: stat/tasmota/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}`

```
- [X] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
`23:08:35 MQT: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["building_heater_circulation","Tasmota2"],"Topic":"tasmota","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[2,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:08:35 MQT: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota-minimal.bin","RestartReason":"Software/System restart","Uptime":"0T00:06:22","StartupUTC":"2020-04-05T22:02:13","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2020-04-04T23:03:08","SaveCount":206,"SaveAddress":"F6000"}}
23:08:35 MQT: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.2.0.1(display)","BuildDateTime":"2020.04.05 00:02:43","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"426/699"}}
23:08:35 MQT: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SmartHome WiFi",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
23:08:35 MQT: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":527,"Free":476,"Heap":25,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000407","8EA0A18F","04049FA1","000000C1","010000C0","00004080","00000000"],"Drivers":"1,2,4,8,10,13,16","Sensors":"1,2,5,6"}}
23:08:35 MQT: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"building_heater_circulation","IPAddress":"10.0.20.53","Gateway":"10.0.20.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.20.1","Mac":"EC:FA:BC:C5:96:4E","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
23:08:35 MQT: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.17","MqttPort":1884,"MqttClientMask":"building_heater_circulation","MqttClient":"building_heater_circulation","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
23:08:35 MQT: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"2020-04-05T22:08:35","Local":"2020-04-05T23:08:35","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00"}}
23:08:35 MQT: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-04-05T23:08:35","Switch1":"OFF","Switch2":"ON","DS18B20-1":{"Id":"011912E9811A","Temperature":20.9},"DS18B20-2":{"Id":"011912ED8B67","Temperature":20.8},"DS18B20-3":{"Id":"0119136E5B70","Temperature":20.9},"TempUnit":"C"}}
23:08:35 MQT: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-04-05T23:08:35","Uptime":"0T00:06:22","UptimeSec":382,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":25,"MqttCount":1,"POWER1":"OFF","POWER2":"ON","Wifi":{"AP":1,"SSId":"SmartHome WiFi","BSSId":"44:D9:E7:FA:9B:23","Channel":6,"RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:07"}}}`

```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
`23:09:51 RUL: SWITCH1#STATE performs "Publish tele/tasmota/Switch1 0"
23:09:51 SRC: Rule
23:09:51 CMD: Gruppe 0, Index 1, Befehl "PUBLISH", Daten "tele/tasmota/Switch1 0"
23:09:51 MQT: tele/tasmota/Switch1 = 0
23:09:55 RUL: SWITCH1#STATE performs "Publish tele/tasmota/Switch1 1"
23:09:55 SRC: Rule
23:09:55 CMD: Gruppe 0, Index 1, Befehl "PUBLISH", Daten "tele/tasmota/Switch1 1"
23:09:55 MQT: tele/tasmota/Switch1 = 1`

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Use Rules for puplishing a mqtt topic instead of triggering a relay directly. I am using IoBroker with the sonoff.plugin on the other side this is why my Mqtt Topic seams to be a little bit confused 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2020-04-05:
In your Status 0 report I can clearly see you are using switchmode 2, not switchmode 1:
```
"SwitchMode":[2,0,0,0,0,0,0,0]
```
Switchmode 2 has the output inverted (0=ON / 1 = OFF) compared to Switchoode 1, then is working as expected. 
Switchmode 0 can output just a TOGGLE signal (2) and again is working as expected.
Please read the [docs](https://tasmota.github.io/docs/Buttons-and-Switches/) about buttons and switches for more information. 
If you have any more question please address them to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for **Tasmota software feature requests and bug reporting**. Troubleshooting and setup assistance is more effective using an interactive forum.

Thanks.

Icefox2k, 2020-04-05:
yes that will be right - i tried  to find a way to go on with my project so i changed to switchmode 2 ! 
in switchmode 2 the following is puplished in the console on is the first publish off the second

`00:16:32 SRC: Rule
00:16:32 CMD: Gruppe 0, Index 1, Befehl "PUBLISH", Daten "tele/tasmota/Switch1 1"
00:16:32 MQT: tele/tasmota/Switch1 = 1
00:16:36 RUL: SWITCH1#STATE performs "Publish tele/tasmota/Switch1 0"
00:16:36 SRC: Rule
00:16:36 CMD: Gruppe 0, Index 1, Befehl "PUBLISH", Daten "tele/tasmota/Switch1 0"
00:16:36 MQT: tele/tasmota/Switch1 = 0`

Or did I get this wrong when connecting my gpio to ground then the switch is ON and should puplish a value of 1 in switchmode 1?

effelle, 2020-04-05:
From the docs:
```
SwitchMode 1
Set switch to follow mode (0 = OFF, 1 = ON)

At the time when the circuit is closed, Tasmota will send ON and opening the circuit sends OFF.
```

Icefox2k, 2020-04-05:
I had read the docs no worry about this but this isn't so
As you said when switch input is connected to ground it should be ON an puplishing 1 in switchmode 1 but this is the result

```
00:37:55 RUL: SWITCH1#STATE performs "Publish tele/tasmota/Switch1 0"
00:37:55 SRC: Rule
00:37:55 CMD: Gruppe 0, Index 1, Befehl "PUBLISH", Daten "tele/tasmota/Switch1 0"
00:37:55 MQT: tele/tasmota/Switch1 = 0
```

effelle, 2020-04-05:
Ok. I'll have a look tomorrow.

ascillato, 2020-04-06:
> In the past i was using the other Trigger Feature for Rules like Switch1#State=1 that function isn't available anymore so i tried with the new Switch1#State Trigger

Sorry that is not right. There isn't any new or old trigger, both triggers were working.

Testing your issue using Tasmota 8.1.0.3:

```lua
01:01:14 CMD: switchmode
01:01:14 MQT: stat/test/RESULT = {"SwitchMode1":2}
01:01:17 CMD: rule1
01:01:17 MQT: stat/test/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":450,"Rules":"on switch1#state do publish stat/MYTOPIC/switch %value% endon"}
01:01:20 RUL: SWITCH1#STATE performs "publish stat/MYTOPIC/switch 1"
01:01:20 MQT: stat/MYTOPIC/switch = 1
01:01:24 RUL: SWITCH1#STATE performs "publish stat/MYTOPIC/switch 0"
01:01:24 MQT: stat/MYTOPIC/switch = 0
```

So, switchmode 2 is behaving like switchmode 1. That is a new BUG. Older versions were working as explained in the docs. Adding BUG label.

ascillato2, 2020-04-06:
> In the past i was using the other Trigger Feature for Rules like Switch1#State=1 that function isn't available anymore so i tried with the new Switch1#State Trigger

Sorry that is not right. There isn't any new or old trigger, both triggers are working.

Testing your issue using Tasmota 8.1.0.3:

```lua
01:12:01 CMD: switchmode 0
01:12:01 MQT: stat/test/RESULT = {"SwitchMode1":0}
01:12:12 CMD: rule
01:12:12 MQT: stat/test/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":450,"Rules":"on switch1#state do publish stat/MYTOPIC/switch %value% endon"}
01:12:16 RUL: SWITCH1#STATE performs "publish stat/MYTOPIC/switch 2"
01:12:16 MQT: stat/MYTOPIC/switch = 2
01:12:19 RUL: SWITCH1#STATE performs "publish stat/MYTOPIC/switch 2"
01:12:19 MQT: stat/MYTOPIC/switch = 2
```
Ok, as expected and as explained in the docs.

```lua
01:13:27 CMD: switchmode1 1
01:13:27 MQT: stat/test/RESULT = {"SwitchMode1":1}
01:13:31 CMD: rule
01:13:31 MQT: stat/test/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":450,"Rules":"on switch1#state do publish stat/MYTOPIC/switch %value% endon"}
01:13:32 RUL: SWITCH1#STATE performs "publish stat/MYTOPIC/switch 0"
01:13:32 MQT: stat/MYTOPIC/switch = 0
01:13:34 RUL: SWITCH1#STATE performs "publish stat/MYTOPIC/switch 1"
01:13:34 MQT: stat/MYTOPIC/switch = 1
```
Ok, as expected and as explained in the docs.

```lua
01:14:13 CMD: switchmode1 0
01:14:13 MQT: stat/test/RESULT = {"SwitchMode1":0}
01:14:18 CMD: rule2
01:14:18 MQT: stat/test/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Free":393,"Rules":"on switch1#state=1 do publish stat/MYTOPIC/switch ON endon on switch1#state=0 do publish stat/MYTOPIC/switch OFF endon"}
01:14:20 KNX: Salida 1 = 1 enviada a 2.2.1
01:14:20 MQT: stat/test/RESULT = {"POWER":"ON"}
01:14:20 MQT: stat/test/POWER = ON
01:14:22 KNX: Salida 1 = 0 enviada a 2.2.1
01:14:22 MQT: stat/test/RESULT = {"POWER":"OFF"}
01:14:22 MQT: stat/test/POWER = OFF
```
Ok, as expected and as explained in the docs.

```lua
01:14:51 CMD: switchmode1 1
01:14:51 MQT: stat/test/RESULT = {"SwitchMode1":1}
01:14:53 CMD: rule2
01:14:53 MQT: stat/test/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Free":393,"Rules":"on switch1#state=1 do publish stat/MYTOPIC/switch ON endon on switch1#state=0 do publish stat/MYTOPIC/switch OFF endon"}
01:14:55 RUL: SWITCH1#STATE=0 performs "publish stat/MYTOPIC/switch OFF"
01:14:55 MQT: stat/MYTOPIC/switch = OFF
01:14:57 RUL: SWITCH1#STATE=1 performs "publish stat/MYTOPIC/switch ON"
01:14:57 MQT: stat/MYTOPIC/switch = ON
```
Ok, as expected and as explained in the docs.

So, there is no Bug.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-04-06 by ascillato2