arendst/Tasmota #8081: can not use MCP230xx-GPIO as (Shutter)Relay

berhomeauto, 2020-04-06:
### PROBLEM DESCRIPTION
My Problem is, that the shuttermodule (setoption80 1) "needs" "Relais" to control shutters.
As I unterstand a "Sensor29 Port" is NOT a Relais. And the MCP230xx driver will not map an output to power like the PFC8574 driver does?
What i did now ist using "dummy" Relais 1-8 on all available ESP-GPIOS and make a "mirror" rule to the ESP-GPIOS. That is working so far.
But it seems a little odd, that I have to "waste" ESP-GPIOS when using an I2C Port-Expander.
The Rule Hook is really ugly (IMHO).

Reading https://tasmota.github.io/docs/MCP230xx/ ist not helpful, because it only says that using an MCP is possible but will not explain how to use and what is more important to me: what is _not_ possible. 

Maybe it is more a thing like Setting up a MCP230xx-GPIO as Powerx/Relaisx in general?
Or am I missing some Point I do not know or understand?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched and asked for the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP-01
- [x] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [ ] IDE / Compiler used: vscode/PlatformIO
- [x] Flashing tools used: PlatformIO esptool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
11:26:53 RSL: stat/tasmota_DAFFF/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
11:26:53 RSL: stat/tasmota_DAFFFF/RESULT = {"Module":{"18":"Generic"}}
11:26:54 RSL: stat/tasmota_DAFFFF/RESULT = {"GPIO0":{"5":"I2C SCL"},"GPIO1":{"157":"LedLink"},"GPIO2":{"6":"I2C SDA"},"GPIO3":{"22":"Relay2"},"GPIO4":{"23":"Relay3"},"GPIO5":{"24":"Relay4"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"25":"Relay5"},"GPIO13":{"21":"Relay1"},"GPIO14":{"26":"Relay6"},"GPIO15":{"27":"Relay7"},"GPIO16":{"28":"Relay8"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
11:31:35 RSL: stat/tasmota_DAFFFF/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":154,"Rules":"ON power1#state DO sensor29 8,%value% ENDON ON power2#state DO sensor29 9,%value% ENDON ON power3#state DO sensor29 10,%value% ENDON ON power4#state DO sensor29 11,%value% ENDON ON power5#state DO sensor29 12,%value% ENDON ON power6#state DO sensor29 13,%value% ENDON ON power7#state DO sensor29 14,%value% ENDON ON power8#state DO sensor29 15,%value% ENDON"}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
none

### EXPECTED BEHAVIOUR
using MCP230xx (Sensor29) GPIOS to control shutterrelays without using "dummy" relais

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
none

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Sensor29 Configuration:
```
BackLog Sensor29 8,6,0; Sensor29 9,6,0; Sensor29 10,6,0; Sensor29 11,6,0; Sensor29 12,6,0; Sensor29 13,6,0; Sensor29 14,6,0; Sensor29 15,6,0
BackLog setoption80 1; ShutterRelay1 1; ShutterRelay2 3; ShutterRelay3 5; ShutterRelay4 7 
```
**(Please, remember to close the issue when the problem has been addressed)**

closed 2020-05-09 by stale[bot]

berhomeauto, 2020-05-09:
Hi Folks,
Maybe my question was too ... not specific enough? not relevant? annoying? boring?

A simple: yes thats the way to do it, or: no, you are completely wrong and maybe telling me why.
That is all i kindly asked for.

But anyway: Tasmota is awesome an thanks for your great work.

ascillato, 2020-05-09:
Sorry about that. Your issue did not have any answer from the community. Seems that very few people are trying to use MCP230xx-GPIO as (Shutter)Relay.
So far, it is not supported in Tasmota. This was requested before but no one was interested on adding support to it.
As this is open source, you can modify the code to add this functionality and if you want, you can provide a Pull Request to add that to Tasmota.
So far is not supported but a PR is always welcome.

MubiTec, 2020-08-29:
So, I guess I am the second one who have this problem. I want to do exactly the same.

dewayneshorts, 2021-05-22:
Me too! I can set them as output and control them in the Web UI but not through any HA system. Anyone found a way around this???

fabian727, 2022-01-29:
same here, playing around with the MCP and thought to expand it for shutters...

barbudor, 2022-01-29:
I'm surprised
MCT230xx outputs are normally recognized as Relays by Tasmota and repond to POWER commands.
From a brief look at the Shutter code, it call `ExecuteCommandPower( )` so it should
What happens ?
What kind of log (Weblog 3) do you see when trying to use ?

fabian727, 2022-01-29:
didn´t expect any feedback on this closed issue.
I didn't test it yet, Seems to work nowadays.
https://tasmota.github.io/docs/Blinds-and-Shutters/
also writes about it.

Nice Job guys! Thanks