arendst/Tasmota #8082: Separate Device Group Name Settings

pcdiem, 2020-04-06:
**Have you looked for this feature in other issues and in the docs?**  
Y

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
[PR #8014](https://github.com/arendst/Tasmota/pull/8014) adds the ability to subscribe to multiple MQTT topics. The same MQTT topics are used by modules that control multiple device groups (PWM DImmer). Being able to subscribe to multiple MQTT topics is a great addition for all devices, not just those that don't use device groups. Also, the purpose of GroupTopic2, 3 and 4 is now confusing.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
I'd like to propose adding settings text values specifically for Device Groups. 
In order to enable [PR #8014](https://github.com/arendst/Tasmota/pull/8014) to work with devices that support multiple device groups, I believe we should split the device group name(s) off into separate settings rather than continuing to use the MQTT topic(s).

This would also support and upcoming PR I plan to submit that implements the DevGroupSend command which allows sending updates to a device group via a command. This command will likely be used in rules to send updates to device groups other than the one(s) the device is a member of. This will require defining the device group names of any device groups the user plans to send updates to because we can only reliably send to known device groups (since we need to not just send a packet but make sure the packet is received and acknowledged by the known members of the group).

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
I could continue to use GroupTopics2, 3 and 4 but that prevents devices that control multiple device groups from taking advantage of [PR #8014](https://github.com/arendst/Tasmota/pull/8014) and keeps up the somewhat confusing dual-use of GroupTopics.

**Additional context**  
_Add any other context or screenshots about the feature request here._
I used the MQTT topic when I wrote device groups support because it was logical that the device group name would match the MQTT topic. However, when a device supports controlling devices in multiple device groups, there is no logical tie between multiple MQTT topics and multiple device group names.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-04-06:
I agree to add a command ``DeviceGroup`` or ``GroupName`` with up to FOUR groups similar like the current ``GroupTopic``.

arendst, 2020-04-06:
If you manage to move all code from ``support_device_groups.ino`` to a new ``xdrv_38_device_groups.ino`` file it makes integration easier as the new command code could be included (and disabled if needed) in this driver file just like any other driver.

It might be challenging with regards to compile time as files tend to be compiled in alphabetical order and you may encounter issues here (which might be solved with prototypes).

pcdiem, 2020-04-06:
How about DevGroupName<x> so it's consistent with the existing DevGroupShare and the, hopefully, upcoming DevGroupSend?

I was thinking the support_ modules were code that is used by multiple driver/sensor module. The existing DevGroupShare command is not compiled in if USE_DEVICE_GROUPS is not defined. If you still feel it's worthwhile, let me know and I'll try to move it to xdrv_38_device_groups.

arendst, 2020-04-06:
Ok. Keep it as it is for now. DevGroupName is fine.

pcdiem, 2020-04-08:
Done. [Add DevGroupName command #8087](https://github.com/arendst/Tasmota/pull/8087)

closed 2020-04-08 by pcdiem