arendst/Tasmota #8085: DeepSleepTime with exact value '10' not working; should work according to documentation

maelcum, 2020-04-06:
### PROBLEM DESCRIPTION
https://tasmota.github.io/docs/Commands/ states:
`DeepSleepTime:   10..86400 = set deep sleep mode time period in seconds.`

When using `DeepSleepTime 10` I get back:
```
    CMD: DeepSleepTime 10
    RSL: RESULT = {"DeepSleepTime":0}
```
Using `DeepSleepTime 11` works fine.

Confirmed in Discord chat by Blakadder:
Blakadder  Today at 5:13 PM
and reproduced here
possibly a bug since 10 should be minimum deepsleeptime

### REQUESTED INFORMATION

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
   yes, with 'is:issue DeepSleepTime'
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [-] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
  problem confirmed by Blakadder on 2020-04-06 CET
- [X] Device used (e.g., Sonoff Basic): Avatar AWP07L
  not device dependend
- [X] Tasmota binary firmware version number used: 8.2.0
  - [X] Pre-compiled
  - [-] Self-compiled
    - [ ] IDE / Compiler used: _____
- [-] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
17:36:53 CMD: Backlog Template; Module; GPIO 255
17:36:54 RSL: RESULT = {"NAME":"AWP07L","GPIO":[56,255,255,255,255,134,255,255,131,17,132,21,255],"FLAG":0,"BASE":18}
17:36:54 RSL: RESULT = {"Module":{"0":"AWP07L"}}
17:36:54 RSL: RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"134":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"131":"HLWBL SELi"},"GPIO13":{"17":"Button1"},"GPIO14":{"132":"HLWBL CF1"},"GPIO15":{"21":"Relay1"},"GPIO16":{"0":"None"}}
```
- [-] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```
17:37:39 CMD: Status 0
17:37:39 RSL: STATUS = {"Status":{"Module":0,"FriendlyName":["DE71364-PW-A5"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":900,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:37:39 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power On","Uptime":"0T00:11:01","StartupUTC":"2020-04-06T15:26:38","Sleep":250,"CfgHolder":4617,"BootCount":17,"BCResetTime":"2020-04-01T19:05:45","SaveCount":129,"SaveAddress":"F9000"}}
17:37:39 RSL: STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-20T14:45:23","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"428/699"}}
17:37:39 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["LVDW@bridgehead.it",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C8000100060000005A00000000000000","00000048","00000000"]}}
17:37:39 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
17:37:39 RSL: STATUS5 = {"StatusNET":{"Hostname":"DE71364-PW-A5","IPAddress":"192.168.254.95","Gateway":"192.168.254.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.254.254","Mac":"84:0D:8E:9B:03:57","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
17:37:39 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-04-06T15:37:39","Local":"2020-04-06T17:37:39","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"06:47","Sunset":"20:00"}}
17:37:39 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
17:37:39 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-04-06T17:37:39","ENERGY":{"TotalStartTime":"2020-04-04T12:13:04","Total":0.005,"Yesterday":0.003,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
17:37:39 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-04-06T17:37:39","Uptime":"0T00:11:01","UptimeSec":661,"Heap":27,"SleepMode":"Dynamic","Sleep":250,"LoadAvg":8,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"LVDW@bridgehead.it","BSSId":"44:4E:6D:84:41:A8","Channel":13,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:09"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
not applicable

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1. enter DeepSleepTime 0 in Console
2. observe result

### EXPECTED BEHAVIOUR
result should be:
17:39:03 CMD: DeepSleepTime 10
17:39:03 RSL: RESULT = {"DeepSleepTime":10}

result is:
17:39:03 CMD: DeepSleepTime 10
17:39:03 RSL: RESULT = {"DeepSleepTime":0}

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
asked Blakadder in Discourse chat:
  should I mention this at github somewhere? make an issue?
  it is rather minor
BlakadderToday at 5:23 PM
  yes, if it is a bug it'll be fixed no matter the size... if its intended i'll change the documentation

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-04-06:
My, my, my. It needs a docs update changing 10 to 11.

maelcum, 2020-04-06:
:) Well, I _did_ ask if I should open an issue for that ...

effelle, 2020-04-06:
Done.

closed 2020-04-06 by effelle