arendst/Tasmota #8089: Zigbee2tasmota crash with presence sensor (RTCGQ01LM)

plutec, 2020-04-06:
### PROBLEM DESCRIPTION
Tasmota crash (Hardware Watchdog appears) when presence sensor RTCGQ01LM (mijiia with no luminance) detects any movement

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X] Device used (e.g., Sonoff Basic): Sonoff Basic + CC2530
- [ ] Tasmota binary firmware version number used: Tasmota 8.2.0.1
  - [ ] Pre-compiled
  - [X ] Self-compiled
    - [X ] IDE / Compiler used: PlatformIO (Visual Studio Code)
- [X ] Flashing tools used: esptool
- [ X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
20:15:15 CMD: Backlog Template; Module; GPIO 255
20:15:15 MQT: stat/Z2T_1/RESULT = {"NAME":"Zigbee","GPIO":[0,0,0,0,0,0,0,0,0,166,0,165,0],"FLAG":0,"BASE":18}
20:15:15 MQT: stat/Z2T_1/RESULT = {"Module":{"1":"Sonoff Basic"}}
20:15:15 MQT: stat/Z2T_1/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"165":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"166":"Zigbee Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
20:15:54 MQT: stat/Z2T_1/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
20:15:54 MQT: stat/Z2T_1/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
20:15:54 MQT: stat/Z2T_1/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
20:16:09 CMD: Status 0
20:16:09 MQT: stat/Z2T_1/STATUS = {"Status":{"Module":1,"FriendlyName":["Z2T_1"],"Topic":"Z2T_1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:16:10 MQT: stat/Z2T_1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Hardware Watchdog","Uptime":"0T00:01:37","StartupUTC":"2020-04-06T19:14:32","Sleep":50,"CfgHolder":4617,"BootCount":49,"BCResetTime":"2020-04-02T17:00:29","SaveCount":91,"SaveAddress":"F7000"}}
20:16:10 MQT: stat/Z2T_1/STATUS2 = {"StatusFWR":{"Version":"8.2.0.1(tasmota)","BuildDateTime":"2020-04-02T22:05:30","Boot":7,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"358/699"}}
20:16:10 MQT: stat/Z2T_1/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["myssid",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
20:16:10 MQT: stat/Z2T_1/STATUS4 = {"StatusMEM":{"ProgramSize":618,"Free":384,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","8FD0E397","047683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,6,7,8,9,10,12,16,18,19,20,21,22,23,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
20:16:10 MQT: stat/Z2T_1/STATUS5 = {"StatusNET":{"Hostname":"Z2T_1-5363","IPAddress":"192.168.0.89","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"XX:XX:XX:XX:XX:XX","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
20:16:10 MQT: stat/Z2T_1/STATUS6 = {"StatusMQT":{"MqttHost":"XXX.XX.XX.XX","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_02D4F3","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
20:16:10 MQT: stat/Z2T_1/STATUS7 = {"StatusTIM":{"UTC":"2020-04-06T19:16:10","Local":"2020-04-06T20:16:10","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:16","Sunset":"19:28"}}
20:16:10 MQT: stat/Z2T_1/STATUS10 = {"StatusSNS":{"Time":"2020-04-06T20:16:10"}}
20:16:10 MQT: stat/Z2T_1/STATUS11 = {"StatusSTS":{"Time":"2020-04-06T20:16:10","Uptime":"0T00:01:38","UptimeSec":98,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"myssid","BSSId":"XX:XX:XX:XX:XX:XX","Channel":6,"RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [X ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
After enable weblog 4, I force the detection with the sensor and the console restart with the following content:
```
20:20:30 MQT: stat/Z2T_1/POWER = OFF
20:20:31 APP: Boot Count 51
20:20:31 CFG: Saved to flash at F4, Count 94, Bytes 4096
20:20:33 ZIG: ZbZNPSent 410000
20:20:34 MQT: tele/Z2T_1/STATE = {"Time":"2020-04-06T20:20:34","Uptime":"0T00:00:13","UptimeSec":13,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"myssid","BSSId":"XX:XX:XX:XX:XX:XX","Channel":6,"RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:07"}}
20:20:35 ZIG: Bytes follow_read_metric = 0
20:20:35 ZIG: {"ZbZNPReceived":"4180020200020603"}
20:20:35 MQT: tele/Z2T_1/RESULT = {"ZbState":{"Status":1,"Message":"CC2530 booted","RestartReason":"Watchdog","MajorRel":2,"MinorRel":6}}
20:20:35 ZIG: checking device configuration
20:20:35 ZIG: ZbZNPSent 2108000F00
20:20:35 ZIG: Bytes follow_read_metric = 0
20:20:35 ZIG: {"ZbZNPReceived":"6108000155"}
20:20:35 ZIG: ZbZNPSent 2102
20:20:35 ZIG: Bytes follow_read_metric = 0
[... 72 lines cut ...]
tele/Z2T_1/RESULT {"ZbState":{"Status":1,"Message":"CC2530 booted","RestartReason":"Watchdog","MajorRel":2,"MinorRel":6}}
tele/Z2T_1/STATE {"Time":"2020-04-06T20:22:44","Uptime":"0T00:00:13","UptimeSec":13,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"myssid","BSSId":"XX:XX:XX:XX:XX:XX","Channel":6,"RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:07"}}
tele/Z2T_1/RESULT {"ZbState":{"Status":50,"MajorRel":2,"MinorRel":6,"MaintRel":3,"Revision":20190608}}
tele/Z2T_1/RESULT {"ZbState":{"Status":3,"Message":"Configured, starting coordinator"}}
tele/Z2T_1/RESULT {"ZbState":{"Status":51,"IEEEAddr":"0x00124B001EFE82FF","ShortAddr":"0x0000","DeviceType":7,"DeviceState":9,"NumAssocDevices":3,"AssocDevicesList":["0x347B","0x500F","0x46EA"]}}
tele/Z2T_1/RESULT {"ZbState":{"Status":0,"Message":"Started"}}
```

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Pair the sensor and move your hand in front of it.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The expected behaviour is don't restart the sonoff and after some period of time detect the "no presence" with "Occupancy":0

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-04-06:
Hi,

"Exception":0 is more likely to a power issue. All sonoff basic are known to have a very limited power supply. If you are using the 3.3v of the sonoff basic, that is your issue. A sonoff basic can't handle more power than itself.

Please, change your sonoff basic for a nodeMCU with a good power supply, or supply 3.3v to your sensors with another power supply.

s-hadinger, 2020-04-06:
Someone already reported the exact same crash, but I cannot reproduce. It was reported that commenting our in `xdrv_23_zigbee_8_parsers.ino` line 537:

```
      // zigbee_devices.setTimer(shortaddr, 0 /* groupaddr */, OCCUPANCY_TIMEOUT, cluster, endpoint, Z_CAT_VIRTUAL_ATTR, 0, &Z_OccupancyCallback);
```

solved the crash. Unfortunately it does not generate the "Occupancy":0 event.

Can you share your compile options?

plutec, 2020-04-07:
I was making some tests with Zigbee, so the only change I did was to uncomment "#define USE_ZIGBEE" in "my_user_config.h"

plutec, 2020-04-07:
I will try with another power source this evening

plutec, 2020-04-07:
I just tested it with another big power supply and the behaviour is the same. 

Also, I checked with another presence sensor (aqara with luminance sensor) and also the Hardware watchdog appears.

plutec, 2020-04-07:
I got another different message (in MQTT): 

tele/Z2T_1/INFO3 {"RestartReason":{"Exception":0,"Reason":"Exception","EPC":["faf0fa2c","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40221e9f","4010102e","402297d9","4020e3c1","402343c4","40249634","40100b19"]}}

s-hadinger, 2020-04-08:
Thanks for your patience. Good news is I can reproduce the crash now. I will investigate.

s-hadinger, 2020-04-08:
Fixed in PR #8096

PR #8096 (2020-04-08): Fix Zigbee crash with Occupancy sensor (#8089)
## Description:

Setting a timer within a timer caused a memory corruption due to the implementation of `std::vector<>`

**Related issue (if applicable):** fixes #8089 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core Tasmota_core_stage
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Tas [... cut]
closed 2020-04-08 by arendst