arendst/Tasmota #8097: Add Sleep and Interval Read functionality to PMS5003 particle sensor to extend life

gururise, 2020-04-08:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
While reading of the PMS series of sensors (PMS3003/PMS5003/PMS7003) is supported, the sensor is set to ACTIVE (continuous) mode which leaves the laser on continuously and reduces the life of the sensor.  The PMS sensors support a PASSIVE mode which allows for sleeping the sensor and waking it up before taking a reading to significantly prolong the life of the sensor.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
I would like an option to take readings either continuously (as is now the default) or at intervals of between 1-255 minutes.  If the sensor is set to take reading at intervals, it will be put to sleep and woken up 30 seconds prior to the reading being taken (the warmup period), then the reading would be taken and the sensor put back to sleep until 30 seconds prior to the next reading.  This solution would require hooking up the TX pin to the sensor to allow Tasmota to send commands (ie. sleep, wake, mode) to the sensor, unless the user decides to keep it in its default mode.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
The alternative is to leave the sensor in its default ACTIVE mode which leaves the laser on continuously. The sensor works fine in this mode; however, its life is degraded.  The laser is rated for 6,000 hours.  (8,760 hours = 1 year).  By sleeping the sensor and waking it up, the life of the device can be extended to many years.

**Additional context**  
_Add any other context or screenshots about the feature request here._
This [blog post](https://www.superhouse.tv/38-diy-air-quality-sensor-part-1-basic-model/) describes the degradation of the sensor after 6,000 hours. (Scroll down to "Further Improvements" section).  Here is the [datasheet/manual](https://rsmith.home.xs4all.nl/files/plantower-pms5003-manual_annotated.pdf) of the PMS5003.

This is my first time using Tasmota; however, I think I can work on a PR for this issue.

**(Please, remember to close the issue when the problem has been addressed)**

gururise, 2020-04-10:
So I've been working on a PR for this and I'm having trouble introducing a new setting variable in settings.h to use as the interval value for minutes.  Introducing a new variable seems to mess up existing settings, and I'm not familiar enough with the settings behavior of Tasmota to feel comfortable introducing a new variable.

However, if I reuse an existing variable (currently uint8_t **novasds_startingoffset**) , everything works as described above.  I'm also thinking of changing the interval time from minutes to seconds instead, but that would require the use of a uint16_t instead.

Any suggestions?  Are there any non-used variables that I can utilize for the interval read time, or would the best approach be to implement a new variable?

closed 2020-04-12 by gururise

bwims, 2022-05-11:
Hi,

is this new functionality documented anywhere?

Thanks!

bwims, 2022-05-11:
Ah, never mind, It's in the PR!

thanks!