arendst/Tasmota #8099: Gosund SP111 random restarts

mgerhard74, 2020-04-08:
### PROBLEM DESCRIPTION
I use a few Gosund SP111 with tasmta 8.2.0 in combination with Domoticz/mosquito. All four Gosund randomly restarts. I can find this in the log:
00:00:09 MQT: tele/sonoff04/INFO3 = {"RestartReason":{"Exception":0,"Reason":"Exception","EPC":["e8402456","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4023c021","40101134","4022d0f6","4020da3d","4022cad8","4024315c","40100c15"]}}

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Gosund SP111
- [x] Tasmota binary firmware version number used: 8.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: ESP Easy Flasher
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
17:33:24 CMD: Backlog Template; Module; GPIO 255
17:33:24 MQT: stat/sonoff04/RESULT = {"NAME":"Gosund SP111 V","GPIO":[57,0,56,0,132,134,0,0,131,17,0,21,0],"FLAG":0,"BASE":45}
17:33:24 MQT: stat/sonoff04/RESULT = {"Module":{"0":"Gosund SP111 V"}}
17:33:24 MQT: stat/sonoff04/RESULT = {"GPIO0":{"57":"Led2i"},"GPIO1":{"0":"None"},"GPIO2":{"56":"Led1i"},"GPIO3":{"0":"None"},"GPIO4":{"132":"HLWBL CF1"},"GPIO5":{"134":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"131":"HLWBL SELi"},"GPIO13":{"17":"Button1"},"GPIO14":{"0":"None"},"GPIO15":{"21":"Relay1"},"GPIO16":{"0":"None"}}

  Configuration output here:
?

- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
no rules

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
17:34:58 CMD: status 0
17:34:58 MQT: stat/sonoff04/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff04"],"Topic":"sonoff04","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:34:58 MQT: stat/sonoff04/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Exception","Uptime":"0T00:18:47","StartupUTC":"2020-04-08T16:16:11","Sleep":50,"CfgHolder":4617,"BootCount":28,"BCResetTime":"2020-04-08T14:21:30","SaveCount":76,"SaveAddress":"F6000"}}
17:34:58 MQT: stat/sonoff04/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-20T14:45:23","Boot":7,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"308/699"}}
17:34:58 MQT: stat/sonoff04/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["AGM",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
17:34:58 MQT: stat/sonoff04/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
17:34:58 MQT: stat/sonoff04/STATUS5 = {"StatusNET":{"Hostname":"Sonoff04","IPAddress":"10.0.0.91","Gateway":"10.0.0.138","Subnetmask":"255.255.255.0","DNSServer":"10.0.0.138","Mac":"D8:F1:5B:A4:D1:AF","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:34:58 MQT: stat/sonoff04/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.142","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A4D1AF","MqttUser":"mosquito","MqttCount":6,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:34:58 MQT: stat/sonoff04/STATUS7 = {"StatusTIM":{"UTC":"2020-04-08T16:34:58","Local":"2020-04-08T17:34:58","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:12","Sunset":"19:31"}}
17:34:58 MQT: stat/sonoff04/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
17:34:58 MQT: stat/sonoff04/STATUS10 = {"StatusSNS":{"Time":"2020-04-08T17:34:58","ENERGY":{"TotalStartTime":"2020-04-08T14:09:22","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
17:34:58 MQT: stat/sonoff04/STATUS11 = {"StatusSTS":{"Time":"2020-04-08T17:34:58","Uptime":"0T00:18:47","UptimeSec":1127,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":6,"POWER":"OFF","Wifi":{"AP":1,"SSId":"AGM","BSSId":"50:D4:F7:14:D8:AB","Channel":2,"RSSI":22,"Signal":-89,"LinkCount":1,"Downtime":"0T00:00:09"}}}
17:34:58 MQT: stat/sonoff04/STATUS12 = {"StatusSTK":{"Exception":0,"Reason":"Exception","EPC":["e8402456","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4023c021","40101134","4022d0f6","4020da3d","4022cad8","4024315c","40100c15"]}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
The restart is random.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
No restarts, because some connected devices (TV; PCs) dont like short loos of ac power.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**
Ok

ascillato2, 2020-04-08:
Hi,

Have you tried as in previous Gosund issues to clean the flash?

Please, try in the console: `reset 3` (this will recalibrate the SDK settings for wifi), After the restart of this command you need to de-energize it and re-energize it again in order to fulfill the calibration.
Then do a `reset 5` (this will erase all your setting except wifi credentials and will clean your flash).
Then please update to latest Tasmota from development branch ( http://thehackbox.org/tasmota/tasmota.bin )
After that, please, reconfigure your device and retest. Thanks.

mgerhard74, 2020-04-09:
Hi, 
thank you for your feedback. The latest Tasmota 8.2.3 did not work, so i had to flash with my ttl-programer the Tasmota 8.2.0 version. I reset all four devices with "reset 3", plug off/on, "reset 5" and manually reconfigured them. Now, i got only two "Exception"-restarts in the last 3 hours of all four devices. This is better than before, but still happens.

ascillato, 2020-04-09:
Please, post the new `status 0` output when this restart happens. Thanks

mgerhard74, 2020-04-10:
Hello ,
here the "status 0" output:

06:36:08 CMD: status 0
06:36:08 MQT: stat/mosquito/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff03"],"Topic":"mosquito","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
06:36:08 MQT: stat/mosquito/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Exception","Uptime":"0T00:04:16","StartupUTC":"2020-04-10T05:31:52","Sleep":50,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2020-04-09T06:15:34","SaveCount":29,"SaveAddress":"F7000"}}
06:36:08 MQT: stat/mosquito/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020.03.20 14:54:51","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"345/699"}}
06:36:08 MQT: stat/mosquito/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["AGM",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00208009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
06:36:08 MQT: stat/mosquito/STATUS4 = {"StatusMEM":{"ProgramSize":578,"Free":424,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000407","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
06:36:08 MQT: stat/mosquito/STATUS5 = {"StatusNET":{"Hostname":"Sonoff03","IPAddress":"10.0.0.4","Gateway":"10.0.0.138","Subnetmask":"255.255.255.0","DNSServer":"10.0.0.138","Mac":"24:62:AB:20:2D:E6","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
06:36:08 MQT: stat/mosquito/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.142","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_202DE6","MqttUser":"mosquito","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
06:36:08 MQT: stat/mosquito/STATUS7 = {"StatusTIM":{"UTC":"2020-04-10T05:36:08","Local":"2020-04-10T06:36:08","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:08","Sunset":"19:34"}}
06:36:08 MQT: stat/mosquito/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
06:36:08 MQT: stat/mosquito/STATUS10 = {"StatusSNS":{"Time":"2020-04-10T06:36:08","ENERGY":{"TotalStartTime":"2020-04-09T06:15:34","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":233,"Current":0.000}}}
06:36:08 MQT: stat/mosquito/STATUS11 = {"StatusSTS":{"Time":"2020-04-10T06:36:08","Uptime":"0T00:04:16","UptimeSec":256,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"AGM","BSSId":"50:D4:F7:14:D8:AB","Channel":2,"RSSI":100,"Signal":-45,"LinkCount":1,"Downtime":"0T00:00:16"}}}
06:36:08 MQT: stat/mosquito/STATUS12 = {"StatusSTK":{"Exception":9,"Reason":"Exception","EPC":["40105fe9","00000000","00000000"],"EXCVADDR":"803fff35","DEPC":"00000000","CallChain":["4026ab16","402493d4","4026ad63","4024352e","40205f68","40243704","402436f8","40105fe9","40280e80","40105fe9","402697c7","40105fe9","40105a96","40105a78","40105fe9","400005e1","4010124c","4000050c","40105fe9","402434e0","402434d4","402491ea","4026ab16","402493d4","4026ad63","4024352e","40205f68","40243704","402436f8","40105fe9","40280e80"]}}

mgerhard74, 2020-04-10:
And from another "Gosund SP111"-device:
07:35:46 MQT: stat/mosquitto/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff01"],"Topic":"mosquitto","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
07:35:46 MQT: stat/mosquitto/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin","RestartReason":"Exception","Uptime":"0T00:01:05","StartupUTC":"2020-04-10T06:34:41","Sleep":50,"CfgHolder":4617,"BootCount":48,"BCResetTime":"2020-04-08T18:53:58","SaveCount":101,"SaveAddress":"F6000"}}
07:35:46 MQT: stat/mosquitto/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020.03.20 14:54:51","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"338/699"}}
07:35:46 MQT: stat/mosquitto/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["AGM",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00208009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
07:35:46 MQT: stat/mosquitto/STATUS4 = {"StatusMEM":{"ProgramSize":578,"Free":424,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000407","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
07:35:46 MQT: stat/mosquitto/STATUS5 = {"StatusNET":{"Hostname":"Sonoff01","IPAddress":"10.0.0.18","Gateway":"10.0.0.138","Subnetmask":"255.255.255.0","DNSServer":"10.0.0.138","Mac":"24:62:AB:34:E6:BF","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
07:35:46 MQT: stat/mosquitto/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.142","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_34E6BF","MqttUser":"mosquito","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
07:35:46 MQT: stat/mosquitto/STATUS7 = {"StatusTIM":{"UTC":"2020-04-10T06:35:46","Local":"2020-04-10T07:35:46","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:08","Sunset":"19:34"}}
07:35:46 MQT: stat/mosquitto/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
07:35:46 MQT: stat/mosquitto/STATUS10 = {"StatusSNS":{"Time":"2020-04-10T07:35:46","ENERGY":{"TotalStartTime":"2020-04-08T18:53:58","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":234,"Current":0.000}}}
07:35:46 MQT: stat/mosquitto/STATUS11 = {"StatusSTS":{"Time":"2020-04-10T07:35:46","Uptime":"0T00:01:05","UptimeSec":65,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"AGM","BSSId":"50:D4:F7:14:D8:AB","Channel":2,"RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:08"}}}
07:35:46 MQT: stat/mosquitto/STATUS12 = {"StatusSTK":{"Exception":0,"Reason":"Exception","EPC":["f0402456","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4023c275","40101134","4022d0f2","4020da3d","4022cad4","40243168","40100c15"]}}

ascillato2, 2020-04-22:
Hi, several fixes has been merged. Please, can you re test with latest version from the development branch? ( http://thehackbox.org/tasmota ) thanks.i

mgerhard74, 2020-04-23:
Sorry, the plug restarts also with version 8.2.0.4:

08:43:29 CMD: status 0
08:43:29 MQT: stat/mosquito/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff02"],"Topic":"mosquito","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:43:29 MQT: stat/mosquito/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota-DE.bin","RestartReason":"Exception","Uptime":"0T00:00:31","StartupUTC":"2020-04-23T07:42:58","Sleep":50,"CfgHolder":4617,"BootCount":445,"BCResetTime":"2020-04-09T06:06:39","SaveCount":594,"SaveAddress":"FA000"}}
08:43:29 MQT: stat/mosquito/STATUS2 = {"StatusFWR":{"Version":"8.2.0.4(717f3b9-tasmota)","BuildDateTime":"2020.04.22 18:06:19","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"355/699"}}
08:43:29 MQT: stat/mosquito/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["AGM",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00208009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
08:43:29 MQT: stat/mosquito/STATUS4 = {"StatusMEM":{"ProgramSize":585,"Free":416,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000407","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
08:43:29 MQT: stat/mosquito/STATUS5 = {"StatusNET":{"Hostname":"Sonoff02","IPAddress":"10.0.0.3","Gateway":"10.0.0.138","Subnetmask":"255.255.255.0","DNSServer":"10.0.0.138","Mac":"24:62:AB:34:E0:01","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
08:43:29 MQT: stat/mosquito/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.142","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_34E001","MqttUser":"mosquito","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
08:43:29 MQT: stat/mosquito/STATUS7 = {"StatusTIM":{"UTC":"2020-04-23T07:43:29","Local":"2020-04-23T08:43:29","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:42","Sunset":"19:54"}}
08:43:29 MQT: stat/mosquito/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
08:43:29 MQT: stat/mosquito/STATUS10 = {"StatusSNS":{"Time":"2020-04-23T08:43:29","ENERGY":{"TotalStartTime":"2020-04-09T06:06:39","Total":0.341,"Yesterday":0.010,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":232,"Current":0.000}}}
08:43:29 MQT: stat/mosquito/STATUS11 = {"StatusSTS":{"Time":"2020-04-23T08:43:29","Uptime":"0T00:00:31","UptimeSec":31,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"AGM","BSSId":"50:D4:F7:14:D8:AB","Channel":2,"RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:05"}}}
08:43:29 MQT: stat/mosquito/STATUS12 = {"StatusSTK":{"Exception":4,"Reason":"Exception","EPC":["400005cb","00000000","00000000"],"EXCVADDR":"00008000","DEPC":"00000000","CallChain":["400005cb","4026c126","401061e1","4026c373","4024508e","40206794","40245264","40245258","4024e3f9","400005cb","402826e0","400005cb","4026add3","400005cb","40105c8e","40105c70","400005cb","400005e1","401061e1","40249fb8","40245040","40245034","4000050c","400005cb","4026c126","401061e1","4026c373","4024508e","40206794","40245264","40245258"]}}

Should i reset with "reset 3" and "reset 5"?

closed 2020-05-23 by stale[bot]