arendst/Tasmota #8100: Basic Outlet crashed with any tasmota 7.2.x - 8.2.3

ryanm352, 2020-04-09:
### PROBLEM DESCRIPTION
I have 3 devices and they're all experiencing the same issue. The Web UI stops responding and seems to crash randomly requiring a physical unplug/re-plug back into an outlet. Though, MQTT will continue to respond to commands.

*TanTan WP3:*

ESP Chip Id | 12902232
-- | --
Flash Chip Id | 0x144051
Flash Size | 1024kB
Program Flash Size | 1024kB
Program Size | 498kB
Free Program Space | 504kB
Free Memory | 27kB

Other Smart Outlets with failures:
TanTan WP3, Potensic JH-G01U, BSD01
**_This issue also occurs when I upgraded a Potensic JH-g01u from 6.5.0 to 7.2.0 then to 8.2.0_**
Previously the Potensic was running flawlessly on 6.5.0 for over a year. (grrr)
This issue also occurs when using the minimal firmware.

I have downgraded to every allowable firmware from 8.2.3 all the way down to 7.2.0. The issue persists. I have also tried the latest core_staging along with 8.2.3 and the issue persists.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff Basic
- [X] Tasmota binary firmware version number used: 7.2.0 - 8.2.3
  - [X] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: Tasmota compiler
- [X] Flashing tools used: Tasmota OTA web ui flasher
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  03:03:38 CMD: Backlog Template; Module; GPIO 255
03:03:38 MQT: stat/pump/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
03:03:38 MQT: stat/pump/RESULT = {"Module":{"1":"Sonoff Basic"}}
03:03:39 MQT: stat/pump/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
03:05:56 CMD: Backlog Rule1; Rule2; Rule3
03:05:56 MQT: stat/pump/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
03:05:56 MQT: stat/pump/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
03:05:57 MQT: stat/pump/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
```
- [x] Provide the output of this command: `Status 0`:
```
03:06:34 MQT: stat/pump/STATUS = {"Status":{"Module":1,"FriendlyName":["Tasmota"],"Topic":"pump","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
03:06:34 MQT: stat/pump/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:13:42","StartupUTC":"2020-04-09T01:52:52","Sleep":50,"CfgHolder":4617,"BootCount":3,"BCResetTime":"2020-04-09T02:50:52","SaveCount":9,"SaveAddress":"FB000"}}
03:06:34 MQT: stat/pump/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-04-09T00:23:40","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"309/699"}}
03:06:34 MQT: stat/pump/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXXXX",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
03:06:34 MQT: stat/pump/STATUS4 = {"StatusMEM":{"ProgramSize":500,"Free":500,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8EC86383","043603A1","000000C7","010000C0","C000F981","00000004"],"Drivers":"1,2,3,4,6,8,10,12,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,3,4,5,6"}}
03:06:34 MQT: stat/pump/STATUS5 = {"StatusNET":{"Hostname":"pump-8024","IPAddress":"192.168.X.X","Gateway":"192.168.X.X","Subnetmask":"255.255.255.0","DNSServer":"192.168.X.X","Mac":"XX:XX:XX:XX:XX:XX","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
03:06:34 MQT: stat/pump/STATUS6 = {"StatusMQT":{"MqttHost":"media","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C4DF58","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
03:06:34 MQT: stat/pump/STATUS7 = {"StatusTIM":{"UTC":"2020-04-09T02:06:34","Local":"2020-04-09T03:06:34","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00"}}
03:06:34 MQT: stat/pump/STATUS10 = {"StatusSNS":{"Time":"2020-04-09T03:06:34"}}
03:06:34 MQT: stat/pump/STATUS11 = {"StatusSTS":{"Time":"2020-04-09T03:06:34","Uptime":"0T00:13:42","UptimeSec":822,"Vcc":3.505,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":49,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXXXX","BSSId":"AC:22:0B:CE:F6:A0","Channel":9,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Simply flash one of the devices above to 8.2.0 connect it to your wifi, and wait. after a few hours, the UI will stop responding.

### EXPECTED BEHAVIOUR
The WebUI should respond.

ascillato2, 2020-04-09:
Please, try the same troubleshooting than previous issues like: https://github.com/arendst/Tasmota/issues/8099#issuecomment-611091542

closed 2020-04-09 by ascillato2

ryanm352, 2020-04-09:
I tried the above with the same results. It’s definitely a bug.