arendst/Tasmota #8106: JSON CONFIG ERROR: 3 => Control character error, possibly incorrectly encoded

Septain21, 2020-04-09:
Hello

I installed tasmota admin on windows10 yesterday without problem because it worked a few hours. Then during a reboot and following an update I have this message:
JSON CONFIG ERROR: 3 => Control character error, possibly incorrectly encoded

What's the problem ? a solution ?

Thank you

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-04-09:
You found where to report the issue.

closed 2020-04-09 by arendst

Jason2866, 2020-04-09:
Tasmo Admin is not a tool programmed or maintained from Tasmota.
It is a 3rd party application for Tasmota. For help or issues go to
https://github.com/reloxx13/TasmoAdmin/issues

Closed since it is NOT a Tasmota issue.

Septain21, 2020-04-09:
Yes thank you