arendst/Tasmota #8109: witty template not complete

fragolinux, 2020-04-09:
antonio@amicum.eu Nutella01

{"NAME":"Witty Cloud","GPIO":[255,255,56,255,17,255,0,0,38,39,255,37,255],"FLAG":1,"BASE":32}

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
flashed a Witty Cloud module with latest dev 8.2.0.3 tasmota.bin file, selected model 32 (Witty Cloud) from module combo box, at restart NO value for the included LDR connected to ADC0 is present in tasmota web gui home... setting the template as per https://templates.blakadder.com/witty_cloud.html or per https://github.com/arendst/Tasmota/blob/development/TEMPLATES.md (both are the same) to: "{"NAME":"Witty Cloud","GPIO":[255,255,56,255,17,255,0,0,38,39,255,37,255],"FLAG":1,"BASE":32}" enabled the analog reading in web gui home as generic 0-1023 values, while enabling the correct template "{"NAME":"Witty Cloud","GPIO":[255,255,56,255,17,255,0,0,38,39,255,37,255],"FLAG":3,"BASE":32}" partially solved the issue, with Illuminance shown as LX from 0 to 65535, but with strange values that (with a smartphone led pointed at ldr at about 10cm distance) move from about 2000 jumping suddenly to 65000 while just moving the same led less than a 1cm nearer to the ldr... in a normally lighted room, values are below 10 LX...

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): 32 Witty Cloud
- [X] Tasmota binary firmware version number used: tasmota.bin 8.2.0.3
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: Tasmotizer
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
15:37:19 CMD: Backlog Template; Module; GPIO 255
15:37:19 MQT: stat/WITTY/RESULT = {"NAME":"Witty Cloud","GPIO":[255,255,56,255,17,255,0,0,38,39,255,37,255],"FLAG":3,"BASE":32}
15:37:19 MQT: stat/WITTY/RESULT = {"Module":{"0":"Witty Cloud"}}
15:37:20 MQT: stat/WITTY/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"56":"Led1i"},"GPIO3":{"0":"None"},"GPIO4":{"17":"Button1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"38":"PWM2"},"GPIO13":{"39":"PWM3"},"GPIO14":{"0":"None"},"GPIO15":{"37":"PWM1"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
15:37:57 CMD: Status 0
15:37:57 MQT: stat/WITTY/STATUS = {"Status":{"Module":0,"FriendlyName":["WITTY"],"Topic":"WITTY","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:37:57 MQT: stat/WITTY/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:11:08","StartupUTC":"2020-04-09T14:26:49","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2020-04-09T15:14:37","SaveCount":14,"SaveAddress":"F6000"}}
15:37:57 MQT: stat/WITTY/STATUS2 = {"StatusFWR":{"Version":"8.2.0.3(ef9ffaa-tasmota)","BuildDateTime":"2020-04-08T17:00:48","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"402/699"}}
15:37:57 MQT: stat/WITTY/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TIM-19074707","Shark-HOME2"],"TelePeriod":300,"Resolution":"55A180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
15:37:57 MQT: stat/WITTY/STATUS4 = {"StatusMEM":{"ProgramSize":580,"Free":420,"Heap":22,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
15:37:57 MQT: stat/WITTY/STATUS5 = {"StatusNET":{"Hostname":"WITTY-3960","IPAddress":"192.168.1.21","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"A0:20:A6:04:0F:78","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:37:57 MQT: stat/WITTY/STATUS6 = {"StatusMQT":{"MqttHost":"mrshark.ddns.net","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_040F78","MqttUser":"admin","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
15:37:57 MQT: stat/WITTY/STATUS7 = {"StatusTIM":{"UTC":"2020-04-09T14:37:57","Local":"2020-04-09T15:37:57","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:10","Sunset":"19:33"}}
15:37:57 MQT: stat/WITTY/STATUS10 = {"StatusSNS":{"Time":"2020-04-09T15:37:57","ANALOG":{"Illuminance":1}}}
15:37:57 MQT: stat/WITTY/STATUS11 = {"StatusSTS":{"Time":"2020-04-09T15:37:57","Uptime":"0T00:11:08","UptimeSec":668,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":1,"POWER":"OFF","Dimmer":10,"Color":"1A1A1A","HSBColor":"0,0,10","Channel":[10,10,10],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"TIM-19074707","BSSId":"20:B0:01:23:0E:93","Channel":6,"RSSI":36,"Signal":-82,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
just select the module 32 from configuration - module page

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
an analog reading showing in tasmota web gui home

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
here the correct module configuration, AFTER selecting adc0 as LIGHT (3)

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2020-04-09:
The luminance shown by the ldr depends mainly by the resistor used on the board. Use `AdcParam3` to tune it.

ascillato2, 2020-04-09:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-04-09 by ascillato2

fragolinux, 2020-04-10:
> Closing this issue as it has been answered.

shouldn't be updated both tasmota/tasmota_template.h and TEMPLATES.md to include the correct template, too? right now selecting the module 32 shows no info in webgui, and the template sets it to just analog and not light

fragolinux, 2020-04-10:
anyway, this seems to have addressed the issues, on witty module the smd resistor is a 471: adcparam 3,470,12518931,-1.405