arendst/Tasmota #8114: grpflg is set for all commands when one or more GroupTopics are not set

pcdiem, 2020-04-10:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
The fix for issue [#8095](https://github.com/arendst/Tasmota/issues/8095) seems to have caused the grpflg to be set for all commands when one or more GroupTopics are not set.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Execute the Hostname command from the console. The current hostname is displayed because the code thinks the command was executed via an MQTT publish to one of the group topics.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The hostname should be set.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-04-10:
Bugger

pcdiem, 2020-04-10:
Can I ask a question? I'm fairly new to embedded development so I don't have a good handle on the cost of things yet. My temporary solution to this was:

  bool grpflg = false;
  uint32_t real_index = SET_MQTT_GRP_TOPIC;
  for (uint32_t i = 0; i < MAX_GROUP_TOPICS; i++) {
    if (1 == i) { real_index = SET_MQTT_GRP_TOPIC2 -1; }
    char * group_topic = SettingsText(real_index +i);
    if (*group_topic && strstr(topicBuf, group_topic) != nullptr) {
      grpflg = true;
      break;
    }
  }

The same as yours except I'm used to thinking that calling SettingsText twice is more expensive both in terms of code size and CPU cycles. Is the compiler able to optimized your patch into a single SettingsText call? Is allocating stack expensive (like in char * group_topic = SettingsText(real_index +i);)?

Thanks for you input, I'm just trying to get a better handle on the right way.

arendst, 2020-04-10:
Using platformIO allows for easy testing and checking and comparing yours and my solution reveals axactly the same code size so I think it is safe to say the compiler is able to optomize my patch.

I have to agree that your implementation using less cycles is (much) better than mine.

It seems that every function is allowed to allocate up to max 4k of stack and I try to keep it as low as possible. Allocating the group_topic pointer is just 4 bytes of the stack which is freed as soon as the for loop is completed so it shouldn't be a problem.

pcdiem, 2020-04-10:
Sorry if this is the wrong place but, can you point me in the right direction to see how "platformIO allows for easy testing and checking"?

arendst, 2020-04-10:
Once compiled it reports the RAM and Flash usage:
```
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [======    ]  56.3% (used 46104 bytes from 81920 bytes)
Flash: [=====     ]  49.5% (used 506792 bytes from 1023984 bytes)
Creating BIN file "C:\temp\.pioenvs\tasmota\firmware.bin" using "C:\temp\.pioenvs\tasmota\firmware.elf"
```

closed 2020-04-10 by arendst